opentrack/opentrack #1259: Bug in interaction between "Toggle" function and custom center pose

shryke2a, 2021-04-17:
If you press "toggle" while you have any deviation entered in "Custom Center Pose" (Options ->Output menu) this deviation will keep adding to the tracker which negates the use of the "toggle" function (used to keep the camera centered on a particular place for instance).

GO63-samara, 2021-04-17:
I have not been able to reproduce this problem. I'm alright. The "Toggle" function works. Show a short video of your problem.

shryke2a, 2021-04-20:
Here you can find the video. 1st movements without custom center pose, I toggle it a few times without issues.

Then I edit the Custom center pose and as soon as I toggle it you can see the effects of the custom pose piling up.

https://user-images.githubusercontent.com/1359399/115410871-4ca5ef00-a1f3-11eb-998f-5b39ca4ca8a5.mp4

GO63-samara, 2021-04-20:
Now I see your problem.
I guess you have `Center at startup` enabled. If you are using the `Toggle` function it is best to disable `Center at startup`. Otherwise, when you turn on the view, you will have all the coordinates set in the `Custom center pose`.
Try disabling `Center at startup`.

shryke2a, 2021-04-22:
I don't believe you see the problem.

The problem is the view keeps spinning at a constant rate when toggled off IF a custom center pose has been set up.

Toggle is supposed to stop all tracking and keep the camera on a fixed point, and this bug makes the functiun useless when a custom center pose has been set up.

Center at startup changes nothing because the problem starts happenning when the view is toggled off and stops happening as soon as it's toggled back on.

This is not a mistake in setting up the app, this is very much a bug, an unwanted behaviour in the programm that makes one of it's function useless when the programm is set up in a particular way.

GO63-samara, 2021-04-22:
So you are talking about the `Toggle` button, and I checked the `Toggle tracking` button.

You mean, if in the `Custom center pose` panel at least one of the coordinates is not equal to 0, then when you stop using the `Toggle` button, the Octopus starts to rotate or shift, right?

This is a serious program bug. 
And here's the same problem: https://github.com/opentrack/opentrack/issues/1210

Finally, I understand you.
Right now I had no idea.

sthalik, 2021-04-23:
I can't see what happens by looking at the code alone, and I still don't have my environment set up.

See https://github.com/opentrack/opentrack/blob/master/logic/pipeline.cpp#L352

The variable `value` is initialized from `newpose` which shouldn't be either modified or leak to subsequent iterations.

GO63-samara, 2021-04-25:
I fixed incorrect rotation and movement of the Octopus when turning off the view with the `[Toggle]` or `[Toggle while held]` buttons.
Indeed, incorrect rotation and movement was showing up in `apply_zero_pos ()`. I added hold_ordered check. Now [Toggle] works correctly.
Confused, but when I tried to make a Pull request, I managed to immediately get into `master`. Take a look here: https://github.com/opentrack/opentrack/commit/0ddcf4eb94fa4ee530702bca2c54e390b3c6057d

zmz125000, 2021-08-16:
FixToggleButton.rar
https://sites.google.com/site/diyheadtracking/
https://drive.google.com/file/d/1-FASGQdVXVONO3_ULGlDHGZeD7GKL_-e/view