opentrack/opentrack #1260: Request: Wine/Proton handling improvements

JT8D-17, 2021-04-18:
The current implementation for handling WINE, WINE wrappers and Proton games essentially precludes Opentrack from being used with more than a single game at once, as Proton games in particular tend to be really picky about versions.
I could not get OT to work with a single Proton game so far.

Tripping stones were:

1) Opentrack assumes that Proton prefixes are stored in 
>  "/.steam/steam/steamapps/compatdata"
or
> "/.local/share/Steam/steamapps/compatdata"

This is not the case when the default Steam library is in another folder on the system.
 
Suggestion: Read the Proton prefix path from the "Wine (System)" field from the UI and assume the default location when not set by the user.

2) Opentrack only reads Proton runtime installations from
> ".local/share/Steam/compatibilitytools.d/"

Again, it would be better if users could just pass a path for when Proton runtimes are installed in other folders.
Maybe add a selector for "Wine (System)" (auto-uses the default WINE system path) and "Wine/Proton (Custom)" (user can pass a custom path; to be stored in the application profile .ini file).

3) To my understanding, WINE/Proton wrappers have to be built against the "libwine" version of the WINE runtime that the application is run in. This method breaks if the WINE runtime version of said application is different to the system's WINE version, requiring an upgrade/downgrade dance and essentially a single Opentrack build version for a particular application.

It would help here if users could pass an array of paths to "[WINE_RUNTIME]/dist/lib/libwine.so" to cmake which would then make a wrapper for each entry, naming the output **opentrack-wrapper-wine-[WINE VERSION].exe.so** for the parent "WINE-RUNTIME" folder or "system" (when using the system's Wine version), e.g. **opentrack-wrapper-wine-system.exe.so**, **opentrack-wrapper-Proton-6.1-GE-2.exe.so**, **opentrack-wrapper-Proton 4.11.exe.so".

The WINE menu in the UI would then allow users to pick the appropriate wrapper for the target WINE/Proton runtime.

Would this be a feasible solution to the issue?

sthalik, 2021-04-20:
> I just dug through the source code for the WINE stuff but could not find any reference to libwine or other external WINE dependencies. If there are any, where are they located?

See `winegcc` invocation in the build system.

What you're proposing is too complicated. Proton is already hard to configure with opentrack as it is. It's Valve who should fix this mess -- for instance, by adding an API to query about Proton versions, not us.

JT8D-17, 2021-04-20:
> See `winegcc` invocation in the build system.

Yeah, I already found that.

> What you're proposing is too complicated. Proton is already hard to configure with opentrack as it is. It's Valve who should fix this mess, not us -- for instance, by adding an API to query about Proton versions.

What are you referring to?
Opentrack already checks multiple locations for a Proton installation and associated prefix. I don't see why it should be too complicated to just have a user pass it an alternative path?
Same for a nonstandard, i.e. WINE binary/runtime location, which is pretty much what Proton is. Or am I missing something here (I'm aware of the LD_LIBRARY_PATH requirement).

Regarding libwine: From WINE 6.0 onward, [image], meaning that backward compatibility with WINE releases < 6.0 is essentially broken. Libwine will also be completely removed in future WINE releases, breaking any module that requires it.However, modules built against WINE 6.0+ are also not compatible with earlier releases. This puts everything into a deadlock.

How feasible would it be to build the wrapper with something WINE-agnostic like mingw?
Do you know if modules built against old libwine releases like from WINE 4.0 are forward compatible and usable by, say, WINE 5.11? Or are they rigidly version locked?
If so, one could maybe provide prebuilt wrappers for older (< 6.0) and newer (>6.0) WINE releases.

I am playing with these ideas in my opentrack fork. Progressed isn't much farther than the frontend stuff at the moment and I'm really only checking how far my shabby C-skills will take me with this.
Anyway, this is my adapted WINE menu based on suggestion 1 and 2:

[image]

A "Proton" string in the runtime path would launch the Proton-specific stuff while other paths would launch WINE-specific stuff.
The easier method of setting the Proton runtime and the Proton prefix from the Steam AppID is still available.

sthalik, 2021-04-22:
> What are you referring to?

The configuration dialog is already overly complicated compared to other output methods.

> Libwine will also be completely removed in future WINE releases
> How feasible would it be to build the wrapper with something WINE-agnostic like mingw?

That's fucking retarded. I certainly don't welcome having to fix it afterward. Let's wait till it breaks then.

Right now the data transfer is done through `mmap(2)` without piping stuff through a socket. I guess it'll still be possible to build against a Wine source tree and  retrieve the underlying file descriptor for a `MapViewOfFile` handle.

> Do you know if modules built against old libwine releases like from WINE 4.0 are forward compatible and usable by, say, WINE 
> 5.11? Or are they rigidly version locked?

Every Wine release is a potential `wineserver` ABI bump. Any compatibility between Wine releases is purely accidental.

> I am playing with these ideas in my opentrack fork. 

Just don't bother sending data over a socket. It most certainly can be done with `mmap(2)`.

> The easier method of setting the Proton runtime and the Proton prefix from the Steam AppID is still available.

Can you please elaborate?

JT8D-17, 2021-04-25:
Revamped the entire WINE/Proton dialog and its underlying mechanics.
The dialog now scans for all valid Proton apps on the system, stores information like runtime and prefix paths and adds them to a selector. Only Proton apps with valid runtime and prefix paths are listed. Probably the least complicated solution for users.

Writing to the profile .ini file works, but exisiting profiles will most likely require updating for the new variables that I'm using.

[image]

Next on the list is revamping the wrapper initialization logic with regard to my new path handling system.

Are those various Steam runtime paths that are passed to the environment required to launch a Proton WINE server from outside Steam?

I hope a successful (first) launch of WINE-wrapper is characterized by finding it in the running processes and finding the paths to the TrackIR and/or Freetrack fake DLLs in the registry of the app's prefix. Because I have no idea how else I can test that the correct Proton instance has been launched.

After that, I'm going to try to tackle the wrapper/winelib issue, but that'll probably require a whole lot of studying beforehand.

JT8D-17, 2021-04-29:
Well, I've made the WINE handler more flexible now. I can basically set the path to any regular WINE release folder that contains /bin/wine and any prefix folder and get a wrapper process in turn.

I also did not observe any versioning issue with the wrapper. Compiling it against WINE 6.7 (system-installed) had the process launch just fine with a non-Arch (Ubuntu) WINE 5.7 runtime from Winehq. Same for compiling against 5.22 (system-installed) and launching it with a non-Arch (Fedora) WINE 6.2 runtime. Each test involved a fresh WINE prefix and generated the correct registry entries for Freetrack and TrackIR. So it seems to be less touchy than I thought. I didn't try it with my primary use case (modded IL-2 1946) yet.

What remains extremely touchy and basically broken, however, is Proton. My modified "module_status wine::initialize()" logic correctly determines the paths to the game's prefix (via app ID) and Proton runtime folder (no matter if installed in any Steam library or ./steam/root/compatibilitytools.d) and sets the "LD_LIBRARY_PATH", "WINEDLLPATH", "WINEPREFIX", "WINEPATH", "WINESERVER", "WINELOADER", "WINEESYNC" and "WINEFSYNC" environment variables according to some other issues mentioned here, but neither Proton 6.1-GE, Proton 6.3 and Proton 6.5-GE works. I have not tested it with other releases (yet).

I've pushed those changes to my [fork](https://github.com/JT8D-17/opentrack), in case anybody wants to have a laugh at substandard C++ code (I learned as I went and it was painful).

sthalik, 2021-04-30:
Getting registered Proton versions out of the `.vdf` registry makes total sense. It should definitely be done this way.

I'm worried about making further improvements to the Wine protocol as they're planning to remove winelib altogether and Proton will soon follow.

JT8D-17, 2021-04-30:
I've also tried to compile the wrapper with MingW as wineg++ apprarently is fully MingW compatible, but it didn't work out at all. Meh.

Continuing compatibility testing on my desktop, I compiled the wrapper against WINE 4.0 (basically the oldest release in the Arch Linux Archive) and it worked with WINE 5.0 and 6.7. Compiling the wrapper against WINE 5.22 only had it work with any newer release.

Regarding Proton, I could not get the wrapper to work at all. Not sure if this is known or new, but Proton's WINE server seems to be sandboxed, as seen with Proton 6.1:

[image]

Was this always the case?

JT8D-17, 2021-05-01:
SMALL SUCCESS

With the wrapper that I had compiled against WINE 4.0, I managed to get successful tracking output in [Reentry](https://store.steampowered.com/app/882140/Reentry__An_Orbital_Simulator/), using Proton-5.9-GE-8-ST.
All it took was selecting Reentry in the WINE/Proton dialog of my fork, enable TrackIR in Reentry itself and start tracking in opentrack.
So this basically reinforces the above hypothesis that older Proton releases behave just like regular WINE releases and that wrappers are forward compatible. It's just the new, sandboxed Proton releases that are causing trouble.

Not sure what to make of this. I've considered a cmake path variable to a custom folder for wine g++ so that users can compile the wrapper against any non-system WINE version, but I think I'm going the really lazy way here and simply redistribute the version 4.0 wrapper as a fallback along with the Freetrack, etc. DLLs and then implement something to select it in the UI.

JT8D-17, 2021-05-01:
Well, the lazy route it is. Users can now keep as many different wrapper versions inside `libexec/opentrack` as long as the filename satisfies the search string (`opentrack-wrapper-wine*.exe.so`).

[image]

I've also investigated that cmake path variable method, but wrapper compilation always fails when trying to compile against a non-system wineg++ and I just don't know why. In case anybody desires to try this further, I'll leave the cmake flag in my fork as `PATH_WINE_GPP`. It should be pointed at a folder containing `/bin/wineg++` and is also used by the `-B` flag of wineg++. It's off by default, which falls back to the `wineg++` system call.

As for the Proton front, versions newer than 5.13 are generally launched inside a sandbox, using Steam's Runtime Soldier. I have no idea how to gain access to the WINE processes running in such said sandbox for launching the wrapper, so I'll just consider any newer Proton release than 5.9 as generally incompatible with opentrack.

JT8D-17, 2021-05-02:
The modified WINE runtime handling from my fork has now passed its application-related test. 
Launched IL-2 1946 with a non-system WINE runtime (because it's still broken with WINE 6.7, at least with mods), pointed opentrack to said runtime (and IL-2's WINE prefix) and I got headtracking output just as I used to. Perfect!

What remains elusive, however, is headtracking in Elite Dangerous (using Proton 5.9-GE-8-ST). The wrapper launches just fine, but the game does not accept input from either npclient.dll or freetrack.dll. I've even tried to move them to a folder in `C:\Progrem Files` in the prefix and pointed the registry path there, but that didn't work. Meh.

(Sorry for the spam.)

sthalik, 2021-05-02:
> As for the Proton front, versions newer than 5.13 are generally launched inside a sandbox, using Steam's Runtime Soldier. I have no idea how to gain access to the WINE processes running in such said [...]

You probably still have access to the Steam directory. Might as well copy/hard link `NPClient{,64}.dll` and the wrapper to either Steam or game directory.

JT8D-17, 2021-05-02:
> You probably still have access to the Steam directory. Might as well copy/hard link `NPClient{,64}.dll` and the wrapper to either Steam or game directory.

I figure adding them to the prefix would also do, but I see the issue more on the side of the wrapper being spawned as a child process of opentrack and thus being unable to "see" the containerized wineserver processes.

JT8D-17, 2021-05-05:
Small success story for Proton 5.9 (barely related, but I'dlike to get the info out there): 
A short test yielded successful headtracking in Elite Dangerous. I think the key was creating `Program Files (x86)/NaturalPoint\TrackIR5` in ED's WINE prefix, i.e. `steamapps/compatdata/359320/pfx/drive_c`, with `TrackIR5` being a symlink to `libexec/opentrack` (copying the DLL files into that folder instead may have also worked). Accoding to the internet, ED is only looking for the DLL(s) that installation directory.
I've also tried editing the registry paths for the Freetrack and TrackIR plugins, but I forgot that those get reset to their defaults when the wrapper starts.

pan-mroku, 2021-06-25:
Hello,

for me Proton 5.9 is no longer available. With 5.13 the wrapper simply finishes (I haven't analyzed why yet) and 5.0 is too old, and ED works much worse with it (it's just unusable, not only worse).
I've tried @JT8D-17's branch with same results. Copying dlls and so files to `NaturalPoint` or ED installation directory didn't help.
How can I help with this issue? Does dropping libwine mean that newer versions don't provide something needed by Opentrack at all, or just that `proto-wine` needs a rework to work with those?

Btw, in https://github.com/JT8D-17/opentrack/blob/main/proto-wine/proton_handling.cpp#L66 the function has to return something at the end. Without `return false` I'm getting SIGABORT due to stack smashing.

Dumb question: should I choose VR option when launching Elite, or should head tracking work in regular version too? It's the first time I'm trying to use headtracking and I'm not sure.

JT8D-17, 2021-06-25:
Proton 5.9 is still available [here](https://github.com/GloriousEggroll/proton-ge-custom/releases/tag/5.9-GE-8-ST).

I have essentially given up on trying to use headtracking in ED as I could only make it work exactly once and never again after that. With Odyssey, I now also need a release from the 6.x branch of Proton to get normal rendering performance, so unless somebody figures out how to hook the wrapper into Proton's bubblewrap container or devises another way to push head tracking data to it, it's essentially a lost cause.

The only time it worked, I picked the regular ED mode, not the VR one.

@pan-mroku I've appended the return condition to that function and pushed the change to my repo (and also merged from upstream).

bekopharm, 2021-06-29:
So close and yet so far.

I tried the same but whatever Wine version I compiled the Glue with, 5.8 to 6.10, seems to have no effect. Whenever I try to run it against a more recent Proton it just seg faults in `ntdll.dll` and is gone - or complains about an ABI mismatch. Even when I try the Wine binary provided by Proton itself! Your fork helped greatly with trying different versions @JT8D-17 but I eventually gave up for now too.

My last hope was getting my own Proton build and run it without Pressure Vessel / Sandboxing and while I can compile good old Wine just fine I fail already at setting up that dreaded build environment Proton wants. Vagrant or Docker - both failed me with useless error messages telling me to "fix my setup". Whoever thought forcing virtualization techniques just to compile some program to "save people from hunting down dependencies" is good idea. This is so frustrating.

Needless to say that it works with plain Wine.

pan-mroku, 2021-07-01:
Would something like [this](https://gitlab.steamos.cloud/steamrt/steam-runtime-tools/-/issues/14) help?
As far as I understand if we ran `pressure-vessel-launcher` on our machines and somehow add a script to ED that in the container would execute opentrack instead of ping in the example, that might work?

Or maybe we should simply create an issue on their gitlab and ask for a way to execute things inside their container?

JT8D-17, 2021-07-02:
Opening an issue on their Gitlab may indeed be the best course of action. However, I have little motivation to do this and try to understand the witchcraft that makes `pressure-vessel` work.

On a more positive note, the combination of IL-2 1946, Opentrack and WINE 6.11 finally seems to work again.

pan-mroku, 2021-07-05:
I managed to make it work with wine version of opentrack. From [here](https://github.com/ValveSoftware/steam-runtime/blob/master/doc/reporting-steamlinuxruntime-bugs.md#getting-a-shell-inside-the-container):

> pressure-vessel has a mode where it will run an `xterm` instead of the game, so that you can explore the container interactively. To enable this, run Steam with the `PRESSURE_VESSEL_SHELL` environment variable set to `instead`

In `xterm` spawned first run the game by `"$@"` and send it to background (in my case `CTRL-Z` `bg` but possibly adding `&` at the end could also work). It needs to be run first because of the `waitforexitandrun` verb in the `%command%`.
Then I've executed `python3  /home/pawel/newhome/SteamLibrary/steamapps/common/Proton\ -\ Experimental/proton run /home/pawel/newhome/SteamLibrary/steamapps/common/Elite\ Dangerous/opentrack_wine/opentrack.exe`

And it works! But it would be nice if we could somehow add this to `%command%` instead of all this writing.

I did try linux version of opentrack, but in the container it lacked the libraries it was built against. And there is no cmake etc there, so I'm not sure how to rebuild it inside the container.

bekopharm, 2021-07-08:
Good catch @pan-mroku 
I'll try this too. It's a lot of hoops to jump. Apparently the SDK runtimes come with compilers n stuff :thinking:

pan-mroku, 2021-07-12:
I have it organized like this:

> ~/newhome/SteamLibrary/steamapps/common/Elite Dangerous/Products/elite-dangerous-odyssey-64#  cat ./shortcut.sh
> #!/bin/bash
> 
> "$@" >/dev/null 2>&1 & sleep 1 && ../../opentrack.sh

> cat  ~/newhome/SteamLibrary/steamapps/common/Elite\ Dangerous/opentrack.sh 
> #!/bin/bash
> 
> exec python3 /home/pawel/newhome/SteamLibrary/steamapps/common/Proton\ -\ Experimental/proton run /home/pawel/newhome/SteamLibrary/steamapps/common/Elite\ Dangerous/opentrack_wine/opentrack.exe>/dev/null 2>&1

and in opentrack I've set it up to start in tray, start tracking when it starts and connected `steamapps/common/Elite Dangerous/Products/elite-dangerous-odyssey-64/EliteDangerous64.exe` to my config.
So when I type in xterm

> ./shortcut.sh "#@"

everything starts up automatically.

JT8D-17, 2021-07-13:
@pan-mroku
Can you try launching only Opentrack's WINE wrapper inside `pressure-vessel`? 
Unless the sandboxing also affects shared memory functions, this may be all that's needed to have (Linux) Opentrack communicate with its wrapper.

pan-mroku, 2021-07-13:
I'd love to, but I have no idea how to do this to be honest. I won't lie, I have little knowledge about opentrack code.
There is a binary for executing that? I think I saw only an .so executed by a process. Or should I write something that would blindly launch that? I remember that shared memory was based on registering for it using the same name. Is it hardcoded?

*Od: *BK ***@***.***>
*Do: *opentrack/opentrack ***@***.***>
*DW: *pan-mroku ***@***.***>; Mention ***@***.***>
*Data: *13 lip 2021 16:57:23
*Temat: *Re: [opentrack/opentrack] Discussion: Wine/Proton handling improvements (#1260)

JT8D-17, 2021-07-14:
I totally forgot that the wrappers are not native Windows executables, so launching them via Proton naturally won't work.

Looks like your approach with running Opentrack's native Windows build in Proton is the only viable option for the forseeable future.

waterlubber, 2021-07-14:
Would a native, thin executable - entirely responsible with interfacing with the game - work? Such an executable could be designed to be environment-agnostic (i.e it would run on Windows, WINE, Proton just fine) and handle translating between some sort of IPC format (eg UDP) to the FreeTrack/NPClient protocol used by Elite. I'm not nearly familiar enough with either Opentrack's code or Windows development to write something like this, but it might be the best option going forward (and would probably be a good deal more reliable than the wrapper business).

Essentially, just a windows .exe that translates Opentrack "Native UDP over network" to FreeTrack. As simple as possible.

JT8D-17, 2021-07-17:
So I've tried @pan-mroku's method of launching ED and the Windows build of OT with a Bash script and I got headtracking output in Odyssey after a few tries. I think the recipe for success is that forking programs out of Steam avoids running them inside _pressure-vessel_. To make the process a bit more "plug and play", I've written a script with minimal configuration, intended for (and only working with) [the minimal ED launcher](https://github.com/Rfvgyhn/min-ed-launcher).

&nbsp;

### Caveats
- Only tested with Proton-6.12-GE-1 from [GloriousEggroll's repo](https://github.com/GloriousEggroll/proton-ge-custom/releases). Regular Proton Experimental and 6.3 outright refuse to work with Odyssey for me.
- Requires a Windows build of OT in `[...]/steamapps/common/Elite Dangerous/z_opentrack-win`. Edit the path in the script if you prefer another location.
- Reading the Proton path from *config_info* in the app's Proton prefix folder may not always work.
- Using this with the standard launcher makes ED throw a path error on process start on my system. I have very little motivation to fix that as [the minimal launcher](https://github.com/Rfvgyhn/min-ed-launcher) is just so much better and much, much faster. If you do decide to test the regular launcher, clean up its processes manually after ED crashes or it'll block a restart.
- Opentrack may steal ED's window focus after launch. Maybe that just happens because I'm running ED in borderless mode.
- I've only tested OT input via UDP (from SmoothTrack running on a phone). No guaranteees for other methods.
- It's best to keep the Linux and Windows profiles for OT separate (i.e. no symbolic linking) as mixing and matching may produce bugs and confusion.

&nbsp;

### Requirements
- ED should work and be playable
- The `MinEdLauncher` binary must be inside `[...]/steamapps/common/Elite Dangerous`
- A [Windows build](https://github.com/opentrack/opentrack/releases) of Opentrack must exist in `[...]/steamapps/common/Elite Dangerous/z_opentrack-win` (unless you want to edit the script path)

&nbsp;

### Installation

1. Download the txt file, rename it to `z_OT_Proton.sh`, make it executable and then put it into `[...]/steamapps/common/Elite Dangerous`
2. Inside `z_OT_Proton.sh`, the value of the `ed_version` variable must be adapted reflect the release level of ED you wish to play, as per [the list of launcher flags](https://github.com/Rfvgyhn/min-ed-launcher#shared) or the game won't launch. "EDO" is the default for me.
3. In the *Properties* dialog for ED inside Steam, set the launch options to
    `konsole -e ./z_OT_Proton.sh %command%`
    where `konsole -e` needs to be replaced with whatever command opens a terminal on your system and executes something. The important part is that it executes the shell script inside the ED folder, followed by `%command%` to run ED.

&nbsp;

### Usage

1. When pressing *"Play"* inside Steam, you'll get a terminal window in which the following happens:
   - The minimal launcher is started
   - The script pauses until ED's main process is detected (in a 5 second interval)
   - The path to ED's Proton folder is read from ED's Protron prefix `config_info` file
   - Once ED's main process is finally running, Opentrack is being started
   - The script pauses until Opentrack's process is detected (in a 5 second interval)
   - The script terminates
2. You should have ED and Opentrack up and running now

&nbsp;

### First run

1. Set Opentrack output to Freetrack, using both the TrackIR and Freetrack protocol
2. Start tracking to check connection to your tracker and to automatically create the required registry entries in the Proton prefix
3. Stop tracking, close Opentrack, close ED
4. Restart ED from Steam. When tracking in OT and when inside the game world, there should now be headtracking input.

Good luck.

[z_OT_Proton.txt](https://github.com/opentrack/opentrack/files/6834855/z_OT_Proton.txt)

pan-mroku, 2021-07-26:
@waterlubber 
AFAIK the wrapper itself works. The problem is that it needs to be launched inside the container and there seems to be no way to do that from outside. So I think that the best way to progress would be to create a way to manually launch the wrapper (via some thin, executable in proton environment, Linux executable) inside the container with the same info OT launches it when we start tracking (that's at least my understanding, that OT passes some info to the wrapper at launch). And then somehow tell OT to connect to "externally launched wrapper" instead of launching it's own.
But I haven't had the time to take a look into OT code, so this is all based on my assumptions how it works.

@JT8D-17 
Nice! Just one nitpick - you are using `otpath="$PWD/z_opentrack-win"` and it so happens that steam enters `steamapps/common/<game_folder>` before executing it. I think more robust would be to use `otpath="$(dirname $0)/z_opentrack-win"` so the location of `z_opentrack-win` is always looked for relative to location of the script, instead of current PWD.

pan-mroku, 2021-07-26:
Yeah, headtracking is messed up in on-foot part of Odyssey. I use a keybind to toggle "zero" output from OT for on-foot stuff.