opentrack/opentrack #1264: Request: OSC output compatibility

dorem-midi, 2021-04-29:
Currently the UDP over network option is not directly compatible with OSC (open sound control).  OSC, I think, is looking for a text string (command) and number (argument) to be received over the UDP port.

I figure that it wouldn't be too hard to convert the headpose data to this format and send it to the UDP port as another output option.  But this is beyond my meager C++ skills.

Compatibility with OSC would allow audio developers to move around sound fields, making virtual spaces more realistic since OSC is a standard in virtual reality and surround audio development.

trsonic, 2021-05-09:
+1 for the OSC support!

I was asked recently if the [head tracker](https://github.com/trsonic/nvsonic-head-tracker) developed by me is capable of sending data to opentrack. The easiest way to integrate both would be by using OSC. I'm happy to discuss this further if anyone is interested on connecting both projects.

Link to the thread:
https://github.com/trsonic/nvsonic-head-tracker/issues/12

Cheers!

sthalik, 2021-05-09:
I'm not adding another RS2322-based tracker. Hatire is already a pain as it is with quirks on each platform.

dorem-midi, 2021-05-09:
Actually, I wasn't originally asking for Opentrack to receive OSC INPUT compatibility.  I was requesting Opentrack OSC OUTPUT compatibility.  This would open up a world of headtracking options to music developers because if Opentrack could send OSC compatible messages out UDP, the Opentrack could be used to control a crapload of music production softwares.

I was thinking about attempting this myself, but my coding skills are not good enough - I can freely admit it.  I think I can see where to manipulate the data, but getting the SDK and modules going on my computer is a big challenge for a newbie like me.

trsonic, 2021-05-10:
@dorem-midi You're right, sorry for capturing your thread.

I thought that adding OSC input/output might be something that @sthalik would want to consider. OSC is one of the most universal and elegant protocols for exchanging motion tracking data.

@sthalik There is no need of adding support for any specific headtracker or software. You could define your own OSC input/output data format and people like me or @dorem-midi would take it from there. Cheers.

sthalik, 2021-05-10:
> @sthalik There is no need of adding support for any specific headtracker or software. You could define your own OSC input/output data format and people like me or @dorem-midi would take it from there. Cheers.

What is it specifically that you need? Your software wants to receive raw/mapped pose data, is that correct? Is there any other data that you wish to receive? What exactly is the OSC API/ABI, got any docs?

trsonic, 2021-05-10:
@sthalik In my case I would like to send OSC data to your app. Head rotation only. Either in a form of Euler angles or quaternions.
E.g. `"/quaternion", float Qw, float Qx, float Qy, float Qz` or `"/ypr", float yaw, float pitch, float roll`

@dorem-midi probably wants your app to output the following OSC messages to a predefined IP/port. Something like this:
`"/ypr", float yaw, float pitch, float roll`; `"/xyz", float x, float y, float z`. Can't really speak for @dorem-midi.

The `"/address"` string is the so-called OSC address. It allows to nicely rout messages across different apps and their functionalities/callbacks.

Some links regarding OSC:
https://en.wikipedia.org/wiki/Open_Sound_Control
https://web.archive.org/web/20030914224904/http://cnmat.berkeley.edu/OSC/OSC-spec.html
http://cnmat.org/OpenSoundControl/

I usually use [JUCE OSC module](https://docs.juce.com/master/group__juce__osc-osc.html) for my C++ projects. I also had some success with this library:
http://www.rossbencina.com/code/oscpack

There are some Qt wrappers available but I have no experience using them.

sthalik, 2021-05-14:
Can you write some stub code that does this? I can't write something I can't test.

Receive data on a separate thread, blocking with possibility of cancellation. Has an initializer and a destructor. Settings can be stubbed. Receives data in -180->180 range and position coordinates in centimeters, all as a float array. Uses this Euler (or properly, Tait-Bryan) angle rotation order: <<https://github.com/opentrack/opentrack/blob/master/compat/euler.cpp>>.

The main thread will copy data once every 4 ms or so, and the callback for receiving it shouldn't spend too many time or CPU cycles. Ideally it should only copy the data under a lock.

dorem-midi, 2021-05-14:
I'm not an expert at this, but I can try.  Probably the entire OSC standard is too much to absorb as we don't need to have every option available.  The way I see it, the user would be given the following options:

1. IP address and UDP port (OSC receiver)
2. Up to 3 address strings that the user can input - let's call it STRING1, STRING2, for example, "/screenrotator/yaw" or "/screenrotator/pitch".  The user should be able to enter the strings because each sound software will respond to different addresses that are called out in their respective documentation.
3. Choice of parameter to output, such as FLOAT YAW, FLOAT PITCH, or FLOAT ROLL
4. Choice of output units (degrees or radians)
5. Possibly a data transmission rate, but I'm not sure if this is defined elsewhere in the code as it also relates to data smoothing (?). 

(Note, my settings example above is tailored for only the angles and not the XYZ position.  Some people may want this, but I currently don't have a need for it.  Implementation for XYZ would be similar, however.)

So when the OSC output is enabled, the data going out would have the format:

STRING1 FLOAT YAW

an example of the output would be something like:
/screenrotator/yaw -5.6379972
/screenrotator/pitch 4.3954344

Additionally, there might be some cases where the output parameters may need to be combined in the same line:

STRING1 FLOAT YAW, FLOAT PITCH, FLOAT PITCH

for example:
/ypr -5.6479972, 4.3954344, 2.53422

I hope this example helps, and sorry if I didn't answer the question well enough.  Possibly @trsonic can weigh in.

hockinsk, 2022-04-07:
+1 for adding OSC Support so it can be used with all the Music & Binaural Headtrackers set to receive yaw,pitch and roll. I've not found a single headtracking software that works, the only one I can get working is the following browser-based one, but it's very CPU intensive.
https://demos.mach1.tech/Facetracking-SpatialPlayer.html

dorem-midi, 2022-04-07:
There are some other people who have made some similar solutions since I opened this. Not sure if it helps, but, I've connected facepose app with a OSC translator as one example 
https://youtu.be/yHo0byXcMPo
And this guy did it with open source face tracking code that he has shared.
https://youtu.be/rOM5t3045Jg

Hope it helps someone, and full OSC support would be even better.

sthalik, 2022-05-06:
Do you have any sample code for it?

SDX-LV, 2022-09-27:
+1 for OSC output to support headtracking with SPARTA Binauraliser and individual HRTFs:
https://sourceforge.net/p/mesh2hrtf-tools/wiki/virtual_surround_with_headtracking_tutorial

thank you.

dorem-midi, 2022-09-27:
Don't know if it helps, but someone built a headtracker specifically for my VR piano library:

https://github.com/blah1898/the-experience-face-tracker

SDX-LV, 2022-09-27:
> Don't know if it helps, but someone built a headtracker specifically for my VR piano library, and the control is OSC:
> 
> https://github.com/blah1898/the-experience-face-tracker

Thanks! I personally have an OSC compatible headtracker right now, but it is not very reliable. OpenTrack has all the input compatibility to use alternative devices which I am very interested in. So the Webcam alternative is Awesome for people who have nothing, but OpenTrack would open possibility to use so many more trackers with even higher performance.

sthalik, 2023-03-21:
Testers needed! Drop the library into the `modules` directory.

[opentrack-proto-osc.zip](https://github.com/opentrack/opentrack/files/11031079/opentrack-proto-osc.zip)

dorem-midi, 2023-03-22:
@sthalik First of all, thanks for the work - steps in the right direction. I did as directed - opened my music app and monitored the port that OpenTrack was sending to.  The stream of data looks really good, and is exactly as you mentioned: /bridge/quat with 4 float values.

However, because OSC is a pretty open framework, every application will need a different type of string + data as mentioned in the earlier part of this thread.  The music software I am using (Reaper + IEM plugin with OSC support) is looking for a different string so the two don't work together (yet).

sthalik, 2023-03-22:
So let's say you get to choose whether to send a bundle with each value being a separate message or just a single message with ypr/quat values. For ypr you probably want radians and degrees as well. Would that exhaust all expected formats?

dorem-midi, 2023-03-22:
I think that's right.  For my application, I need a separate string for each value, yet other apps might require one string with multiple values.

sletonqu, 2023-03-27:
> Testers needed! Drop the library into the `modules` directory.
> 
> [opentrack-proto-osc.zip](https://github.com/opentrack/opentrack/files/11031156/opentrack-proto-osc.zip)
> 
> The message is `/bridge/quat` with 4 float values.

You module work.

I would like to create one with a Head pose protocol (kinect) (used in Steinberg Cubase DAW)
/head_pose UserId, X, Y, Z, pitch, yaw, roll
an get this output
/head_pose 0, 0, 0, 0, -pitch,, -yaw, -roll

ABoredBunny, 2023-05-30:
Great work! While a bit more niche, I would like if it also send XYZ over OSC. So for things like IEM Room Encoder to have body tracking. If it isn't too much extra work.

dorem-midi, 2023-05-30:
@ABoredBunny I think the issue is writing the module so that the OSC output stream is configurable to the many output formats.  The current version has all of the data there, it just needs to be reformatted, and @sletonqu used OSCII-bot to translate each OSC line to something that his software could take as input.  Problem is that OSCII-bot is a really old scripting utility and is not supported on some modern OS's.

Anyway, if your OS supports OSCII-bot, you could try it. IEM plugin compatibility is what I'm after too.

ABoredBunny, 2023-05-30:
@dorem-midi It doesn't send XYZ values, it sends headtracking as quadtrinos. I'm not worried about the headtracking values. I can make it work using reaktor, together with reapers routing ability. But XYZ would be nice.

@sthalik 
Also towards how it should be.
The ability to change the message "header" and polarity. But also the send format. Also the ability to send Yaw, Pitch, and Roll. The option to select the address per value. 
Of course the best option would probably be to code an implementation per software and just having a dropdown. But that sounds like more work. while this should work with most things that use OSC.

But that is just my opinion.
[image]

dorem-midi, 2023-05-30:
Thanks @ABoredBunny , I'd forgotten about that UI element - it's nearly the same yes.

beggehorn, 2023-11-26:
@ABoredBunny 

Thanks for this! I'm trying to find a way to connect opentrack to an app that supports headtracking data in IEM Scene Rotator format and this seems like it might be the solution I've been looking for. Unfortunately, I only own Reaktor5 and can't open your .ens file (created in Reaktor6). I've been trying to recreate your patch and am not sure i have everything correct. Here's my screenshot... I managed to create and connect all of the modules (matching all of the options in the "Function" tab for all of them) but I'm wondering if I'm missing something as I don't see the same icons in the lower right of each box (the "musical notes"). Any tips? Thx!

[image]

beggehorn, 2023-11-27:
I got this all working today. Turns out the "Router 1->M" needed to be set to 'mono' in the Properties "Function" tab. Once I checked that box, signal started flowing.

Thx!!

beggehorn, 2023-12-01:
While I do have this working, it would be amazing to be able to avoid the OSC>Reaktor translation. Would love any updates on getting this all implemented directly inside OpenTrack.

I'm still trying to wrap my head around it all and due to some existing bugs in some of the software I'm mapping to, I'm not 100% dialed in. It's been difficult for me to know what is causing some of the issues I'm having because I'm not sure if Reaktor is not configured correctly of if the bugs are affecting things. FWIW, I'm using this setup to create a headtracking solution for immersive (binauralized) audio monitoring and some of the audio signals are snapping around to different positions with small head movements (?).

ABoredBunny, 2023-12-03:
> While I do have this working, it would be amazing to be able to avoid the OSC>Reaktor translation. Would love any updates on getting this all implemented directly inside OpenTrack.
>
> I'm still trying [...]

Yeah, I forgot to mention that I set Scene rotator to yaw-pitch-roll,
not roll pitch yaw.

I do believe the OSC module needs some work. @beggehorn

ABoredBunny, 2023-12-03:
Also somewhat unrelated, but I tried to compile opentrack with aruco and OSC support on Linux, and this might have been the most conveluted thing I have tried to do. I know that Opentrack on Linux is not supported and more for devolpers. But I'm not sure every devolper is a masochist(maybe most are, I don't know).

Having the wiki instruct how to install without dependencies is kind of pointless, because they are important. Having to search through github issues to find out you need a specific version of a library, or through random internet forums. Is not a reasonable thing to ask of anyone.

Linux is getting more popular as a gaming platform, because of things like the steam deck, and valve still supporting old amd gpu's in there drivers. At least update the wiki to show how to install the dependencies, please.(although a flatpak would be even better.)

Also I'm not sure if the makefile or something is setup right because it seems to be searching for the osc library files inside the 'git' but it is missing the files.

onurbas, 2024-01-05:
OSC output of Yaw, Pitch and Roll info works great. As @dorem-midi requested, it would be great if we could also transfer OSC info for X, Y, Z axes. Is there any updates on that front?

beggehorn, 2024-02-01:
> Yeah, I forgot to mention that I set Scene rotator to yaw-pitch-roll, not roll pitch yaw.
>
> I do believe the OSC module needs some work. @beggehorn

One other issue I'm having at the moment. With my adaptation of your Reaktor patch, I'm not receiving Roll value changes in SceneRotator from Reaktor. The Roll values are jittering but do not change with a head roll movement (tilting head left or right). They do however show values +/- 4 degrees when turning my head (yaw). 

If you have a moment, could you tell me if something's wrong in in the attached Reaktor 5 ensemble? I tried my best to recreate your Reaktor6 version that I couldn't open but it's clearly not working 100%...

[Reaktor 5 - BME opentrack to scene rotator.zip](https://github.com/opentrack/opentrack/files/14129717/Reaktor.5.-.BME.opentrack.to.scene.rotator.zip)

beggehorn, 2024-02-01:
I just came across this open source OSC bridging app:

https://github.com/trsonic/nvsonic-head-tracker/releases

While I'm currently having trouble getting it to connect to my destination app (APL Virtuoso), I've had no trouble feeding it from opentrack. The nvsonic headtracker bridge app features a virtual head model as well which helps ensure the mapping is correct.

ABoredBunny, 2024-02-14:
@beggehorn I'm sorry for the delay. I compared yours, and mine and found out that you set the Router to mono, I set the ensemble to mono. This makes a difference because the inverter has some sort of issue with the mono signal. Which made it so the X axis of your tracking data didn't get send.

To fix it go into Ensemble(click the background of the node panel)
and set voices to 1. And uncheck the mono option in the router node. And the result should be the same as my ensemble(outside of my meters being number based, but that doesn't matter)

sthalik, 2024-02-16:
> OSC output of Yaw, Pitch and Roll info works great. As @dorem-midi requested, it would be great if we could also transfer OSC info for X, Y, Z axes. Is there any updates on that front?

It's easy to add XYZ data in centimeters.

dorem-midi, 2024-02-17:
Looks like I'll have to try that OSC bridge app when I get a chance...

trsonic, 2024-02-17:
Kudos to @sthalik for implementing the OSC output.

I'm working on an update of the [OSC bridge app](https://github.com/trsonic/nvsonic-head-tracker). I've altered the OSC input settings to match the default OSC output from opentrack. Latest code at the [develop](https://github.com/trsonic/nvsonic-head-tracker/tree/develop) branch. Binaries will follow.

beggehorn, 2024-02-17:
Nice work!!

Ben

Ben Eggehorn

haryo-s, 2024-05-02:
Hi, I want to look into the OSC output implementation and it's unclear to me how I need to specify the oscpack build directory in cmake. 

I've cloned oscpack and built it in a separate folder, and I've looked into [opentrack/proto-osc/CMakeLists.txt](https://github.com/opentrack/opentrack/blob/master/proto-osc/CMakeLists.txt) but it doesn't explain how the include folder should be structured

sthalik, 2024-05-03:
@haryo-s, it should be linked to the oscpack source directory.

```console
burzum opentrack-depends/oscpack/build % ls -l include
lrwxrwxrwx 1 sthalik None 13 Mar 21  2023 include -> ..
```

haryo-s, 2024-05-03:
I completely overlooked the opentrack-depends repository. 🤦
Thank you very much!

sthalik, 2025-05-01:
How are you guys doing with the current version? Did you get what you want from the bridge app or is further work on opentrack necessary?