opentrack/opentrack #1267: PS3 eye camera no longer commercially available.

cofilippo, 2021-04-30:
So far the recommendation always has been to get a PS3 eye camera. This choice is no longer a viable option, as it's no longer commercially available.

In order to secure its longevity, the project needs to select a new standard camera and include instructions on how to prepare it for use with OpenTrack.

albertolg, 2021-05-02:
Aliexpress is full of eyecams

https://www.aliexpress.com/af/ps3-eye.html?d=y&origin=n&SearchText=ps3+eye

cofilippo, 2021-05-02:
> 
> 
> Aliexpress is full of eyecams
> 
> https://www.aliexpress.com/af/ps3-eye.html?d=y&origin=n&SearchText=ps3+eye

Those are Chinese Knock-offs, not the real deal. Also Ali is not a a serious channel for recommendations, too many fraudulent stores and also import costs. A suitable model has to be available locally in western markets.

albertolg, 2021-05-03:
I use them and are working perfect

albertolg, 2021-05-04:
> 
> 
> > Aliexpress is full of eyecams
> > https://www.aliexpress.com/af/ps3-eye.html?d=y&origin=n&SearchText=ps3+eye
> 
> Those are Chinese Knock-offs, not the real deal. Also Ali is not a a serious channel for recommendations, too many fraudulent stores and also import costs. A suitable model has to be available locally through retail channels.

fraudlent stores? never meet one, if something was late or never arrived I always got a refund, so for me aliexpress is fine

vjetten, 2021-05-07:
bought one for 8.50 euro on ebay. Lets be environmentally friendly and re-use hardware! Chiense knockoffs work fine because it is not exactly rocket science to build a camera.

sthalik, 2021-05-07:
Honestly the Chinese have been getting better with their engineering. It's no longer accurate to describe their wares as poor quality knockoffs.

sthalik, 2021-08-29:
I've now added support for the MJPEG media format. 60 FPS cameras working over USB 2.0 should now work.