opentrack/opentrack #1269: Keybinding Request

sutex, 2021-05-07:
Can you please add support for the F13 to F24 keys.

OpenTrack listening for global hotkeys sent to it's window, when using Voice software to control OT, also activates action when in games.  Using modifiers still has the same affect IE:  In Star Citizen rShift + rCTRL+H  is for OT _Center Head Tracking._  This happens but it also fires off a flare the 'H' 

You may suggest just use another keys. But that is problematic as the base keyboard keys are used up, hence the modifiers.

Having OT adding support for the F13 to F24 keys will help heaps!.

sthalik, 2021-05-07:
It's already been implemented as per #1012.

sutex, 2021-05-07:
I tried editing the ini with F13 but my Voice software would not active it. Do I use the keycode my voice software says it (124)

sutex, 2021-05-07:
Just to be clear Keybinding #1012 does not work

sutex, 2021-08-31:
Trying to add Virtual keys F13 -F24  to OpenTrack
This is not  working, for me, What am I doing wrong is there a trick to adding VKeys in OpenTrack something I don't know?

Fish1956, 2021-09-19:
I can confirm, F1-F12 keys are recognised by Opentrack, 2021.1.2, but not F13-F24 keys, as original poster described. Hope this gets fixed/implemented soon.  
Great software, Many thanks.

SneakyMcSnakey, 2023-12-29:
Hello, I would like to note that this still doesn't work for me, any Function key past F12 doesn't register when trying to bind it.

sutex, 2023-12-29:
Over two years, been waiting for this to be fixed, the only reply has been **"It's already been implemented as per https://github.com/opentrack/opentrack/issues/1012."**

SneakyMcSnakey, 2023-12-29:
I was hoping that commenting might get some new attention :P

WinkelCode, 2024-01-16:
👋 Hey everyone, same thing here. I am trying to use F13 to center my view. I tried manually writing `F13` into `default.ini`, but that didn't seem to work. (Note: This is the same format as F1-F12 assignments).

sutex, 2024-01-24:
I have no idea why this is being ignored, by the Dev, in an answer or implementation. As these keys F13-F24 keys, do not work

sthalik, 2024-01-24:
Alright, at this rate you'll nag me to death.

Here's a test build. I'm going to need keycodes from F10 to F24. Open the executable by dragging it onto a Windows Console window to see the output such as:
```console
DEBUG [F:\dev\opentrack\dinput\keybinding-worker.cpp:226]: key 29
DEBUG [F:\dev\opentrack\dinput\keybinding-worker.cpp:226]: key 207
```
[opentrack-keytest.zip](https://github.com/opentrack/opentrack/files/14036203/opentrack-keytest.zip)

sutex, 2024-01-24:
DEBUG [F:\dev\opentrack\dinput\keybinding-worker.cpp:226]: key 68  - F10
DEBUG [F:\dev\opentrack\dinput\keybinding-worker.cpp:226]: key 87  - F11 
DEBUG [F:\dev\open
[image]
track\dinput\keybinding-worker.cpp:226]: key 88   -F12

I used the keyboard for F10 - F12
And tested the same keys with Voice Macro software using a voice command "Test Key" and it sent the key code to the terminal window for the F10 - F12 keys. Yet I was unable to  to get a response from the  terminal window from any of  the  F13 -F24 keys  

I then tried my LGS software that as the F13 -F24 keys , only the F10 key worked. 

I hope this helps, no evern sure if how I'm going about is the problem , but others here maybe able to help 
Thanks for seeing to this issue best of wishes , your nag :)
[image]

sutex, 2024-04-17:
I'm not sure what's happening; if my post above was of any benefit, no one has commented.

sthalik, 2024-04-21:
@sutex according to your post, nothing can be done on opentrack's side.

sutex, 2024-05-10:
So my testing method was sound?. And the issue is, what then? I had the impression you had implemented it: https://github.com/opentrack/opentrack/issues/1012. but it was no longer working.

sthalik, 2024-05-10:
It was performed correctly, but the result was that the keys aren't being emitted. I can't do anything unless they are. Maybe your keyboard software from https://github.com/opentrack/opentrack/issues/1269#issuecomment-1907910842 can disable key programmability in order to fix the issue?

sutex, 2024-05-10:
> It was performed correctly, but the result was that the keys aren't being emitted. I can't do anything unless they are. Maybe your keyboard software from [#1269 (comment)](https://github.com/opentrack/opentrack/issues/1269#issuecomment-1907910842) can disable key programmability in order to fix the issue?

Maybe others reading, will be able to do the test, as I'm not sure how to go about  _disable key programmability in order to fix the issue_  :)