opentrack/opentrack #1271: Hatire communications protocol

beeeeep, 2021-05-08:
Is there any documentation about the form of the data packages being sent from the arduino? I want to send translation data on X,Y,Z axes

GO63-samara, 2021-05-08:
For Arduino the following Hatire code works:

```
typedef struct
{
  int16_t  Begin;	  	// 2  Debut
  uint16_t Cpt ;     		// 2  Compteur trame or Code info or error
  float    YPR[3];		// 12 [Y, P, R]    gyro
  float    XYZ[3];   		// 12 [x, y, z]    Acc
  int16_t  End ;   	  	// 2  Fin
} type_hat;

type_hat hat;

void FT_Data()
{
       ++(hat.Cpt) %= 999;

	hat.YPR[0] = yaw;
	hat.YPR[1] = pitch;
	hat.YPR[2] = roll;
	
	hat.XYZ[0] = x;
	hat.XYZ[1] = y;
	hat.XYZ[2] = z;
	
       Serial.write((byte*)&hat,30);
}
```
Good luck!

GO63-samara, 2021-05-08:
I forgot after "type_hat hat;" you need to initialize synchrobytes:

type_hat hat;
hat.Begin = 0xAAAA;
hat.End	= 0x5555;

Now everything is correct.