opentrack/opentrack #1272: Flickering when using OpenTrack

ttenor12, 2021-05-09:
I have been using OpenTrack with Delanclip Fusion and a modded PS3 Camera for 3 months already and have never had this issue, used to use it with Smoothtrack and didn't have this issue either. Today, I started experiencing some kind of flickering today with OpenTrack, with both 2.3.12 and 2.3.13. Here's a link with a video showing the issue (Video is too big and can't upload the direct file): https://drive.google.com/file/d/1OFoBmh1SDSXc_9Iduu_-_SxOl-7bb8Ey/view?usp=sharing

As you can see, I'm using Accela as the filter. If I switch filters, I get different results. They all flicker but not as much as Accela, started using Hamilton but the screen jitters a lot, so I set the deadzone to max, that's the only thing stopping the jittering but my head basically turns at 15 fps, doesn't feel smooth at all.

At first, I thought I was having hardware issues (Maybe the IR lights of my Delanclip may have been flickering?) but confirmed they are working fine, also, made sure it's not the camera I'm using either as the exact same issue happens when I use Smoothtrack from the phone.

I'm also thinking it is not the Opentrack software itself but maybe the plugins? As the exact same thing happens with FaceTrackNoir and I guess they use the same plugins. I have tried uninstalling OpenTrack more than once, but issue doesn't seem to be resolved and I'm not sure if by uninstalling every single headtracking software I have installed also removes the plugins. If not, how can I remove them so I can try to reinstall them?

GO63-samara, 2021-05-09:
Maybe there was another Windows update today?
If you turn off all filters, the problem disappears?
Show the Mapping properties settings. Perhaps the reason is in them.

ttenor12, 2021-05-09:
Hello,

Thank you for your response. It definitely seemed to be a software issue as the Delanclip Fusion was working properly, did a system reboot and the issue was resolved.

Regards,

-------- Mensaje original --------

Fecha: 9/5/2021 9:50 a. m. (GMT-06:00)
A: opentrack/opentrack ***@***.***>
CC: ttenor12 ***@***.***>, Author ***@***.***>

ttenor12, 2021-05-09:
Hey, yeah it looks like the windows update definitely did something because I did a system reboot and the issue was fixed. Oh Microsoft, this is the second time it messes with headtracking solutions. Anyway, thank you so much for your response.