opentrack/opentrack #1274: Euro Truck Simulator 2 bug

Shadow0013HUN, 2021-05-15:
Hi there,

I have an issue with this software. During my ETS2 gameplay, I turn my head to any direction, but the software is seemingly going to sleep, and doesn't calculate any movement at all, meaning game data values don't change at all regardless if I move or not, however before I launch the game, head tracking works fairly well. 
Then after a certain amount of time (anywhere between 5 seconds to half a minute), it comes back to life, and does it's job again, and this cycle just keeps repeating.

Also, I was not able to figure out, how to make my head go outside of the truck, (you know, when you turn the ingame camera past 90 degrees to the left, it leans out of the window) so I could actually reverse into a parking bay. I tried mapping, where after a certain point, it maxes out, but it was still very low turning angle.
So, is there a way to multiply every number by a certain lambda value, without creating deadzones? Because the other annoying thing is, that instead of you scaling everything up, all you do is create deadzones, where big jumps occur, and I'm not sure how to manipulate those values to prevent this effect.

CPU: i7 4720HQ Boosted up to 3.3GHz
RAM: 8 GB
Camera: Trust GXT 1160 Vero Streaming (1080p 30fps)
I have the latest OpenTrack and AITrack installed, with the settings recommended by the video that was uploaded here.

Looking forward to your answers.

sthalik, 2021-05-15:
> I turn my head to any direction, but the software is seemingly going to sleep, and doesn't calculate any movement at all

Run the game in a window, then monitor CPU usage, the main app window and aitrack. Also disable Windows 10 game mode.

Shadow0013HUN, 2021-05-16:
My CPU usage is between 40-60%, so its hardly under strain.
The moment I put any other window (doesnt have to be exactly ETS2, Chrome does this too) as active window, it either stops calculating, or does it very laggily. Raw data changes correctly in opentrack, game data stays the same.
And I have turned off game mode.