opentrack/opentrack #1280: Only run as service- Minimized

oykma, 2021-05-26:
Hi, and thanks for your time.

I downloaded Opentrack for Windows, and app on phone, every thing Works, but then i marked in Minimize when running, and now i cant get it running and get to setup etc.

It runs in Service, but wont open.

Try to uninstall and install again but  it wont open, only run minimized.

Any file i can delete og rename or any thing/file , that remember that setting ”minimized”, so i can get back to this normal  setup

sthalik, 2021-05-27:
The only related option I can think about is "start in tray". But even that doesn't behave the way you describe it.

GO63-samara, 2021-05-27:
Maybe you need to click on "show hidden icons" in the tray?