opentrack/opentrack #1281: Nintendo RVL-CNT-01-TR (Wii remote) not registering in OpenTrack

M-Chimiste, 2021-05-28:
Hello,

I recently tried to use the wii remote camera in order to do 3 point tracking.  I've been able to connect the remote via bluetooth and can confirm it's working with the dolphin emulator.  Though when I try to use opentrack I just see an image of a wii remote and I'm not able to get the camera to register.  I'm not certain what I can do to trouble shoot this.  Any help is appreciated and I'm certainly willing to help write a tutorial for this since this seems to not be an incredibly straightforward effort.

Example of what I'm currently seeing:

[image]

DavidThijs, 2022-06-05:
I also have issues making it detect the wii contoller although it shows up in W10 x64 bluetooth connected devices. I do can use the wiimote as a mouse pointer, so at least it connects with the accelerometer. Not sure if the ir cam is properly exposed. Btw, has anyone a copy of freetrack v2.2 still laying around. All mirrors are dead and the main site only shows v2.1 for download.

Colibrisim, 2023-01-22:
Hi everybody, I have the same issue. 

> Hello,
>
> I recently tried to use the wii remote camera in order to do 3 point tracking. I've been able to connect the remote via bluetooth and can confirm it's working with the dolphin emulator. [...]

Hi M-Chimiste, I have the same problem. Didi you find a solution ?