opentrack/opentrack #1287: Opentrack doesn't get video from Droidcam (but did before)

zartko, 2021-06-09:
Something odd is happening - I'm using Droidcam to get 720p video from my smartphone and a few days ago everything was working great (using aruco -- paper marker tracker, but no other input module can get video either).

Now, I can still see the camera sources as before (DroidCam Source 2 and DroidCam Source 3) but once I start tracking the preview windows is just grey, no content whatsoever.

Nothing I can tell has changed on the computer in these few days.
Processor is AMD Ryzen Threadripper 1950X and OS is Windows 10 pro build 19042.1023

I've tried turning of Windows Defender and running both DroidCam and Opentrack as administrator.

Tried uninstalling and reinstalling opentrack 2.3.13 and tried using an older (2.3.12) version in its portable form, no luck.
In the event manager, I noticed with the portable version that opentrack-tracker-wii.dll was sometimes causing faults, deleted it, again nothing.

Event manager logs no errors I can link to opentrack when running the non-portable 2.3.13 version.

I've verified Droidcam works with other apps, including FacetrackNoir and both sources work (DroidCam Source 2 and DroidCam Source 3).

zartko, 2021-06-10:
Using ManyCam does fix this but somethings still seems wrong - for one, ManyCam seems to introduce latency and stutter and overall everything is unstable. ManyCam had to be reinstalled after failing to launch after an original working test and OpenTrack takes a long time to open and then again a fairly long time to start the tracker. I'm using this with Falcon BMS and while trying this out it crashed a few times which I have no proof for but suspect is due to the tracking glitchines.
CPU usage of both programs is overall very low in all cases, hovering at around 1 to 2%.

Webcamoid didn't fix things, in fact in the end it wouldn't even start any more.

For now I've put the Aruco route on hold and am going to use TrackAR on Android and send it via UDP to OpenTrack and this seems to work as well as that method can and I notice no weird issues (meaning OpenTrack itself seems to be still working great, it's just the video part that seems to be problematic and Falcon BMS stayed stable during testing the Android TrackAR method).

GO63-samara, 2021-06-10:
Suddenly appearing problems with video trackers are most often associated with the latest Windows updates. Wait for the next Windows update and the problem will most likely go away by itself.

zartko, 2021-06-10:
> Suddenly appearing problems with video trackers are most often associated with the latest Windows updates. Wait for the next Windows update and the problem will most likely go away by itself.

Ok, I'll give it a shot after a new round of updates.

ragedoog, 2022-05-24:
I've got the same issue but instead of using aruco paper tracker I was using UDP over network with droid cam and aitrack and everything worked perfectly well for 2 days and now I can't get it to work. Everything was exactly the same as it was when I was using it for the first time so I don't have any clue of what happened...

please let me know if any of you solved the problem

YosypC, 2025-03-08:
I have the same problem. Windows Camera detects it fine, while Opentrack doesn't. 2024.1.1

bukowa, 2026-10-14:
Hey any updates?

vido89, 2026-10-15:
@bukowa 
Create new stream with srcrcpy

./scrcpy --video-source=camera --camera-size=1280x720 --camera-fps=30 --v4l2-sink=/dev/video10 --no-playback --no-audio