opentrack/opentrack #1293: when looking down, the octopus rises upwards

weczi, 2021-07-03:
when looking down, the octopus rises upwards
if i want to see "my feet" i am going up

weczi, 2021-07-03:
version 2.3.13

Slion, 2021-07-05:
You can invert the axis in your profile I think.

weczi, 2021-07-05:
> You can invert the axis in your profile I think.

and what it could give me?
i think you dont understand

GO63-samara, 2021-07-05:
In OT-2.3.12, the Octopus showed the Y-axis inverted.
In OT-2.3.13, the Octopus shows the Y-axis correctly, in accordance with the position of the pilot's head in the flight simulator. The Octopus display does not affect the Opentrack output coordinates.
@Slion correctly said that if the work of the Pitch or Y axes does not match the head, then you need to invert them in Options-Output.
It's not even a problem.

weczi, 2021-07-06:
> In OT-2.3.12, the Octopus showed the Y-axis inverted.
> In OT-2.3.13, the Octopus shows the Y-axis correctly, in accordance with the position of the pilot's head in the flight simulator. The Octopus [...]

sorry but it not a solution, problem is different:
when I invert Y-axis then: looking down (**pitch down**), the octopus rises looking down and crouch (**Y axis**) so its wrong Y-axis should move at all!

GO63-samara, 2021-07-06:
Nothing is clear from your description.
Show screenshots of your **Options-Output** and **Options-Relative translation** settings.
It is also helpful to show a video of your problem.

> Y-axis shouldn't move at all

> pitch down should move Y-axis

One task contradicts another, right?

weczi, 2021-07-07:
YES!!!
When I want pitch down i ALSO move up (Y-axis) - together.

Options-Output  are blank - standard like in 2.3.12 and older versions.
What is a difference when I invert Y? why Y even move? I dont want go up or down (Y-axis) I want to pitch down (look my feet).
If I invert Y-axis I will go down - not up - so still the problem with moving Y-axis.
From begining I am wrote about problem with PITCH. Pitching should move Y-axis at all!
Its a different move - different axis. Y should move at all when I want pitch - what you dont understand?

Pitch its a different move
[https://photos.app.goo.gl/TN8nL38keLQS3P4h6](url)

GO63-samara, 2021-07-07:
We still don't know which tracker you have.
We still haven't seen your Opentrack settings.
It is impossible to help.

weczi, 2021-07-07:
i have my own tracker IRTrack 
[www.irtrack.pl](url)
thats my profile
[DCS vPRO2.txt](https://github.com/opentrack/opentrack/files/6780106/DCS.vPRO2.txt)

GO63-samara, 2021-07-07:
Now I can see that you have PointTracker. Perhaps his settings give the effect of Pitch on the Y-axis.
Check the input to see if the **Y**-axis changes as you rotate to **Pitch** in the `Raw tracker data` panel.
If changes, you need to calibrate the offset in `PointTracker Settings`-`Model position`-`Start calibration`.

The rest of the Opentrack settings are normal. Maybe you still need to try disabling `Relative translation`-`Disabled`.

weczi, 2021-07-07:
Thank you - and that is a big help! thank you
I will check the calibration - hope this help. 
I will let you know.

weczi, 2021-07-08:
Thank you very much! it was about calibration.
Now everything working fine! Thx!

closed 2021-07-08 by weczi