opentrack/opentrack #1294: Windows 11

Len62, 2021-07-07:
@sthalik  If anyone is interested, I was able to get OT working in Windows 11. Just had to install the MSVCP140 library.  Using the cleye driver with a PS3 camera. Haven't figured out how to use the included PS3 driver though.

sthalik, 2021-07-07:
> Haven't figured out how to use the included PS3 open driver though.

Use Zadig software to install the libusbK driver onto the camera's `interface 0`, after uninstalling the CL-Eye driver.

I've personally had a problem where it won't work when plugged into a USB 3.0 controller, only a USB 2.0 one. In case of problems, connect it to a black port and not a blue one.

GO63-samara, 2021-07-07:
In addition to **msvcp140**, **vcruntime140** is also often needed.
@sthalik It would be useful to include these dlls in the next Opentrack releases, wouldn't it?

VivienHardgas, 2021-07-12:
if im allowed here i would like to mention another problem.

opentrack causes my windows 11 system to freeze while using it and the envent viwer says this 
[image]

GO63-samara, 2021-07-12:
If you have **OT-2.3.12** just delete `opentrack-tracker-wii.dll` from `modules` folder.
In **OT-2.3.13**, **WiiPointTracker** boots without error (on Windows 10).

VivienHardgas, 2021-07-13:
i updated the version to the newest build its more stable on win 11 and dont causes my system to freeze anymore? the eventviwer now dont make trouble with OT

bastianb051, 2023-06-21:
Hello,
I have the following problem: 
when I press start in open track, the program crashes. 
Here is the log from cmd: 
DEBUG [F:\dev\opentrack\api/plugin-support.hpp:106]: library "kinect-face" failed: "Cannot load library C:\\Program Files\\opentrack\\modules\\\\opentrack-tracker-kinect-face.dll: Das angegebene Modul wurde nicht gefunden."
[ INFO:0@2.334] global registry_parallel.impl.hpp:96 cv::parallel::ParallelBackendRegistry::ParallelBackendRegistry core(parallel): Enabled backends(0, sorted by priority): N/A
[ INFO:0@2.344] global videoio_registry.cpp:232 cv::`anonymous-namespace'::VideoBackendRegistry::VideoBackendRegistry VIDEOIO: Enabled backends(3, sorted by priority): DSHOW(1000); CV_IMAGES(990); CV_MJPEG(980)

can you help me?