opentrack/opentrack #1296: Tracking issue with FreePIE UDP in OT version 2.3.11; 2.3.12 & 2.3.13

CatWithLightSabers, 2021-07-09:
I use OT + Android phone (Samsung Galaxy S2 & S4) for years. I use the FreePIE UDP Input protocol with FreePIE IMU on the phone. Since version 2.3.11 there is an issue with the "Game data" axis values. When I roll the phone and see the Raw tracker data change the Game data pitch AND roll change. Same with pitch. I see the Raw tracker data pitch changing but the Game Data pitch AND roll change. Yaw is unaffected.

This means when I look down for example my view goes down but also rolls. When I roll my head to the right my view rolls to the right but also looks up. It  is like roll and pitch axis are combined.

I rolled back version by version and I can confirm the issue does not exist in 2.3.10 and below. There, when I pitch my head the ingame view pitches accordingly. When I roll my head the ingame view rolls correctly as well.

GO63-samara, 2021-07-09:
Hey!
This is a really interesting and useful question.

In **OT-2.3.10** and earlier, centering simply subtracted the center coordinates from the current coordinates. And this caused a bug in IMU trackers, which manifested itself as a jump in the Yaw view when the input Yaw crossed the +/-180 degrees border.
In **OT-2.3.11** and the following versions, improved centering is applied. Centering has been improved just for IMU trackers. When you press the centering, the current orientation of the tracker is memorized and from that moment it is considered the base one.

Your problem is that the **tracker is tilted to your head**.

Possible solutions to your problem:

- Mount the tracker straight on the head, so that at the center position of the head, the Roll coordinate on the `Row tracker data` panel is equal or close to zero. This is the **best solution**.
- Center in your tracker software and no longer center in Opentrack. Set `Center at startup` - `OFF`.
- Stop using Roll in the simulator. To do this, set `Options`-`Output`-`Roll`-`Disabled`. Hmm, this needs to be checked, it may not work.

CatWithLightSabers, 2021-07-18:
Hi GO63-samara,

 

I tried another app ("Head Tracker" from the playstore) that lets me calibrate the tracker on my head instead of calibrating with Opentrack on my PC and you are right this does not produce the issue. However it uses the "UDP over network" input instead the "FreePIE UDP receiver" input I used before.

So I got this working with OT-2.3.13 however it is not as smooth as my previous solution with OT-2.3.10 with FreePIE IMU. I have tried multiple head tracking systems and apps like 3-point tracker clips and also on a baseball hat, I tried facetrackNOIR and facetracking through androidcam.

OT-2.3.10 with FreePIE IMU is almost flawless for me. Even though it provides only 3 axis, I usually don't use the Z and Y axis anyway and I combine the X axis with the roll axis. This way my view translates and rotates left and right when I roll my head which feels very naturally. I get 4 axis out of a 3 axis system.

 

I stick around with the old version for now but it would be great if it would be supported in the future. It would be a shame if that solution would become impossible due to lack of software/update support. I use an old Samsung Galaxy S2 as the head tracker. It works wirelessly, has its own power supply, there is no fumbling around with light sources/brightness etc. like with LED based systems and there is also no problem with LEDs being blocked/out of view for a camera.

 

I noticed the bug you mentioned where the Yaw view jumps crossing the 180 degrees border and I got around that by simply using the "add +180 degree" to the yaw axis option in the FreePIE UDP receiver input.

 

Cheers,

Michael

 

Gesendet: Samstag, 10. Juli 2021 um 04:47 Uhr

GO63-samara, 2021-07-18:
> OT-2.3.10 with FreePIE IMU is almost flawless for me. Even though it provides only 3 axis, I usually don't use the Z and Y axis anyway and I combine the X axis with the roll axis. This way my view [...]

DIY IMU tracker can even provide 5DOF:
https://sites.google.com/site/diyheadtracking/home/razor-ahrs-5dof-20191019-samodelnyj-inercionnyj-5dof-treker

GO63-samara, 2021-07-18:
> I stick around with the old version for now but it would be great if it would be supported in the future. It would be a shame if that solution would become impossible due to lack of software/update support.

@CatWithLightSabers and @sthalik.

It seems like it makes sense to introduce a choice of several centering methods in Opentrack.
1. Subtraction of the center point (as in OT-2.3.10). Convenient with PointTracker. Not suitable for 360 degree wireless VR.
2. Centering matrix (as in OT-2.3.11 and later). It is perfect for wireless VR 360 degrees. The only centering option without an input gimbal lock.
3. For IMU trackers and video trackers, a combined option would be convenient. First, the Yaw and Pitch centering matrix is calculated, assuming Roll = 0. The Yaw and Pitch axes are centered. The current Roll coordinate is then considered the center one and subtracted from the input Roll values, as in the first option. With this method, there will also be a gimbal lock at the input (when |Pitch| = 90). But the phone or the clip can be attached to the head tilted, not strictly vertically.

For @CatWithLightSabers, the third option would work.

sthalik, 2021-07-19:
@GO63-samara these options look like the best thing that's been suggested so far. Do you feel like implementing it? I don't want to steal your idea in that case.

GO63-samara, 2021-07-19:
Well, I'll try to do this. But it will take some time.

GO63-samara, 2021-07-24:
I started to solve this problem.

So far I managed to make a centering model in Delphi, with several options:
1. Corrected the centering of the OT-2.3.10 version by gluing the borders +/- 180 degrees. This centering option is now suitable in most cases. But this variation has behavior similar to "gimbal lock" near |Pitch| = 90.
2. Left the centering option from OT-2.3.11, but I converted it to quaternion form. In this form, less memory is needed, and the performance is even higher. This option is indispensable for wireless VR 360 degrees. But precise orientation of the sensor on the head is required. There is no "gimbal lock".
3. Simulated a Roll compensation option where the IMU sensor is tilted. It works even without the "gimbal lock". This option will be especially useful for owners of IMU trackers and a laptop when the screen is installed below (or above) eye level.

- Tested the option with tilted sensor compensation according to Roll and Pitch, but it turned out that it works exactly the same as option 1, but in implementation it is much more complicated, so I abandoned it.
- I think the most popular option will still be centering option 1, as in OT-2.3.10, with corrected rotations across the border +/- 180 degrees.
- I think it makes sense to add option "0" - `no centering`, for comparison with other options.

On nearest Monday will try to start transferring this work to Opentrack.

GO63-samara, 2021-07-30:
I managed to do everything I wanted: https://github.com/GO63-samara/opentrack/commit/723b98296a5636f561e3903560da9f1aa1797b82
I entered the "Centering method" combobox next to the "Center at start up" selection

Opentrack now uses several centering methods.
Select the centering method from the drop-down list: `[Options]-[Shortcuts]-"Centering method"`.

- **Disabled**: Centering is disabled.
- **Point**: Point centering, the same as in OT-2.3.10, but without jumps when crossing the border +/- 180 degrees. Well suited for video trackers, although near the input |Pitch| = 90 it has strange behavior similar to "gimbal lock".
- **Wireless VR 360**: Centering using a centering matrix or quaternion, the same as in OT-2.3.11...OT-2.3.13. Ideal for wireless VR headset for 360 degree viewing, no "gimbal lock". But it requires precise installation of the sensor or clip on the head so that in the central position the Roll angle on the "Raw tracker data" panel is close to zero.
- **Roll compensated VR 360**: It is a development of the previous version. It does not require precise installation of the sensor on the head. The sensor or clip can be tilted along the Roll, even at a 45 degree angle. Unlike the first option, there is no "gimbal lock". This centering option is suitable in most cases, especially for owners of an IMU tracker and laptop, when the screen is installed below (or above) eye level.
Do not forget to click [OK] to save the centering method selection to the profile file.

Hint: Now it is possible to reset the centering without stopping the tracking. To do this, while tracking, switch to [Disabled], then press [Center]. This will reset the centering for all centering options.

Is it time to make a Pull request? or will there be any comments?

sthalik, 2021-07-31:
The code looks good, and any comments I've made are purely stylistic. You know your math so you can merge it directly if you want.

GO63-samara, 2021-07-31:
@CatWithLightSabers
I understand that our conversations with @sthalik did not make you feel better.
Therefore, I suggest you and everyone else try this **OT-2-3-13-FixCentering.rar** fix:
https://sites.google.com/site/diyheadtracking/home/opentrack/opentrack-download/OT-2-3-13-FixCentering.rar?attredirects=0&d=1

[image]

You only need to replace three files in **OT-2.3.13** with the files of the same name from this archive.

Good luck!

CatWithLightSabers, 2021-08-01:
Hi guys,

 

I just tried it and it works flawlessly! Thanks for all your efforts, it is very much appreciated!

 

Cheers,

Michael

 

Gesendet: Sonntag, 01. August 2021 um 04:07 Uhr

 

https://sites.google.com/site/diyheadtracking/home/opentrack/opentrack-download/OT-2-3-13-FixCentering.rar?attredirects=0&d=1