opentrack/opentrack #1298: Default Capture Pin settings being reverted to 1080p30fps on ELP camera

Sephirotic, 2021-07-16:
I suppose this is more of a Point Tracker 1.1 plugin issue than OpenTrack itself, but I imagine this could be a relatively easy fix. 

I have got a very nice neat camera from Aliexpress which is a OV2710 sensor with CS mount lens with MJPEG 1080p30fps 720p60fps and even 640x480p120fps modes available. Manual exposure, removeable IR filter and very low latency. It would be awesome tracking camera, except its driver does not save the last capture pin mode you used and always revert back to 1080p30fps mode. When I set up the mode I wish to use on PointTracker 1.1 and click "Start" on open track, the actual mode is always 1080p@30fps, and I notice the resolution is just a middle crop of the image not an actual capture pin change. 

Is there any way to enforce a capture pin mode when clicking "start" on open track? Thanks a lot.

GO63-samara, 2021-07-18:
Could it be related to the **MJPEG** format of the camera's output?
Here's something similar: https://github.com/opentrack/opentrack/issues/1244

Sephirotic, 2021-07-19:
> Could it be related to the **MJPEG** format of the camera's output?
> Here's something similar: #1244

Thanks for the reply. I commented in that ticket, but it seems the OP there misdjudged the real reason the camera does not work in 60 or 120fps. It is not an issue about MJPG or YUY2, because you can set up 1280x720p as resolution and still get 30fps (because it is just a crop from 1920x1080), if it was truly changing to yuy2 input mode, than you wouldn't get more than 6fps.

N7-Hash, 2021-07-19:
Interesting find, and thanks for the detailed explanation. I hope this problem is resolved.

Sephirotic, 2021-07-19:
> Interesting find, and thanks for the detailed explanation. I hope this problem is resolved.

I've found a workaround. 

Apparently my initial assumption was also wrong. The input settings into the camera settings tab from point tracker 1.1 was toggling incorrect modes into the camera pin driver. It had nothing to do with reverting default settings like I initially thought. 

Whenever I put 640x480 @ 60 or 120 fps (which are actual modes in the camera pin), the "status" info when initialized the point tracker was always 30 fps. It didn't respect the FPS I had chosen. The same Happened for all other actual resolutions: 1280x720@60fps, or  800x600@60 fps. I was getting 10 for 720p and 20 fps for 800x600 (which indicated it actually was working in YUY2 mode like the other user suspected). However, when I changed the resolution to an uneven number close to the actual resolution I wanted, for example, 1200x720, @ 60fps, then the camera pin switched to the correct mode: 1280x720p@60fps (and MJPEG). 
I tried 640x480 @120hz, and I got around 75~81fps (oscillating around this. It seems Point Tracker can't process 120fps even with the high single core boost of a 5900x).  The result is awesome in DCS, the quality of the camera is much superior than the PS3eye and having better Dynamic Range also meant I could make the exposure darker so that it could isolate the IR leds even with the window open in a sunny day (with sun rays being blocked by a curtain, of course). 
I have made two videos comparing the PS3 eye camera with the ELP OV2710 Hd Usb camera with CS Mount here:

https://youtu.be/ELABP2SgUyw
https://youtu.be/RNwth7gtsc4

The second video tests input lag with a 240fps slow motion capture from my smartphone. Actual lag is about 13 frames (or 50ms) for the ELP camera in both 640x480@120hz mode or 1280x720@60, and 17 frames for the PS3 eye camera (70ms, 20 more than the ELP HD camera.)