opentrack/opentrack #1299: Smooth transition when Centering

StarSpaceCommander, 2021-07-21:
More of a request than issue.

Would it be possible to include an option/function which enables a smooth transition when you center or stop tracking? So that it doesn't have this artificial snap into center, but slows a bit down smoothly toward center to look more like natural head movement.

Thank you for the great software!

StarSpaceCommander, 2021-07-22:
While at that. An option, where you define a small cone around the center, within which the view slowly centers itself, would be greatly appreciated. :)

GO63-samara, 2021-07-23:
> or stop tracking

Do you mean when "custom center pose" is used?
I would prefer that in the "stop" mode the view is not at zero, but at the "custom center pose" point, then smoothing is not required.

But when centering, smoothing makes sense so as not to lose orientation in battle for a while.
I like it request!

StarSpaceCommander, 2021-07-23:
It should work with custom center pose, too, but what I specifically mean is that if you re-center, or toggle off the tracking and the view snaps back to centered forward view, the view that is snapping back to center should kind of slow down at the very end a bit. That way it would not look so linear and robotic, but a bit more natural.

Imagine you look to the side and then turn your head back forward, the view doesn't snap to center, it moves quickly, but naturally to center, because a human doesn't snap his head like a robot/turret, but moves it naturally non-linear. That is the effect that would make the re-centering feature look and feel better.

If my explanation isn't clear, I can show a video clip as a demonstration.

GO63-samara, 2021-07-24:
> If my explanation isn't clear, I can show a video clip as a demonstration.

Of course, we want to see the video. In a well-shown problem, half of the solution will already be held!

sthalik, 2021-07-28:
> I would prefer that in the "stop" mode the view is not at zero, but at the "custom center pose" point, then smoothing is not required.

With the "zero" keybinding, it's meant to display itself at origin plus the center pose offset. With the "stop" keybinding, it's meant to display itself at last pose plus offset. Isn't this the case?

GO63-samara, 2021-07-28:
When the [Zero] button is pressed, everything works correctly, `Game data` == `Custom center pose`.
When I press the [Stop] button, all coordinates are equal to zero on the `Game data` panel, but should be equal to the `Custom center pose`.
Or shouldn't they? What is the difference between [Zero] and [Stop] buttons?

> it's meant to display itself at origin plus the center pose offset

What does at `origin` mean? Is it all zeros?

sthalik, 2021-07-29:
So in <https://github.com/opentrack/opentrack/blob/master/logic/pipeline.cpp#L468>:

```c++
    if (hold_ordered) value = output_pose;
    else value = apply_zero_pos(value);
```

What happens if you change it to,

```c++
    if (hold_ordered)
        value = output_pose;
    value = apply_zero_pos(value);
```

sthalik, 2021-07-29:
So in \<https://github.com/opentrack/opentrack/blob/master/logic/pipeline.cpp#L445>:
This can be removed, but then:

Change

```c++
    if (hold_ordered) value = output_pose;
    else value = apply_zero_pos(value);
```

```c++
    output_pose = value;
    auto value_ = apply_zero_pos(value);
```

As `hold_ordered` is already reset to `output_pose`. The problem is saving the custom offset in `output_pose` in the first place. Then:

```c++
    ev.run_events(EV::ev_finished, value);
    libs.pProtocol->pose(value, raw);

    QMutexLocker foo(&mtx);
    output_pose = value;
    raw_6dof = raw;
```

Replace

```c++
    ev.run_events(EV::ev_finished, value_);
    libs.pProtocol->pose(value_, raw);

    QMutexLocker foo(&mtx);
    // REMOVE THIS
    //output_pose = value;
    raw_6dof = raw;
```

GO63-samara, 2021-07-29:
Well, you've almost solved the problem. Why don't you add this fix?
I now have to finalize the centering fix.

sthalik, 2021-08-18:
Done.

StarSpaceCommander, 2021-10-17:
Glad you resolved a problem.

Back to my original post. I looked into it again and noticed it is not so much a smoothness thing, but if you use head tracking and move your head to the center, you do it with a natural speed.

I request the ability to simply be able to adjust the speed in the software at which the view re-centers. So that it doesn't snap to center unnaturally, but looks as if the user is simply moving the head to the center. I'd like it to look and feel natural and in accordance with how the user looks around.

sthalik, 2021-10-18:
That's not particularly useful, as the typical user wants to know the precise center position immediately after depressing the button. The typical user also doesn't want to spend time unnecessarily while in the heat of the battle.

IOW, you'll get used to it.

DarinDev1000, 2025-02-23:
I also would like an option to change the speed at which the camera recenters when pressing zero