opentrack/opentrack #1308: Axes coupling solution with mapping

cro888888, 2021-08-06:
Can I in any way solve coupling problem(in my case:pitch/yaw coupling when pitch approach 90deg up) with mapping? Like map pitch to yaw +/-ndeg to original yaw to yaw curve?

GO63-samara, 2021-08-07:
Hey!
Your question is not clear.

1. What does `solve coupling problem` mean? Does this mean `gimbal lock`?
2. Is the problem showing up in the Opentrack output coordinates or in some game?
3. What tracker is used?
4. Which filter?

dssrtfx, 2021-09-03:
Do you mean since your tracker may not be 100% parallel with your face's vertical plane, it shows yaw more obviously near pitch limits?

sthalik, 2021-09-03:
See if the new centering modes added by @GO63-samara help with your problem.