opentrack/opentrack #1312: After using OpenTrack, hat switch stops working in DCS and X plane

Jetsy85, 2021-08-13:
Hi, I am using OpenTrack in DCS and X Plane 11. In both games, the moment I used OpenTrack for the first time, I lost the ability to change views or look around using joystick hat or keyboard commands! This issue is present whether OpenTrack is open, running or not. All keyboard key bindings are still assigned in both games, however they have had zero functionality during game play since installing OpenTrack. Is there a way of restoring this lost functionality in my games?

GO63-samara, 2021-08-13:
The problem is that you are using the `freetrack 2.0 Enhanced` output protocol. When you clicked `Locate DLL` and selected `NPClient.dll`, that selection was remembered in Windows. Now, if you load the game, even if Opentrack is not running, `NPClient.dll` will also load and will not allow you to use the buttons and joystick for change views.
To unlink the game from Opentrack, rename the `Opentrack` folder or the `NPClient.dll` library. This barbaric way works, but I don't know any other way.
One more conclusion - to avoid this problem, use a different output protocol, for example `mouse emulation`. This works in many games.

Jetsy85, 2021-08-13:
Thanks so much!! I will give it a try!

sthalik, 2021-08-19:
Just use a batch file. Execute:

```
  reg delete HKCU\SOFTWARE\NaturalPoint\NATURALPOINT\NPClient Location\Path /f
```

GO63-samara, 2021-08-20:
It would be nice to add a `[Forget dll]` button next to the `[Locate dll]` button which does exactly that. I think so.

sthalik, 2021-08-21:
It's only useful for users while they evaluate whether they want to use opentrack in their sim setup. People graduate to either using opentrack fulltime, or not using it at all.

I should add a mode that copies the .dll to another directory, for Elite: Dangerous.

Is there anything in particular to be done before releasing `2021.1.0`?

GO63-samara, 2021-08-21:
> It's only useful for users while they evaluate whether they want to use opentrack in their sim setup. People graduate to either using opentrack fulltime, or not using it at all.

It is also useful for testing different Opentrack output protocols. Otherwise, NPclient.dll may interfere with proper operation.

> Is there anything in particular to be done before releasing 2021.1.0?

I like the name of version `2021.1.0`.
In my opinion, at least a short Readme about the new `Neuralnet tracker` is missing, about its configuration.
I was going to add translation of a few new lines to Options and change flags when changing localization. I'm not going to do this until Monday. I also don't like the phrase `"Never translate the application interface"`. "Never" hints at some kind of irreversibility of the action. I translated it into Russian as `"Использовать английский интерфейс"` `(Use the English interface)`. It seems more logical to me. What do you think about this?
In general, I have no urgent comments.