opentrack/opentrack #1315: semantic versioning woes

sthalik, 2021-08-18:
As you know, I'm awful at using semantic versioning. opentrack should've been bumped to 2.4.0 ages ago. I'm thinking of changing x.y.z(.patchlevel) to year.minor.patchlevel. If you have anything against that, please speak now.

sthalik, 2021-08-18:
Paging @GO63-samara @DaMichel @Slion.

dbaarda, 2021-08-18:
My 2c; semantic versioning is very important for libraries and API's since they tell you important things about compatibility of the interface;

1. if the major number has changed, the library is incompatible in breaking ways, and you probably can't even link against it, but even if you could your program will probably crash. Programs should specify they require a particular major number.
2. If the minor number is incremented, additional features have been added, but it's still backwards compatible and will work for you. Programs should specify they require a minor number >= some value which includes all the features used by your program.
3. If the patch number is incremented, some non-API changing fixes have been applied. It's largely irrelevant for programs except that they should always use the largest patch number (most bug fixes) for a given minor version.

These rules are usually used by putting the major number on the end of the built library so programs dynamically link against the right major version of the library, and then have the programs check the minor version number is large enough.

So you should not abandon semantic versioning if you have an API and particularly a library other people are using. If you don't then use whatever you like... yy.mm.dd-rev would be fine. Sometimes people use semantic versioning for their API's interface, and some other version numbering system for the rest of their program, giving you things like "v2021.06.25-1 provides API v2.4.9", but that can get messy if you are not treating your library/API like a separate mini-project.

I believe Debian packagers have some tools to check library changes to help enforce semantic versioning.... I know the Debian packager of librsync requested we constrain our library exports to the documented interface so that their tooling would be happy if we changed internal functions without bumping the semantic versioning.