opentrack/opentrack #1323: Sony Camera driver instructions

Iknowlongnamerightwhatwasithinking, 2021-08-31:
Can you please explain how to install the new camera driver. In steps. 
   
"Use the zadig program <https://zadig.akeo.ie/> to install the libusb0 driver onto interface 0 of the camera. Don't use the WinUSB driver!"
   
What is "libusb0" driver ?   

Where is the Camera driver? 

Thanks!

S!

sthalik, 2021-08-31:
Run the zadig app, you're gonna get a selection of drivers. Select libusb0. libusbK may or may not work and WinUSB will straight out give awful performance.

Iknowlongnamerightwhatwasithinking, 2021-08-31:
There is no Libusb0 ...

sthalik, 2021-08-31:
libusb-win32 (v1.2.6.0)

Sorry. updating release notes

Iknowlongnamerightwhatwasithinking, 2021-08-31:
ok right, will try that right now. S!

Iknowlongnamerightwhatwasithinking, 2021-08-31:
looks good. Camera FPS looks right this time. Awesome. : - ) Thank you.

sthalik, 2021-08-31:
Can you get the LED to turn itself off when stopping tracking?

Iknowlongnamerightwhatwasithinking, 2021-08-31:
The RED led turns on and off with the "start" button ( tracking = on ) . The BLUE one stays on full time. OK ?

sthalik, 2021-08-31:
Yup.

Iknowlongnamerightwhatwasithinking, 2021-08-31:
Glad to help. OpenTrack rules. : -)