opentrack/opentrack #1324: Crash when starting tracking with neuralnet

FRad1, 2021-09-02:
I had the Universal PS3 Eye Driver 1.0 beta 2 driver installed. I uninstalled it and followed all the instructions of the opentrack 2021.1.0 release to install the new driver. All good but when I try to start tracking it crashes with this event:

> Faulting application name: opentrack.exe, version: 0.0.0.0, time stamp: 0x612dd42d
Faulting module name: opentrack-tracker-neuralnet.dll, version: 0.0.0.0, time stamp: 0x612dd493
Exception code: 0xc0000409
Fault offset: 0x00140a65
Faulting process id: 0x2a7c
Faulting application start time: 0x01d7a02b2a030d3d
Faulting application path: C:\Track\opentrack-2021.1.0-win32-portable\install\opentrack.exe
Faulting module path: C:\Track\opentrack-2021.1.0-win32-portable\install\modules\opentrack-tracker-neuralnet.dll
Report Id: aa5d7d82-be29-420b-aed4-0fa98af17e7d
Faulting package full name: 
Faulting package-relative application ID:

GO63-samara, 2021-09-02:
In my opinion this is not related to the neuralnet tracker. It has to do with the camera driver. Try running point tracker. It looks like there will be the same error.

I would try installing libusb-win32 again. How to install libusb-win32 was explained in more detail by @blueeagle69 here: https://github.com/opentrack/opentrack/issues/1135

FRad1, 2021-09-02:
> 
> 
> In my opinion this is not related to the neuralnet tracker. It has to do with the camera driver. Try running point tracker. It looks like there will be the same error.
> 
> I would try installing libusb-win32 again. How to install libusb-win32 was explained in more detail by @blueeagle69 here: #1135

Yeah, same problem with point tracker. Already reinstalled the driver and restarted the PC, same problem. BTW, AITrack does not have any problems using my ps3 eye camera.

sthalik, 2021-09-03:
In Task Manager, on the details tab select the crashing opentrack process, right click and select "create dump file". Zip the file and paste it into this text box.

FRad1, 2021-09-03:
>In Task Manager, on the details tab select the crashing opentrack process, right click and select "create dump file". Zip the file and paste it into this text box.

It's not crashing anymore, just cant use it because of the "PS3 Eye driver error: camera init failed: Access denied (insufficient permissions)" Error

dssrtfx, 2021-09-03:
I have the same issue, the "insufficient permissions" alert box and refusal to start tracking.
My process was:

1. Turn off Windows automatically installing drivers
2. Go through Device Manager and Uninstall Device (plus checkmarking Remove Driver) on every entry that said PS3 Eye
3. Uninstalling CLEye driver using Apps and Features
4. Unplug PS3 Eye
5. Restart computer
6. Plug in PS3 Eye
7. Run Zadig
8. Install lubusb-win32 1.2.6.0
9. Unplug then replug PS3 Eye
10. Start Opentrack 2021.1.0
11. Select "PS3 Eye open driver"
12. Click "Start"

As the OP said, this is not a crash, this is clicking "Start" tracking, Opentrack freezing for 6-ish seconds, then an alert popup saying "PS3 Eye driver error: camera init failed: Access denied (insufficient permissions)" pops up, you click Close on the alert popup, and click "Stop" tracking because the camera output window doesn't show up in Opentrack nor is it tracking input, it's just stuck there until you click "Stop" tracking.

Edition	Windows 10 Pro
Version	21H1
Installed on	‎11/‎21/‎2020
OS build	19043.1202
Experience	Windows Feature Experience Pack 120.2212.3530.0

sthalik, 2021-09-03:
Try using the different USB controller. If you're using USB 3.0,l use USB 2.0, or vice versa.

dssrtfx, 2021-09-03:
Tried switching from USB 3 to USB 2, same issue.

FRad1, 2021-09-03:
Mine also is a ps3 eye device name USB Camera-B4.09.24.1

dssrtfx, 2021-09-03:
Using Zadig to Replace Driver with libusbk instead of libusb-win32 seems to let the camera function successfully, but see #1325 for why I'll still use opentrack version 2.3.13 for the time being.

I do appreciate all of you who work on Opentrack, it's been an indispensable part of my gaming life for a good decade and...a half? It's been a long time.

sthalik, 2021-09-03:
> I do appreciate all of you who work on Opentrack, it's been an indispensable part of my gaming life for a good decade and...a half? It's been a long time.

It's been nine years. Damn, time flies fast.

It's a problem with the libusb driver if we can't predetermine whether to use libusb-win32 or libusbK. @dssrtfx what Windows version are you using? Does it work for you with both USB controllers or just one? @FRad1 can you check the suggestion of using libusbK as well?

dssrtfx, 2021-09-03:
For the Windows version, in one of my posts above I pasted my build info that shows up in the System settings page, or am I misunderstanding?

FRad1, 2021-09-03:
>@FRad1 can you check the suggestion of using libusbK as well?

With libusbK the pointracker works but the camera is at like 5 fps (very slow) and neuralnet just crashes opentrack when you hit start tracker:

>Faulting application name: opentrack.exe, version: 0.0.0.0, time stamp: 0x612dd42d
Faulting module name: opentrack-tracker-neuralnet.dll, version: 0.0.0.0, time stamp: 0x612dd493
Exception code: 0xc0000409
Fault offset: 0x00140a65
Faulting process id: 0x1fb4
Faulting application start time: 0x01d7a0db313f9e73
Faulting application path: C:\Track\opentrack-2021.1.0-win32-portable\install\opentrack.exe
Faulting module path: C:\Track\opentrack-2021.1.0-win32-portable\install\modules\opentrack-tracker-neuralnet.dll
Report Id: b95f3433-541c-4f1a-b3b5-4598f8fd6fc6
Faulting package full name: 
Faulting package-relative application ID:

FRad1, 2021-09-03:
libusb-win32 now makes even [AITrack](https://github.com/AIRLegend/aitrack/releases) crash after seconds of starting opentrack track, I'm going back to [PS3 Eye Driver 1.0 beta 2 driver.](https://github.com/jkevin/PS3EyeDirectShow/releases)

sthalik, 2021-09-03:
@FRad1 grab the `2021.1.1` build from <<http://opentrack.misaki.pl>> and see if the crash is gone.

MikeChugunov, 2021-09-03:
Encountered same issue on `2021.1.1` and `2021.1.0_pre1` , but launching with open driver  on `2.3.13` helped somehow and now camera works in latest releases.

FRad1, 2021-09-03:
Now I'm using the [PS3 Eye Driver 1.0 beta 2](https://github.com/jkevin/PS3EyeDirectShow/releases) driver with the neuralnet and it works great.

sthalik, 2021-09-04:
So is the neuralnet crash still present on `2021.1.1`? Unless any of you manage to confirm it, it's not possible to fix.

FRad1, 2021-09-04:
@sthalik 

>@FRad1 grab the 2021.1.1 build from <http://opentrack.misaki.pl> and see if the crash is gone.

Confirmed working without problems with lubusb0 v1.2.6.0 on msfs2020 using that build

petsild, 2021-09-04:
Hi,
Unfortunately, even the test build with the Open driver does not work for me.
I'm happy to return to the prehistoric driver.

seece, 2021-09-05:
Hello,

I'm also having this "camera init failed: Access denied" issue with the new built-in PS3 driver in opentrack-2021.1.2. I uninstalled old drivers (both iPi and CL) and also removed their inf files with `pnputil /delete-driver something.inf /uninstall` after checking the file names with `pnputil -e`. I followed the installation instructions shown on the [2021.1.2 release page](https://github.com/opentrack/opentrack/releases/tag/opentrack-2021.1.2).

The Universal PS3 Eye Driver 1.0 beta 2 does not work either: only a black screen in shown in VLC Direct Show.
The opentrack-2021.1.1-test version also shows the same "Access denied" message.

For what it's worth, AITrack v0.6.5-alpha *does* display a camera image but it's clearly corrupted with some pixel columns missing.

After messing with all the drivers, Zadig only lists a single "USB Camera-B4.09.24.1" device instead of the two "interfaces" that were there before. Don't know what happened there.
I also tried another PS3 camera and it behaves the same way, so the camera itself should be fine.

petsild, 2021-09-06:
What is the difference between libusb-win32 vs libusbk drivers?

FRad1, 2021-09-11:
With 2021.1.2 its again not working, does nothing when you press "Start", everything goes grey but nothing happens and I only can click the "Stop" button , neuralnet and pointtracker arent working.