opentrack/opentrack #1325: 2021.1.0 Not creating TrackIR device?

dssrtfx, 2021-09-03:
I was having problems with the new PS3 Eye open driver, so decided to try the neuralnet tracker.
It's tracking my face with a C920 perfectly, so I tried it out with DCS World and BMS 4.35, merely copying the config I had working with the previous Opentrack and PS3 Eye open driver versions and changing the Input to "neuralnet tracker," again, just changing the camera.
Neither game detected a "TrackIR" device so headtracking will not work. How to fix?

sthalik, 2021-09-03:
Verify registry key exists as string, `Path`, with location of `HKEY_CURRENT_USER\SOFTWARE\NaturalPoint\NATURALPOINT\NPClient Location`.

els2, 2021-09-03:
Opentrack 2021.1.0 is not being recognized by the games, I tried it on Falcon BMS, and Strike Fighters 2. The previous version 2.3.13 is working and being recognized by the games correctly, so I went back to using it. I use AITRACK (UDP over network) and it worked correctly in the previous version.

sthalik, 2021-09-03:
@els2 see whether the above registry paths exists, and whether the fake `trackir.exe` gets started as with previous version. The freetrack protocol settings dialog should look like this:

[image]

els2, 2021-09-03:
> @els2 see whether the above registry paths exists, and whether the fake `trackir.exe` gets started as with previous version. The freetrack protocol settings dialog should look like this:
> 
> [image]

I checked the registry path and it is found, and the freetrack protocol settings dialog is present. In the previous version, Opentrack 2.3.13, everything worked fine:
[image]

[image]

sthalik, 2021-09-04:
See if the build `opentrack-2021.1.1-test.7z` from <<http://opentrack.misaki.pl>> fixes the freetrack problem.

GO63-samara, 2021-09-04:
@els2
So you have an empty path in the `NPClient Location` registry key. All you need to do is click **[Locate DLL]** and select **NPclient.dll** from the `modules` folder.
This will fix the `NPClient Location` registry key.

sthalik, 2021-09-04:
@GO63-samara  That won't fix it. The path will get reset to an incorrect value again.

els2, 2021-09-04:
> See if the build `opentrack-2021.1.1-test.7z` from <http://opentrack.misaki.pl> fixes the freetrack problem.

OK, thank you very much, with this fix you indicated now Opentrack 2021.1.0 is working correctly. Thank you all for your help.

dssrtfx, 2021-09-16:
As for me, 2021.1.2 seems to work! Thank you for doing whatever it was!

closed 2021-09-16 by dssrtfx