opentrack/opentrack #1326: antivirus hit

DarkWolf303, 2021-09-04:
got this after running a new install (2021.1.1) from the .exe installer. 

"The file C:\Program Files (x86)\opentrack\modules\opentrack-tracker-kinect-face.dll is infected with Gen:Variant.Razy.902046 and was moved to quarantine. It is recommended that you run a System Scan to make sure your system is clean."

freecodecampster, 2021-09-20:
Trojan:Script/Wacatac.B!ml detected! Could the authors submit to Microsoft  so it passes Windows Defender to get the all clear.

makattack, 2021-09-24:
Looks like Windows Defender is no longer alarming on the standalone archive and its contents.

tenevas666, 2021-11-11:
> Trojan:Script/Wacatac.B!ml détecté! Les auteurs pourraient-ils soumettre à Microsoft afin qu’il passe Windows Defender pour obtenir le tout clair.

Oui cela serait le minimum.....car j'ai sans cesse un positif trojan a chaque fois

sthalik, 2021-11-19:
See #1371

closed 2021-11-19 by sthalik