opentrack/opentrack #1328: Not able to install the libusb-win32 (v1.2.6.0) driver

RLONE1, 2021-09-06:
I can't for the life of me seem to be able to install the libusb-win32 (v1.2.6.0), It just fails every time no matter what I try, any ideas?

Len62, 2021-09-08:
I managed to get it installed earlier this evening. What are the steps you are taking to install it?

RLONE1, 2021-09-08:
I've tried uninstalling the driver, replacing the driver, installed cl-driver, downloaded the driver files via zadig and tried installing via windows driver update, tried almost all my ports but nothing works. It gives me failed to install. I'm starting to think it's the actual hardware in the cam that there is something wrong with.

Len62, 2021-09-08:
Here's how I did it: Uninstall the camera via device manager, deleting the cl-eye driver. Unplugged the camera, plugged it back in and ran zadig-2.6.exe. Under options in zadig I selected list all devices. From the drop down menu select the camera, make sure it is the interface zero. Next to the green arrow make sure "libusb-win32 (v1.2.6.0)" is selected. I then clicked reinstall driver and it went about installing the driver, took a few minutes, once I rebooted my system OT could find and use the cam.

RLONE1, 2021-09-09:
[image]

Tried doing exactly what you said and still, nothing. I really don't get it.

RLONE1, 2021-09-09:
Ok, so I managed to install the driver, by disabling driver signature enforcement but opentrack does not find my cam for some reason now.

Len62, 2021-09-09:
Great! Good info for others that may find the same issue on their systems!

RLONE1, 2021-09-09:
Yeah, always a good feel when your troubleshooting finally pays off ^^ And if i'm not mistaken people should install libusbK instead of libusb-win32 as it is more advanced in development and works better for newer OS's like win10/11

mangkoran, 2021-09-10:
Might out of OP topic a bit: Looking at your USB device name, seems you use the PS3 Eye camera. Do you able to use the camera as webcam?

RLONE1, 2021-09-10:
> Might out of OP topic a bit: Looking at your USB device name, seems you use the PS3 Eye camera. Do you able to use the camera as webcam?

No. Try installing WinUSB driver for that maybe.

mangkoran, 2021-09-10:
I have tried WinUSB and it said "Camera failed to connect". It seems that it can only be used with `libusb` driver, of which `opentrack` based on. Thank you for your response, and I won't further asking this topic here.

ThoWmas31, 2021-09-18:
i ran a similar problem, i'm pretty sure to have the  right USB driver installed 

[https://i.imgur.com/dVuicx4.png](url)
and i got this error ( i tried to install and run as administrator nothing change)
[https://i.imgur.com/4mGeXZw.png](url)

what i'm doing wrong ? 
It's way than a easy operation to do ....

WF000, 2021-09-18:
Thanks for the info guys.
Please, one question tho, .. 
 Can you use PS3eye "Microphone" with this universal libusb driver, does it get detected in windows10. ?

ThoWmas31, 2021-09-18:
Not able to use like a microphone for me

ThoWmas31, 2021-09-18:
working  with libusbK

sthalik, 2021-09-18:
> Can you use PS3eye "Microphone" with this universal libusb driver, does it get detected in windows10. ?

Yes. In case of problems, check Device Manager.

RLONE1, 2021-09-26:
> 
> 
> Not able to use like a microphone for me

I'm able to use the mic but sometimes it's finicky to get working so you might have to use different ports and or clean unplugged device data, something that is possible to do with a software called driver booster. That has fixed the problem for not being able to use the mic at-least for me.