opentrack/opentrack #1331: Use PS3 Eye Driver as

mangkoran, 2021-09-08:


mangkoran, 2021-09-08:
Sorry I mistyped the Enter key. I just found out opentrack's repository of the open-source PS3 Eye driver is the most recent/updated (https://github.com/opentrack/PS3EYEDriver). I would like to use the driver as standalone, not for opentrack purpose (in another words, use it for just webcam). Is it possible? If so, how do I do that? I already read from "Release" page that I need to use zadig, but I'm not really sure how to do it.

mangkoran, 2021-09-10:
What I am looking for is use case like https://github.com/jkevin/PS3EyeDirectShow. However, the driver from that repository seems unable to provide 60 FPS and above. The repository also looks like unmaintained.

mangkoran, 2021-09-10:
> PS3 Eye open driver
> opentrack now has a standalone driver for the PS3 Eye, and it's mature enough for its existence to be advertised more widely. Now you don't have to pay for (or pirate, for that matter) the CL Eye driver that's buggy, slow, unmaintained and broken with every other Windows 10 update. The crashes are gone and there's less CPU usage. It's also free as in beer.

What I thought about this description is that `opentrack`'s PS3 cam driver should replace CL Eye driver, which should also able to provide the functionality of CL Eye driver. CMIIW.

sthalik, 2021-09-11:
You can still use the CL-Eye driver if you want. The open driver is only listed when there's libusb already installed as the driver.

Normally it helps trying to connect the camera to a different USB controller, and trying both libusb-win32 and libusbK.

mirh, 2023-08-18:
How about media foundation?
Presumably that has way more samples and boilerplate available that aren't decades old. 

p.s. also of note that the original inspirit repo has a lot of pending PRs that look pretty juicy

sthalik, 2023-08-19:
> also of note that the original inspirit repo has a lot of pending PRs that look pretty juicy

I think I merged all things across forks that were relevant when creating my own fork. The microphone branch wasn't relevant to opentrack though. The libusb driver is being attached solely to the camera device, leaving the mic to be supported by the OS which it can already do.

mirh, 2023-08-19:
Ok yes, sorry, you have a point. 
You did eventually merge just about everything up to around late 2018 (except perhaps the very minor https://github.com/inspirit/PS3EYEDriver/pull/42 and https://github.com/inspirit/PS3EYEDriver/pull/48). 
There's still [one](https://github.com/opentrack/PS3EYEDriver/compare/master...jkevin:PS3EYEDriver:master) or [two](https://github.com/opentrack/PS3EYEDriver/compare/master...marcel303:PS3EYEDriver:master) missing RGB-related fixes though.

sthalik, 2023-08-19:
Flipping and the test pattern were implemented, and RGBA wasn't needed. Changing the resolution can be easily implemented but opentrack won't benefit from it.

closed 2023-11-05 by sthalik