opentrack/opentrack #1332: camera::info::num_channels support in video-opencv?

DaWelter, 2021-09-09:
Hi

The field `num_channels` is currently ignored in `opencv_camera_impl::cam::start()`. Should it be supported? 

I mean, should the `num_channels` setting guarantee that `get_frame_()` returns an image with respective channels?

... within what can be reasonably expected of course, i.e. 1 and 3 ;-)

If so it looks like a bug. I could fix it I guess. The video capture api of opencv has some fields we could try. And if that does not work we could convert the frame with `cv::cvtColor`.

For reference
https://docs.opencv.org/3.4/d4/d15/group__videoio__flags__base.html
https://answers.opencv.org/question/196617/right-way-to-setup-cvvideocapturemodes/

DaWelter, 2021-09-09:
P.s. how about an API which returns both a monochrome image as well as RGB (or BGR?!)?

The motivation is both PT and NNT use color images to draw information overlays in, but their actual pose algorithms only use monochrome images.

With a clever subroutine we could minimize the number of color conversions and have relatively idiot proof client code.

sthalik, 2021-09-10:
Treat the input `num_channels` value as advisory. The API may return a color image since grayscale output is non-standard. Right now the code in the PT tracker asks for grayscale, but is indifferent toward what is actually returned.

Returning both isn't very useful as the entire point of the channel count selector is reducing CPU usage, and getting rid of the crappy Bayer filter. Otherwise you'd want to return a color image and run `cv::cvtColor` anyway. Doesn't matter which side of the API it's at.

DaWelter, 2021-09-18:
> Doesn't matter which side of the API it's at.

Exactly, that's why I figured we could eliminate duplicated code. But maybe the code pieces are too tracker-specific to come up with a clean "high-level-API". So that's fine for me.

> Treat the input num_channels value as advisory

:+1:

closed 2021-09-18 by DaWelter