opentrack/opentrack #1333: massive stutters in tracking-translation

PhntmV2, 2021-09-10:
Hey there. First of all Im sorry if Im talking about a problem that already has been solved. I just couldnt find it.
My problem is the following: Suddenly after over a year of succesfully using opentrack on DCS, the tracking translation starts to lag massively everytime I start the headtracking. My camera and the head-clip do seem to work fine. The dots on the left side look as the always did, aswell as the tracking itsself works very fine. Sadly the translation lags super hard and makes it unusable for me. 
I downloaded the newest version of opentrack, just to make sure. That didnt change anything. 
[image]

Is there anything I can try or do? 
Thank you!

GO63-samara, 2021-09-10:
The yellow cross on the screen shows that `Model position` is not set. Therefore, the offsets are also incorrect.
You just need to calibrate the offset in `PointTracker Settings` - `Model position` - `Start calibration`.
There was the same issue here: https://github.com/opentrack/opentrack/issues/1293

PhntmV2, 2021-09-11:
Thanks for your help. I did the calibration but sadly that didnt really help. I still have stutters. Have a look at this video. I recorded it after the calibration. The octopus is lagging still.

https://user-images.githubusercontent.com/90453726/132966366-922ab7cf-6c2e-4f28-bc78-b6e4a49dd53d.mp4

GO63-samara, 2021-09-12:
The delay may be due to settings:

- Accela filter. Too much smoothing and too large dead zones. To check, try disabling the filter.
- There is a delay in the output of the Octopus itself. It is displayed with an FPS of about 20. But this does not affect the Opentrack's output coordinates.
- Low FPS of a videocam. This is not visible in the video.

In addition, the video shows that you have set very sharp Mapping curves, especially in the rear hemisphere. Do you really need it?

PhntmV2, 2021-09-13:
It seems to be the accela filter. But lowering the smoothing and the deadzones doesnt change anything. Switching the filter does work for me tho. Is there a downside in using a different filter as accela? 
Thanks so much!

GO63-samara, 2021-09-13:
Then try the Hamilton filter. There are positive reviews. A full description of the Hamilton filter settings (in Russian) is here:
https://sites.google.com/site/diyheadtracking/устройства-обзора-для-авиасимуляторов/opentrack/opentrack-фильтр-hamilton