opentrack/opentrack #1334: Build from opentrack source with steamvr

Nobatgeldi, 2021-09-11:
Hi there,

I did everything right to build with MS Visual Studio 2019 and build completed successfully. But when I run opentrack there is no Steam VR inpurt selection. 
[image]

When I download latest portable opentrack, I can select Steam VR input. 

[image]

I build for x64 system.

Nobatgeldi, 2021-09-11:
When I check Cmake GUI log, there is no steamvr modul.

Selecting Windows SDK version 10.0.19041.0 to target Windows 10.0.19043.
User settings file 'sdk-paths-COVISART@MSVC-windows.cmake' doesn't exist
module api
module compat
module csv
module dinput
module filter-accela
module filter-ewma
module filter-hamilton
module user-interface
module logic
module migration
module executable
module options
module pose-widget
module proto-fgfs
module proto-freetrack
module proto-win32-mouse
module proto-simconnect
module proto-udp
module spline
module tracker-freepie-udp
module tracker-fusion
module tracker-hatire
module tracker-joystick
module tracker-s2bot
module tracker-test
module tracker-udp
module video
Compile flags:
   CXX: /DWIN32 /D_WINDOWS /W3 /GR /EHsc
   CXX_RELEASE: /MD /O2 /Ob2 /DNDEBUG
Link flags:
   LINK_SHARED: /machine:x64
   LINK_SHARED_RELEASE: /INCREMENTAL:NO
Static archive flags:
   STATIC: /machine:x64
   STATIC_RELEASE: 

Configuring done
Generating done

GO63-samara, 2021-09-11:
You need to use `opentrack-depends`.

The complete build of Opentrack from source is described here (in Russian):
https://sites.google.com/site/diyheadtracking/устройства-обзора-для-авиасимуляторов/opentrack/opentrack-сборка-из-исходников
According to the replies of the thread https://github.com/opentrack/opentrack/issues/1228

Good luck!

sthalik, 2021-09-16:
Run `git submodule update --init` on the opentrack-depends repo.

Nobatgeldi, 2023-07-07:
After clonnig repo, I cloned  [opentrack-depends](https://github.com/opentrack/opentrack-depends) repo using `--recurse-submodules`. Problem solved

closed 2023-07-07 by Nobatgeldi