opentrack/opentrack #1335: Opentrack crash whe I press Start

eudesrafael, 2021-09-17:
Sorry if it was already discussed but I couldn't find a topic with a solution to me. Opentrack crashes every time I press the start button. I use it some months ago and it was fine, now this problem appears. I updated the installation to the latest version but I did not have success. If someone with more experience could help I would appreciate it a lot.

GO63-samara, 2021-09-18:
To find out what's the matter:

- Launch `Opentrack` from the command line of a `cmd` window.
- Click `[Start]`.
- Copy the log from the `cmd` window here.

eudesrafael, 2021-09-18:
Mate, I have no idea of what happened but simply by launching through cmd, the software is working right now. I even closed and launched through the desktop shortcut like yesterday and everything seems to be fine. Anyway I will paste the log here if it helps in some way. Thanks!!!

DEBUG [F:\dev\opentrack\api/plugin-support.hpp:107]: library "kinect-face" failed: "Cannot load library C:\\Program Files (x86)\\opentrack\\modules\\\\opentrack-tracker-kinect-face.dll: NÒo foi possÝvel encontrar o m¾dulo especificado."
[ INFO:0] global F:\dev\opentrack-depends\opencv\modules\core\src\parallel\registry_parallel.impl.hpp (96) cv::parallel::ParallelBackendRegistry::ParallelBackendRegistry core(parallel): Enabled backends(0, sorted by priority): N/A
[ INFO:0] global ..\modules\videoio\src\videoio_registry.cpp (217) cv::`anonymous-namespace'::VideoBackendRegistry::VideoBackendRegistry VIDEOIO: Enabled backends(3, sorted by priority): DSHOW(1000); CV_IMAGES(990); CV_MJPEG(980)

*Obs: NÒo foi possÝvel encontrar o m¾dulo especificado = Was not possible to find the specified module.

GO63-samara, 2021-09-18:
> Was not possible to find the specified module.

`kinect-face` does not work on `Windows 7`. Requires `Windows 8` or newer.

eudesrafael, 2021-09-18:
I actually use Windows 10, but as I said, is working fine now! (At least with point tracker which is what I use)

closed 2021-09-18 by eudesrafael

sthalik, 2021-09-18:
FWIW, the Kinect message is coming out on startup from the module loader. All modules are always loaded, he's not necessarily using it.

GO63-samara, 2021-09-18:
> All modules are always loaded

FWIW, this is not very good, it wastes memory. Guess it has something to do with static linking.

GO63-samara, 2021-09-18:
Perhaps it need to advise users to delete unused dlls from the modules folder to save memory?

sthalik, 2021-09-19:
By loaded, I mean `LoadLibraryEx` for the dll files. They need their metadata, as well as being able to run their config prior to starting tracking. It's not like the trackers are getting instantiated. It's not worth the trouble with static initializer and destructor leaks to save just a little bit of memory. Memory usage is low when tracking is stopped.