opentrack/opentrack #1337: Tracking Freezes with PS3 Eye

IGotIn, 2021-09-18:
Used to have this issue long long ago but it vanished (maybe when updating to 2.3? I honestly don't know) and has now returned.

Was using opentrack 2.3 with a PS3 eye for eons with the CLEye driver version 5.3.0.0341, everything worked great.
Only issue I would have is sometimes the camera wouldn't be detected when it was first plugged in, after a few tries it would be recognised and I could begin tracking and it would continue tracking for as long as i kept it running, not once would it freeze.

Sometimes the tracking would go a lil wacky and start yawing and pitching steadily ion random directions but re-centering the tracking would fix it and I put that issue down to a reflection or something being picked up by the tracker.

Recently I installed opentrack version 2021.1.2 and tried to have a go with the open ps3 eye drivers.

I've tried using zadig to install libusb-win32 (v1.2.6.0) as suggested, tracking still freezes after a few minutes
libusbK (v3.1.0.0) Freezes the same way.
and even the 'do not use' but still works WinUSB (v6.1.7600.16385) and that freezes too.
Running opentrack as Admin, Freezes still
Tried "do not shut off this device to save power" still getting freezing.

Everytime it freezes I can just restart tracking and continue like nothing has happened.
When frozen the image remains in the tracking window twitching ever so slightly, same for the raw data.

I've tried every combination I can think of for uninstalling-rebooting-reinstalling, replacing, clean install etc. for the USB drivers but everytime I get the same issue, freezing after a few minutes.
On the flip side however the camera is always detected first try now :D Though I'd much prefer to have that issue back instead of this intermittent freezing issue lol.

IGotIn, 2021-09-18:
Froze again after I rolled back to 2.3.13 , tried all the drivers, same issue.

Drew a pentagram on the floor and did a system restore.

Back on Opentrack 2.3.11 and the PS3 eye is on Code Labs driver 5.3.0.341 from the beforefore times of 2012 and all is solid as a rock thus far, no freezing for an hour now, before I would have a crash under ten minutes consistently.

Win10 x64 by the way, should've mentioned that.

GO63-samara, 2021-09-18:
If you are using multiple versions of Opentrack, you may need to point to the correct `NPclient.dll` when you change versions:
`freetrack protocol settings` - `Locate DLL` - select `NPclient.dll` from the `modules` folder - `Open` - `OK`.

Also, to determine the problem, try running Opentrack from the command line, like here: https://github.com/opentrack/opentrack/issues/1335

IGotIn, 2021-09-18:
Here it is...

DEBUG [F:\dev\opentrack\api/plugin-support.hpp:107]: library "kinect-face" failed: "Cannot load library C:\\Program Files (x86)\\opentrack\\modules\\\\opentrack-tracker-kinect-face.dll: The specified module could not be found."
DEBUG [F:\dev\opentrack\api/plugin-support.hpp:107]: library "rift-025" failed: "Cannot load library C:\\Program Files (x86)\\opentrack\\modules\\\\opentrack-tracker-rift-025.dll: The specified procedure could not be found."
DEBUG [F:\dev\opentrack\api/plugin-support.hpp:107]: library "rift-042" failed: "Cannot load library C:\\Program Files (x86)\\opentrack\\modules\\\\opentrack-tracker-rift-042.dll: The specified procedure could not be found."
DEBUG [F:\dev\opentrack\api/plugin-support.hpp:107]: library "rift-080" failed: "Cannot load library C:\\Program Files (x86)\\opentrack\\modules\\\\opentrack-tracker-rift-080.dll: The specified procedure could not be found."
DEBUG [F:\dev\opentrack\api/plugin-support.hpp:253]: duplicate lib "opentrack-tracker-rift.dll" ident "Oculus Rift runtime 1.4.0 -- HMD"

Just reinstalled 2021.1.2 and got rid of the old 2.3.13 folder, made sure 2021 was pointing to the correct dll too and it's still running without crashes thus far.
This also fixed the problem I had of [black text](https://github.com/opentrack/opentrack/issues/1336) in the tracker preview window, everything there is back to normal as well.

Still running the CLEye drivers from 2012 though, worried the freezing might rear its ugly head oncemore by installing the open drivers but I'll give them a go oncemore.

IGotIn, 2021-09-18:
So I had CLeye drivers, I could see the red text of how many pixels with PS3 eye selected in opentrack I could also select the ps3 open driver and still see an image though with black text instead of red. 
Uninstalled Code Labs drivers via add/remove and installed the libusb-win32 as recommended with zadig.
Got this... 
DEBUG [F:\dev\opentrack\api/plugin-support.hpp:107]: library "kinect-face" failed: "Cannot load library C:\\Program Files (x86)\\opentrack\\modules\\\\opentrack-tracker-kinect-face.dll: The specified module could not be found."
DEBUG [F:\dev\opentrack\api/plugin-support.hpp:107]: library "rift-025" failed: "Cannot load library C:\\Program Files (x86)\\opentrack\\modules\\\\opentrack-tracker-rift-025.dll: The specified procedure could not be found."
DEBUG [F:\dev\opentrack\api/plugin-support.hpp:107]: library "rift-042" failed: "Cannot load library C:\\Program Files (x86)\\opentrack\\modules\\\\opentrack-tracker-rift-042.dll: The specified procedure could not be found."
DEBUG [F:\dev\opentrack\api/plugin-support.hpp:107]: library "rift-080" failed: "Cannot load library C:\\Program Files (x86)\\opentrack\\modules\\\\opentrack-tracker-rift-080.dll: The specified procedure could not be found."
DEBUG [F:\dev\opentrack\api/plugin-support.hpp:253]: duplicate lib "opentrack-tracker-rift.dll" ident "Oculus Rift runtime 1.4.0 -- HMD"
[ INFO:0] global F:\dev\opentrack-depends\opencv\modules\core\src\parallel\registry_parallel.impl.hpp (96) cv::parallel::ParallelBackendRegistry::ParallelBackendRegistry core(parallel): Enabled backends(0, sorted by priority): N/A
libusb: error [winusbx_claim_interface] could not access interface 0: [87] The parameter is incorrect.
ps3eye: device claim interface error: -3
ps3eye: usb error Access denied (insufficient permissions) (-3)
libusb: error [winusbx_claim_interface] could not access interface 0: [87] The parameter is incorrect.
ps3eye: device claim interface error: -3
ps3eye: usb error Access denied (insufficient permissions) (-3)

So I replaced the driver with libusbK and again replaced that with libusb-win32 using zadig.

Works but I get black text in the tracking window and the freezing problem returns after 10-20 minutes, logs...

DEBUG [F:\dev\opentrack\api/plugin-support.hpp:107]: library "rift-025" failed: "Cannot load library C:\\Program Files (x86)\\opentrack\\modules\\\\opentrack-tracker-rift-025.dll: The specified procedure could not be found."
DEBUG [F:\dev\opentrack\api/plugin-support.hpp:107]: library "rift-042" failed: "Cannot load library C:\\Program Files (x86)\\opentrack\\modules\\\\opentrack-tracker-rift-042.dll: The specified procedure could not be found."
DEBUG [F:\dev\opentrack\api/plugin-support.hpp:107]: library "rift-080" failed: "Cannot load library C:\\Program Files (x86)\\opentrack\\modules\\\\opentrack-tracker-rift-080.dll: The specified procedure could not be found."
DEBUG [F:\dev\opentrack\api/plugin-support.hpp:253]: duplicate lib "opentrack-tracker-rift.dll" ident "Oculus Rift runtime 1.4.0 -- HMD"
[ INFO:0] global F:\dev\opentrack-depends\opencv\modules\core\src\parallel\registry_parallel.impl.hpp (96) cv::parallel::ParallelBackendRegistry::ParallelBackendRegistry core(parallel): Enabled backends(0, sorted by priority): N/A
ps3eye: payload error, data[1]=78
ps3eye: payload error, data[1]=79
ps3eye: payload error, data[1]=78
ps3eye: payload error, data[1]=79
ps3eye: payload error, data[1]=79
ps3eye: payload error, data[1]=78
ps3eye: payload error, data[1]=78
ps3eye: payload error, data[1]=79
ps3eye: payload error, data[1]=78
ps3eye: payload error, data[1]=78
ps3eye: payload error, data[1]=79
ps3eye: payload error, data[1]=79
ps3eye: payload error, data[1]=79
ps3eye: payload error, data[1]=78
ps3eye: payload error, data[1]=79
ps3eye: payload error, data[1]=79
ps3eye: payload error, data[1]=79
ps3eye: payload error, data[1]=79
ps3eye: payload error, data[1]=78
ps3eye: payload error, data[1]=78

after that I can restart and the tracking will pick up again

IGotIn, 2021-09-18:
Restarted the machine to see if any gremlins ran away, same thing just happened much quicker this time, frozen screen, same driver libusb-win32 v1.2.6.0 on Opentrack 2021.1.2

> 
> DEBUG [F:\dev\opentrack\api/plugin-support.hpp:107]: library "kinect-face" failed: "Cannot load library C:\\Program Files (x86)\\opentrack\\modules\\\\opentrack-tracker-kinect-face.dll: The specified module could not be found."
> DEBUG [F:\dev\opentrack\api/plugin-support.hpp:107]: library "rift-025" failed: "Cannot load library C:\\Program Files (x86)\\opentrack\\modules\\\\opentrack-tracker-rift-025.dll: The specified procedure could not be found."
> DEBUG [F:\dev\opentrack\api/plugin-support.hpp:107]: library "rift-042" failed: "Cannot load library C:\\Program Files (x86)\\opentrack\\modules\\\\opentrack-tracker-rift-042.dll: The specified procedure could not be found."
> DEBUG [F:\dev\opentrack\api/plugin-support.hpp:107]: library "rift-080" failed: "Cannot load library C:\\Program Files (x86)\\opentrack\\modules\\\\opentrack-tracker-rift-080.dll: The specified procedure could not be found."
> DEBUG [F:\dev\opentrack\api/plugin-support.hpp:253]: duplicate lib "opentrack-tracker-rift.dll" ident "Oculus Rift runtime 1.4.0 -- HMD"
> [ INFO:0] global F:\dev\opentrack-depends\opencv\modules\core\src\parallel\registry_parallel.impl.hpp (96) cv::parallel::ParallelBackendRegistry::ParallelBackendRegistry core(parallel): Enabled backends(0, sorted by priority): N/A
> ps3eye: payload error, data[1]=78
> ps3eye: payload error, data[1]=79
> ps3eye: payload error, data[1]=79
> 

Having a go now at replacing the libusb-win32 drivers with libusbK v3.1.0.0

sthalik, 2021-09-19:
The only things that come to mind is disabling USB selective suspend in power options, and trying the other usb controller -- one has black interior, the other is blue. Otherwise it may be controller-related.

We could try restarting the camera when it hangs, but for that it's necessary to include debug info for the camera subprocess. Let's try this only in case when the other solutions fail.

IGotIn, 2021-09-19:
As far as I'm aware I had disabled usb power saving on the camera along with the host it was showing as connected to through device manager's 'view by connection' in each test but I can make doubley sure on the next go round.

I can have a go with a 3.0 port at some point when I get the time and will update accordingly.

Either way, when connected to USB 1.1/2.0 (black) all drivers start throwing out payload errors until the camera locks up and requires and requires a restart through opentrack, whereby the camera resumes where it left off throwing payload errors until it freezes again.
Also with all other drivers the centre pivot yellow cross and pixel count red text is all black and practically invisible against the mostly black background of the tracking window.

The only exception is the CL-eye-driver (5.3.0.0.341) which zadig sees as WinUSB (v5.3.0.341) instead this driver just occasionally struggles with detecting the camera upon connection to the USB port. 
After a bit of plugging in, out, in, out and shaking all about it eventually gets detected properly.
Once connected, detected and tracking starts it  throws out no payload errors and can function indefinitely from what I've seen so far.
This driver also restores the centre pivot yellow cross and red text for the pixel count.

A most perplexing conundrum!