opentrack/opentrack #1339: Installing the new Ps3 eye open drivers

sutex, 2021-09-20:
Followed these instruction, maybe you can add then to the main post.
Download Zadig.

    Run the executable and click Options/List All Devices
    On the dropdown box find USB Camera Interface 0 (it'll have other text after it, but this is the one you need for the PS3 Eye).
    Below that dropdown you have two others. In the right-hand dropdown, select libusb-win32.
    Lastly, click Replace Driver

Using the libusb-win32 the PS3 Eye driver error: camera init failed, with **Access denied (insufficient permissions)** 

Switched to **libusbK** and that worked. Question, is libusbK a worst option than using the  CL Eye driver?

And how do you fix this **Access denied (insufficient permissions)**?
Running latest win 10 in admin mode and ticked the box for admin for OT.

ounet, 2021-09-20:
I confirm taht I have the same issue and if i use the libusbk drivers that works. I have the same question as sutex.

sthalik, 2021-09-20:
libusbK is the better option due to lower CPU usage. Use whichever works. On my end libusbK doesn't work while libusb-win32 does.

AncestralHeritage, 2021-11-20:
I just wanted to say that I had the same issue and the same solution to sutex. Just installed the libusbK driver and now everything works.

IceNbrn, 2025-01-12:
> I just wanted to say that I had the same issue and the same solution to sutex. Just installed the libusbK driver and now everything works.

I used libusbK driver as well and now it works.  Thanks!