opentrack/opentrack #1340: [major] "new centering modes that fix the long-standing problem with yaw affecting pitch and vice versa" is broken only now for me

u-an-i, 2021-09-22:
With your current release, default settings or settings taken over from March version (idk, i didn't change anything now), I now have the issue

"yaw affecting pitch"

with iOS app SmoothTrack as input.

Note: this is only yaw left right which both lead to pitch down, not the other way round, but possibly a bit of roll too. The input is iPhone SE 2020 front camera looking up from in front (using a smartphone stand). Such an issue I did not have with the previous release. On the contrary, it was "perfect". But perhaps SmoothTrack does/did its own workaround for you.

GO63-samara, 2021-09-22:
**OT-2021.1.2** uses several alignment methods:

Select the centering method from the drop-down list: `[Options]`-`[Shortcuts]`-`"Centering method"`.

- **Disabled**: Centering is disabled.
- **Point**: Point centering, the same as in OT-2.3.10, but without jumps when crossing the border +/- 180 degrees. Well suited for video trackers, although near the input |Pitch| = 90 it has strange behavior similar to "gimbal lock".
- **Wireless VR 360**: Centering using a centering matrix or quaternion, the same as in OT-2.3.11...OT-2.3.13. Ideal for wireless VR headset for 360 degree viewing, no "gimbal lock". But it requires precise installation of the sensor or clip on the head so that in the central position the Roll angle on the "Raw tracker data" panel is close to zero.
- **Roll compensated VR 360**: It is a development of the previous version. It does not require precise installation of the sensor on the head. The sensor or clip can be tilted along the Roll, even at a 45 degree angle. Unlike the first option, there is no "gimbal lock". This centering option is suitable in most cases, especially for owners of an IMU tracker and laptop, when the screen is installed below (or above) eye level.

Do not forget to click `[OK]` to save the centering method selection to the profile file.

**Hint**: Now it is possible to reset the centering without stopping the tracking. To do this, while tracking, switch to `[Disabled]`, then press `[Center]`. This will reset the centering for all centering options.

Which centering method did you choose?

u-an-i, 2021-09-23:
"Roll compensated VR 360" was set.

"Point" works fine for me.

Thank you!

closed 2021-10-15 by u-an-i