opentrack/opentrack #1341: Weird non deadzone when using curve reponse....

Iknowlongnamerightwhatwasithinking, 2021-09-22:
So I have bought this up before but no fix. 

If I add a bit of DZ curve to my Yaw response in mapping properties I get very noticeable Non Dead Zone. 
this is particularly bad when zoomed in in game. 

[image]

As you can see here there should be no movement around the zero range but there is. a lot. 
this has bugged me before and ignored. Well I am back! 

If I do this:

[image]

the issue is gone BUT I lose the curve.... which is what I want. ok. 

Thanks!

GO63-samara, 2021-09-25:
So add one or more points between input 5 and 20. Then you have both a static dead zone and a response curve:

[image]

Using a static dead zone is inconvenient, try selecting a dynamic dead zone in the filter.
Also, if you need to look exactly at the crosshair, it is more convenient to use the `[Zero]` or `[Zero while held]` button.

Iknowlongnamerightwhatwasithinking, 2021-09-25:
There is a problem with the response curve. 

That problem is some kind of bump that is baked into the response curve.

A simple curve from zero shows this problem. There should be no strange "non dead zone" at all. 

In your pic you can even see it at point 5. That weird slight rise. 

In my first pic where the response curve is very low but still not zero 
from about 0 to 2 , you can see that weird bump.  
That is the problem. There should be no "undead zone' between 0 and 2 , 
but there is. it's there, I have seen it a lot and it should not be there. 
The response at 0 should be very very low, then rise up till it's linear. 
What I get is this weird lose bit around 0 and THEN the very very low response 
with the rate changing till it's constant. 

I should not need to add other points, which change what I want, and using other 
buttons to get what I should be getting anyway is just a work around and not addressing the 
core issue.  What is that weird "undead zone" when I use a simple curve in my response? 

Do you want video evidence? 

This is an old problem. This is second time I have brought this up. 
The last time I got the same work around but the problem remains. 

In your pic with your solution you have moved the weird bit our to point 5. 
If you look carefully you can see it in game. It's a slight change in response 
when you move into that part of the curve. If you track back and forth over it you can 
see your pov judder slightly.  It's not cool. 

The one thing I do not want when looking around is the picture to jump. 
and that is exactly what happens over this tiny weird response. 
It's in the way the response departs from zero.  

S!

GO63-samara, 2021-09-26:
So your problem is that when the input Yaw is between 0 and 5, the output Yaw is not always 0?
I have tried but failed to replicate this problem. Everything works for me correctly.
Then send your profile file.
What tracker are you using?

Iknowlongnamerightwhatwasithinking, 2021-09-29:
Away from main PC right now. 

I use the point tracker. Three point cap. 

The problem is that with the curve like in the very top picture. 
There is a loose zone ( it's small ) where there some yaw output when there 
should be very little. It is like a small zone where there is too much output , 
then when I move my head further it stops and follows the curve. 

I don't your filter dead zone and I don't use the "zero" function hot key thing. 

Imagine this: from 0 to maybe .25  the output is not following the output curve. 
It is free to move around quiet a lot. Like it's a worn out bearing with "slop". 
If I zoom in in game this becomes very noticeable. 
I would prefer to have the curve such as in the top picture, which would awesome 
and suit my many various "looking"  styles. BUT the weird lose zone is distracting. 

I can see I am going to have to make a video to show, you... 

Me.

Iknowlongnamerightwhatwasithinking, 2021-11-23:
This issue is still an issue. Release 2021.2.0 still does it. 

Just saying.... Squeaky wheel etc.

sthalik, 2021-12-19:
See if this build works. https://www.dropbox.com/s/8ink2czg1d83ms7/opentrack-master-20211216_02.7z

The first point(s) used as the deadzone must have a Y value of zero.

5YNTH3T1K, 2021-12-20:
Hello. 
Are we on the same page?   If I do this: 

[image]

or this:

[image]

I still get a "Non Dead Zone" ( NDZ )  where my FOV can fidget a tiny bit before the curve starts. 

I think this it what it "feels" like. If you had the "game data" show hundredths of a degree it would show the NDZ.  
IE Game Data Yaw = O.00 -> 0.50  ->  0.99 -> 1.00 etc Because the issues is under 1 degree it does not show up. 

[image]

I will check this with Rise of Flight to see if it is the same. 

Cheers! 

5ynth.

sthalik, 2021-12-21:
Try adding a fourth point in the graph. Catmull-Rom splines require 4 points. Normally it's possible to fake it, but not so much in your case.

5YNTH3T1K, 2021-12-21:
So the graph is not showing the true output just a "what you see is not what you get" output.... interesting. 

Will try the forth point when I get back to PC. 

 Ah... just read wiki on what a spline is. So the spline is not starting at zero on the X line, it is starting at maybe at .30 AND the 0 to .30 is a tiny ramp. like in my diagram. 
 Is there some way to show the actual output in 00.00 of degrees? Some third party software that reads the output , like DIview but for tracking ... just wondering. 

Cheers.

sthalik, 2021-12-21:
> Is there some way to show the actual output in 00.00 of degrees?

Drag and drop your .ini into this text box.

5YNTH3T1K, 2021-12-23:
Note: 
It would not let me drop the file here so I copied the contents. 
I think the four point curve has cured that NDZ issue. 
However I am not sure if there is still something weird around 0 to 0.30 ... so I will keep an eye it. 
Does the spline start working when you add more than one point? This is curious. Surely 0,0 would 
automatically count as a point in the spline? Which means you only need to add three points. 
And a simple curve only needs three points... so why four? Of course an S curve needs five ..... 
This is all very interesting! 

[migrations]
last-migration-at=20180428_00~

[opentrack-ui]
keycode-center=Ctrl+C
z-source-index=6
roll-source-index=6
x-source-index=0
y-source-index=1
center-at-startup=false
centering-mode=1

[tracker-pt]
camera-name=PS3 Eye open driver
camera-fps=75
threshold-slider=@Variant(\0\0\0\x7f\0\0\0\x18::options::slider_value\0@Z\xc0\0\0\0\0\0\0\0\0\0\0\0\0\0@o\xe0\0\0\0\0\0)
min-point-size=8
max-point-size=20
active-model-panel=1
enable-point-filter=false
point-filter-limit=@Variant(\0\0\0\x7f\0\0\0\x18::options::slider_value\0?\x84z\xe1G\xae\x14{?\x84z\xe1G\xae\x14{?\xf0\0\0\0\0\0\0)
automatic-threshold=false
point-filter-coefficient=@Variant(\0\0\0\x7f\0\0\0\x18::options::slider_value\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0@\x10\0\0\0\0\0\0)
model-centroid-z=0
model-centroid-y=0

[opentrack-mappings]
yaw-max-value=10
pitch-max-value=10
pitch-max-output-value=-180
x-max-value=10
y-max-value=10
y-alt-axis-sign=true

[spline-yaw]
points="@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x4@\x4\0\0\0\0\0\0@,\0\0\0\0\0\0@\x12\0\0\0\0\0\0@D\x80\0\0\0\0\0@\x1c\0\0\0\0\0\0@X\0\0\0\0\0\0@$\0\0\0\0\0\0@f\x80\0\0\0\0\0)"

[spline-pitch]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x4@\x4\0\0\0\0\0\0@2\0\0\0\0\0\0@\x14\0\0\0\0\0\0@Q\0\0\0\0\0\0@\x1c\0\0\0\0\0\0@^\x80\0\0\0\0\0@\"\0\0\0\0\0\0@f\x80\0\0\0\0\0)

[accela-sliders]
rotation-sensitivity=@Variant(\0\0\0\x7f\0\0\0\x18::options::slider_value\0?\xc9\x99\x99\x99\x99\x99\x9a?\xa9\x99\x99\x99\x99\x99\x9a@\x4\0\0\0\0\0\0)
rotation-deadzone=@Variant(\0\0\0\x7f\0\0\0\x18::options::slider_value\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0?\xc9\x99\x99\x99\x99\x99\x9a)
translation-deadzone=@Variant(\0\0\0\x7f\0\0\0\x18::options::slider_value\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0?\xf0\0\0\0\0\0\0)
translation-sensitivity=@Variant(\0\0\0\x7f\0\0\0\x18::options::slider_value\0?\xc3\x33\x33\x33\x33\x33\x33?\xa9\x99\x99\x99\x99\x99\x9a?\xf8\0\0\0\0\0\0)

[proto-freetrack]
used-interfaces=0

[video-ps3eye]
exposure=@Variant(\0\0\0\x7f\0\0\0\x18::options::slider_value\0@C\0\0\0\0\0\0\0\0\0\0\0\0\0\0@o\xe0\0\0\0\0\0)
gain=@Variant(\0\0\0\x7f\0\0\0\x18::options::slider_value\0@.\0\0\0\0\0\0\0\0\0\0\0\0\0\0@O\x80\0\0\0\0\0)

[modules]
filter-dll=accela

[spline-Y]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x2?\xe0\0\0\0\0\0\0\0\0\0\0\0\0\0\0@\0\0\0\0\0\0\0@R\xc0\0\0\0\0\0)

[spline-Z]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x1@<\0\0\0\0\0\0\0\0\0\0\0\0\0\0)

[spline-roll]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x1@ep\0\0\0\0\0\0\0\0\0\0\0\0\0)

[spline-X]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x2@\b\0\0\0\0\0\0\0\0\0\0\0\0\0\0@\x14\0\0\0\0\0\0@R\xc0\0\0\0\0\0)

[alt-spline-Y]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x2?\xe0\0\0\0\0\0\0\0\0\0\0\0\0\0\0@\0\0\0\0\0\0\0@R\xc0\0\0\0\0\0)

5YNTH3T1K, 2021-12-23:
[image]

Well in this case only four.  https://en.wikipedia.org/wiki/Flat_spline

sthalik, 2021-12-23:
> Does the spline start working when you add more than one point? This is curious. Surely 0,0 would automatically count as a point in the spline?

When you add more than one point (and the first point not placed on [`x`, 0]) an implicit (0, 0) point gets added.

If you add three points with no initial point on (`x`, 0), then it's enough to make a well-formed Catmull-Rom spline.

> Which means you only need to add three points. And a simple curve only needs three points... so why four?

Catmull-Rom splines are the stiffest type of Hermite splines. They go exactly through the control points, even at the cost of singularities at other places.

What you want to do is place points relatively evenly. Try adding a trailing point at (`x`, 180). See how the spline misbehaves across multiple preceding points as you drag it further to the right (i.e. increase `x`).