opentrack/opentrack #1342: segfault with udp on linux

StanberyTrask, 2021-09-26:
getting a crash when trying to use udp on linux mint. i get an error:
```
DEBUG [/home/stanbery/Downloads/opentrack-opentrack-2021.1.2/logic/pipeline.cpp:225]: nan check failed for: value function: void pipeline_impl::pipeline::logic() line: 464
Segmentation fault (core dumped)
```
i get the vague idea that its the ``#define nan_check(...)`` on line 244 based off the error but i hardly code and cant say for sure.

sthalik, 2021-09-26:
Need backtrace.

StanberyTrask, 2021-09-26:
dont know how relevant this is but it only crashes once data starts getting received 
```
Segmentation fault (core dumped)
*** Segmentation fault
Register dump:

 RAX: 00005571abebba40   RBX: 00005571ab87f559   RCX: 00005571abe713c0
 RDX: 0000000000000000   RSI: ffffffff80000000   RDI: 00005571ab87f540
 RBP: 00005571ab87f540   R8 : 000000000000000b   R9 : 00005571ab87f750
 R10: 00005571ab87f748   R11: 0000000000070ea2   R12: 0000000080000000
 R13: 0000000000000001   R14: 00007f7f877fdb08   R15: 00007f7f877fd9c0
 RSP: 00007f7f877fd8f0

 RIP: 00007f7ffe71a424   EFLAGS: 00010a12

 CS: 0033   FS: 0000   GS: 0000

 Trap: 0000000e   Error: 00000004   OldMask: 00000000   CR2: abebba40

 FPUCW: 0000037f   FPUSW: 00000020   TAG: 00000000
 RIP: 00000000   RDP: 00000000

 ST(0) 0000 0000000000000000   ST(1) 0000 0000000000000000
 ST(2) 0000 0000000000000000   ST(3) 0000 0000000000000000
 ST(4) 0000 0000000000000000   ST(5) ffff b000000000000000
 ST(6) ffff d000000000000000   ST(7) d000 d000000000000000
 mxcsr: 1ff9
 XMM0:  0000000000000000000000008e78e38e XMM1:  0000000000000000000000008e78e38e
 XMM2:  0000000000000000000000008e78e38e XMM3:  0000000000000000000000008e78e38e
 XMM4:  0000000000000000000000008e78e38e XMM5:  0000000000000000000000008e78e38e
 XMM6:  0000000000000000000000008e78e38e XMM7:  0000000000000000000000008e78e38e
 XMM8:  0000000000000000000000008e78e38e XMM9:  0000000000000000000000008e78e38e
 XMM10: 0000000000000000000000008e78e38e XMM11: 0000000000000000000000008e78e38e
 XMM12: 0000000000000000000000008e78e38e XMM13: 0000000000000000000000008e78e38e
 XMM14: 0000000000000000000000008e78e38e XMM15: 0000000000000000000000008e78e38e

Backtrace:
/home/stanbery/Downloads/opentrack-opentrack-2021.1.2/install/libexec/opentrack/opentrack-spline.so(_ZNK13spline_detail6spline18get_value_internalEi+0x44)[0x7f7ffe71a424]
/home/stanbery/Downloads/opentrack-opentrack-2021.1.2/install/libexec/opentrack/opentrack-spline.so(_ZNK13spline_detail6spline17get_value_no_saveEd+0x69)[0x7f7ffe71a7e9]
/home/stanbery/Downloads/opentrack-opentrack-2021.1.2/install/bin/../libexec/opentrack//opentrack-filter-accela.so(+0x5878)[0x7f7ff8989878]
/home/stanbery/Downloads/opentrack-opentrack-2021.1.2/install/bin/../libexec/opentrack//opentrack-filter-accela.so(+0x5e2b)[0x7f7ff8989e2b]
/home/stanbery/Downloads/opentrack-opentrack-2021.1.2/install/libexec/opentrack/opentrack-logic.so(_ZNK13pipeline_impl8pipeline18maybe_apply_filterERKN10simple_mat3MatIdLi6ELi1EEE+0x41)[0x7f7ffe73f001]
/home/stanbery/Downloads/opentrack-opentrack-2021.1.2/install/libexec/opentrack/opentrack-logic.so(_ZN13pipeline_impl8pipeline5logicEv+0x2aa)[0x7f7ffe73ff3a]
/home/stanbery/Downloads/opentrack-opentrack-2021.1.2/install/libexec/opentrack/opentrack-logic.so(_ZN13pipeline_impl8pipeline3runEv+0x1f8)[0x7f7ffe740618]
/lib/x86_64-linux-gnu/libQt5Core.so.5(+0xbe9d2)[0x7f7ffd53b9d2]
/lib/x86_64-linux-gnu/libpthread.so.0(+0x9609)[0x7f7ffcda3609]
/lib/x86_64-linux-gnu/libc.so.6(clone+0x43)[0x7f7ffd061293]
```

sthalik, 2021-09-26:
Compile with debug info and run under `valgrind --leak-check=none`.

StanberyTrask, 2021-09-26:
```
valgrind --leak-check=no ./opentrack

==535174== Memcheck, a memory error detector
==535174== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==535174== Using Valgrind-3.15.0 and LibVEX; rerun with -h for copyright info
==535174== Command: ./opentrack
==535174== 
qt5ct: using qt5ct plugin
==535174== Conditional jump or move depends on uninitialised value(s)
==535174==    at 0xBF6532C: ??? (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.32)
==535174==    by 0xC43E801: g_closure_invoke (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.6400.6)
==535174==    by 0xC452813: ??? (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.6400.6)
==535174==    by 0xC45DBBD: g_signal_emit_valist (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.6400.6)
==535174==    by 0xC45EB9B: g_signal_emit_by_name (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.6400.6)
==535174==    by 0xC446E75: g_object_set_valist (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.6400.6)
==535174==    by 0xC447763: g_object_set (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.6400.6)
==535174==    by 0xBE5BB89: ??? (in /usr/lib/x86_64-linux-gnu/qt5/plugins/styles/libqgtk2style.so)
==535174==    by 0xBE5CF92: ??? (in /usr/lib/x86_64-linux-gnu/qt5/plugins/styles/libqgtk2style.so)
==535174==    by 0x4BAAA79: QStyleFactory::create(QString const&) (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.12.8)
==535174==    by 0x4BD215F: QProxyStyle::QProxyStyle(QString const&) (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.12.8)
==535174==    by 0xBE2C40E: Qt5CTProxyStyle::Qt5CTProxyStyle(QString const&) (in /usr/lib/x86_64-linux-gnu/qt5/plugins/styles/libqt5ct-style.so)
==535174== 
==535174== Conditional jump or move depends on uninitialised value(s)
==535174==    at 0x6A62F04: g_utf8_offset_to_pointer (in /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.6400.6)
==535174==    by 0xBF68B83: ??? (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.32)
==535174==    by 0xBF692A3: ??? (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.32)
==535174==    by 0xBF695E3: ??? (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.32)
==535174==    by 0xC43E801: g_closure_invoke (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.6400.6)
==535174==    by 0xC452961: ??? (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.6400.6)
==535174==    by 0xC45DBBD: g_signal_emit_valist (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.6400.6)
==535174==    by 0xC45E0F2: g_signal_emit (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.6400.6)
==535174==    by 0xC0FD69E: gtk_widget_realize (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.32)
==535174==    by 0xC0FEA47: gtk_widget_set_parent (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.32)
==535174==    by 0xBF97EFA: gtk_fixed_put (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.32)
==535174==    by 0xC43E801: g_closure_invoke (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.6400.6)
==535174== 
==535174== Conditional jump or move depends on uninitialised value(s)
==535174==    at 0x6A62F43: g_utf8_offset_to_pointer (in /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.6400.6)
==535174==    by 0xBF68B83: ??? (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.32)
==535174==    by 0xBF692A3: ??? (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.32)
==535174==    by 0xBF695E3: ??? (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.32)
==535174==    by 0xC43E801: g_closure_invoke (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.6400.6)
==535174==    by 0xC452961: ??? (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.6400.6)
==535174==    by 0xC45DBBD: g_signal_emit_valist (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.6400.6)
==535174==    by 0xC45E0F2: g_signal_emit (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.6400.6)
==535174==    by 0xC0FD69E: gtk_widget_realize (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.32)
==535174==    by 0xC0FEA47: gtk_widget_set_parent (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.32)
==535174==    by 0xBF97EFA: gtk_fixed_put (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.32)
==535174==    by 0xC43E801: g_closure_invoke (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.6400.6)
==535174== 
==535174== Conditional jump or move depends on uninitialised value(s)
==535174==    at 0x51D1601: QImageData::checkForAlphaPixels() const (in /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5.12.8)
==535174==    by 0x521A197: QRasterPlatformPixmap::createPixmapForImage(QImage, QFlags<Qt::ImageConversionFlag>) (in /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5.12.8)
==535174==    by 0x521A35D: QRasterPlatformPixmap::fromImage(QImage const&, QFlags<Qt::ImageConversionFlag>) (in /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5.12.8)
==535174==    by 0x5211252: QPixmap::fromImage(QImage const&, QFlags<Qt::ImageConversionFlag>) (in /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5.12.8)
==535174==    by 0x535152A: QBrush::texture() const (in /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5.12.8)
==535174==    by 0x4B6ED53: QWidgetPrivate::paintBackground(QPainter*, QRegion const&, int) const (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.12.8)
==535174==    by 0x4B72BB8: QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.12.8)
[... cut ...]
==535174==     in use at exit: 17,895,521 bytes in 130,801 blocks
==535174==   total heap usage: 377,031 allocs, 246,230 frees, 204,896,814 bytes allocated
==535174== 
==535174== For a detailed leak analysis, rerun with: --leak-check=full
==535174== 
==535174== Use --track-origins=yes to see where uninitialised values come from
==535174== For lists of detected and suppressed errors, rerun with: -s
==535174== ERROR SUMMARY: 20680 errors from 13 contexts (suppressed: 2 from 2)
Segmentation fault (core dumped)
```

sthalik, 2021-09-26:
Do you get the same crash when opening the mapping window?

This doesn't make any sense. `fmax` isn't something that can do an out-of-bound memory read. Is there something unusual about the compiler you're using, or the `glibc` version?

StanberyTrask, 2021-09-26:
both gcc and glibc seem to be up to date
```
gcc --version && ldd --version
gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

ldd (Ubuntu GLIBC 2.31-0ubuntu9.2) 2.31
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Written by Roland McGrath and Ulrich Drepper.
```
``cc --version`` also returns the same info as ``gcc --version``
and i have no real experience with compiler errors like this so i have no idea if any of this is "wrong"

sthalik, 2021-09-29:
Are you passing `-march=cpu` or `-msse`/`-mavx` in CFLAGS? Does using simply `-O2` help?

StanberyTrask, 2021-09-30:
no special flags during the build, just ``cmake -DCMAKE_BUILD_TYPE=RelWithDebInfo . && make -j6 && make install -j6``. Could you run me through how to get disassemble and register info like you want? I'm not well versed in debugging this far.

sthalik, 2021-09-30:
When it crashes under `gdb`, `bt` should show `fmax` as `frame \#0`. Then type `disassemble` and `info registers` as two separate commands and send the output.

StanberyTrask, 2021-09-30:
including the bt results because now there's no mention of fmax that i can see
```
(gdb) bt
#0  spline_detail::spline::get_value_internal(int) const
    (this=this@entry=0x5555558a75d0, x=x@entry=-2147483648)
    at /home/stanbery/Downloads/opentrack-opentrack-2021.1.2/spline/spline.cpp:104
#1  0x00007ffff7f1c478 in spline_detail::spline::get_value_no_save(double) const (this=0x5555558a75d0, x=x@entry=inf)
    at /home/stanbery/Downloads/opentrack-opentrack-2021.1.2/spline/spline.cpp:85
#2  0x00007ffff218f8cc in accela::<lambda(double)>::operator()
    (__closure=<optimized out>, x=inf)
    at /home/stanbery/Downloads/opentrack-opentrack-2021.1.2/filter-accela/ftnoir_filter_accela.cpp:132
#3  do_deltas<accela::filter(double const*, double*)::<lambda(double)> >(const double *, double *, accela::<lambda(double)> &&)
    (deltas=deltas@entry=0x5555558a77a8, output=output@entry=0x7fff915a1b40, fun=...)
    at /home/stanbery/Downloads/opentrack-opentrack-2021.1.2/filter-accela/ftnoir_filter_accela.cpp:36
#4  0x00007ffff218fc79 in accela::filter(double const*, double*)
    (this=0x5555558a7530, input=0x7fff915a1b78, output=0x7fff915a1b40)
    at /home/stanbery/Downloads/opentrack-opentrack-2021.1.2/filter-accela/ftnoir_filter_accela.cpp:131
--Type <RET> for more, q to quit, c to continue without paging--
::Mat<double, 6, 1> const&) const (this=this@entry=0x5555563adb50, value=...) at /home/stanbery/Downloads/opentrack-opentrack-2021.1.2/api/../compat/simple-mat.hpp:255
#6  0x00007ffff7f40153 in pipeline_impl::pipeline::logic() (this=0x5555563adb50) at /home/stanbery/Downloads/opentrack-opentrack-2021.1.2/logic/pipeline.cpp:463
#7  0x00007ffff7f4072b in pipeline_impl::pipeline::run() (this=0x5555563adb50) at /home/stanbery/Downloads/opentrack-opentrack-2021.1.2/logic/pipeline.cpp:595
#8  0x00007ffff6d419d2 in  () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#9  0x00007ffff65aa609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#10 0x00007ffff6865293 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
(gdb) disassemble
Dump of assembler code for function spline_detail::spline::get_value_internal(int) const:
   0x00007ffff7f1c060 <+0>:	endbr64 
   0x00007ffff7f1c064 <+4>:	sub    $0x18,%rsp
   0x00007ffff7f1c068 <+8>:	movss  0x78e8(%rip),%xmm2        # 0x7ffff7f23958
   0x00007ffff7f1c070 <+16>:	test   %esi,%esi
   0x00007ffff7f1c072 <+18>:	js     0x7ffff7f1c07c <spline_detail::spline::get_value_internal(int) const+28>
   0x00007ffff7f1c074 <+20>:	movss  0x78e0(%rip),%xmm2        # 0x7ffff7f2395c
   0x00007ffff7f1c07c <+28>:	mov    %esi,%eax
   0x00007ffff7f1c07e <+30>:	movss  %xmm2,0xc(%rsp)
   0x00007ffff7f1c084 <+36>:	pxor   %xmm1,%xmm1
   0x00007ffff7f1c088 <+40>:	sar    $0x1f,%eax
   0x00007ffff7f1c08b <+43>:	xor    %eax,%esi
   0x00007ffff7f1c08d <+45>:	sub    %eax,%esi
   0x00007ffff7f1c08f <+47>:	mov    $0x1fff,%eax
   0x00007ffff7f1c094 <+52>:	cmp    $0x1fff,%esi
   0x00007ffff7f1c09a <+58>:	cmovg  %eax,%esi
   0x00007ffff7f1c09d <+61>:	mov    0x68(%rdi),%rax
   0x00007ffff7f1c0a1 <+65>:	movslq %esi,%rsi
=> 0x00007ffff7f1c0a4 <+68>:	movss  (%rax,%rsi,4),%xmm0
   0x00007ffff7f1c0a9 <+73>:	callq  *0xc951(%rip)        # 0x7ffff7f28a00
   0x00007ffff7f1c0af <+79>:	movss  0x78a9(%rip),%xmm1        # 0x7ffff7f23960
   0x00007ffff7f1c0b7 <+87>:	callq  *0xce4b(%rip)        # 0x7ffff7f28f08
   0x00007ffff7f1c0bd <+93>:	movss  0xc(%rsp),%xmm2
   0x00007ffff7f1c0c3 <+99>:	add    $0x18,%rsp
   0x00007ffff7f1c0c7 <+103>:	mulss  %xmm0,%xmm2
   0x00007ffff7f1c0cb <+107>:	pxor   %xmm0,%xmm0
   0x00007ffff7f1c0cf <+111>:	cvtss2sd %xmm2,%xmm0
   0x00007ffff7f1c0d3 <+115>:	retq   
End of assembler dump.
(gdb) info registers
rax            0x5555563c3cf0      93825007369456
rbx            0x80000000          2147483648
rcx            0x5555563b5310      93825007309584
rdx            0x0                 0
rsi            0xffffffff80000000  -2147483648
rdi            0x5555558a75d0      93824995718608
rbp            0x5555558a75d0      0x5555558a75d0
rsp            0x7fff915a1980      0x7fff915a1980
r8             0xb                 11
r9             0x5555558a77e0      93824995719136
r10            0x5555558a77d8      93824995719128
r11            0xc2b2f             797487
r12            0x5555558a75e9      93824995718633
r13            0x1                 1
r14            0x5555558a77d8      93824995719128
r15            0x7fff915a1ba8      140735631989672
rip            0x7ffff7f1c0a4      0x7ffff7f1c0a4 <spline_detail::spline::get_value_internal(int) const+68>
eflags         0x10a12             [ AF IF OF RF ]
cs             0x33                51
ss             0x2b                43
ds             0x0                 0
es             0x0                 0
fs             0x0                 0
gs             0x0                 0
```

sthalik, 2022-09-30:
Can you confirm this with  [opentrack 2022.3.0](https://github.com/opentrack/opentrack/releases/tag/opentrack-2022.3.0)?

closed 2024-06-07 by sthalik