opentrack/opentrack #1346: could I use some of code and models in my commercial software?

frankzhang196, 2021-10-04:
If could, what do I need to do?

sthalik, 2021-10-04:
You need to comply with the ISC license. That is, list the contributors and the ISC license's copyright statement. There are also libraries opentrack depends upon, see `3rdparty-notices/`.

See: <<https://github.com/opentrack/opentrack/blob/master/OPENTRACK-LICENSING.txt>> but this list is out of date and you need to extract additional copyright statements from the source tree.

DaWelter, 2021-10-04:
The pose estimation model (head-pose.onnx) of the neural network tracker is a no-go I guess since it was created using some stuff which prohibits commercial use.

See https://github.com/opentrack/neuralnet-tracker-traincode/blob/master/readme.md. 

@sthalik , I should really finish the licensing section on datasets. Sorry about that. I postponed it because I could not find the information for some. However I did train on a derivative work of https://www.tugraz.at/institute/icg/research/team-bischof/lrs/downloads/aflw/, which limits use to non-commercial research purposes.

DaWelter, 2021-10-04:
@sthalik I completed the docu to the best of my abilities (not being a lawyer and such ;-) )

frankzhang196, 2021-10-05:
thanks for your reply!  @sthalik @DaWelter