opentrack/opentrack #1349: Z axis problem

Anjelu03, 2021-10-10:
I am not an expert so i won't be very specific but i'll try to be as clear as possible:
I have a problem with neuralnet tracker: z value goes from -75 to 75 even if I don't move. How can i solve it?

Anjelu03, 2021-10-19:
Idk. I use it for dcs and I decided to disable z axis and set an axis on my hotas for zoom. It's not a solution but it allows me to play

TheGrimReaper13, 2021-12-06:
any change this will be fixed in the near future?

jemackay, 2022-01-17:
I am having the same problem as described by @KaiKruschel. Is the cause of the problem known and just needs a fix or perhaps the fix is tricky? Not seeing much discussion on this. Can I help at all?

sthalik, 2022-01-17:
@DaWelter this seems legit.

kjerandp, 2022-05-08:
There seems to be a workaround for this issue.

Assign a key to re-center. After re-centering, the Z-axis will work. At least it did for me.

mjcarter, 2022-06-07:
Also had this issue, another workaround is to use the Kalman filter

PieterKuyperAlbertyn, 2022-07-21:
I found a fix that worked for me. Go to mapping -> z,  then change the max input and max output to 300 or 600 cm

XenoPL, 2022-08-16:
I've tried NeuralNet tracker in 2022.03.0. Definitely it doesn't like Accela filter, when face is around 80cm from camera it goes crazy  zoom loop, it seems Kalman filter is the best option for this tracker. It handles loosing track way better and a bit further away.
Another thing is it still doesn't center properly on start, z-axis is zoomed despite of proper calibration. Recentering by hand solves this problem.
On the sidenote key capture on Plasma-wayland session does not work. You can bind key to the command, but even if Opentrack window has focus, keypresses does nothing.

DaWelter, 2022-08-28:
Crazy zoom loop when you are only 80cm from the camera sounds like there is something wrong. Try playing with the ROI Zoom in the tracker settings.

dawidrogowicz, 2022-09-05:
> I've tried NeuralNet tracker in 2022.03.0. Definitely it doesn't like Accela filter, when face is around 80cm from camera it goes crazy zoom loop, it seems Kalman filter is the best option for this [...]

Thanks! that turned out to be the issue for me as well, kalman filter works great after centering it

DaWelter, 2022-09-07:
Strange. I can reproduce it. Will take a look @sthalik . Recentering fixes it for me though. Hence I never noticed ...

Pieloth, 2022-10-01:
IMHO
Neuralnet + Accela issue seems to be due to 2 reasons:

1. Neuralnet Tracker Z values

As far as I can see, Neuralnet Tracker provides too high Z-axis values, compared to reality
As @XenoPL states, and in my case also, at 80cm from the Webcam, the Z value is estimated to something around -160cm to -180cm, which appears to be way too high. 
When Z reaches around -180cm, then Z goes crazy in Accela
Note that calibration adds also around -17cm to the Z value, for the face-to-neck translation 
Note also that reducing a bit the ROI zoom in Neuralnet settings, to have my face totally covered by the white circle (0.85 gives better stability), will make also the Z value even higher 

Now, not "real" distances is not really an issue, as we then "map" those sensor values to real angles and pos, and that's why other filters than Accela can work fine 

So what happens with Accela?

2. Accela filter

Browsing through the Accela code, I suspect the Z-axis high values around 180 of Neuralnet, will hit the following code in filter-accela/ftnoir_filter_accela.cpp, starting line 153:

```
    for (unsigned k = 0; k < 6; k++)     <-- All 6 DOF parameters are processed within this loop
    {
        output[k] *= dt;
        output[k] += last_output[k];
        if (fabs(output[k]) > half_turn) output[k] -= copysign(full_turn, output[k]);    <-- This line Ok for angles, but KO for pos

        last_output[k] = output[k];
    }

```

Here, all 6 DOF params, 3 pos + 3 angles, are processed the same way, and a 180/360 correction is applied to all of them, although I think it should be applied to **only** the 3 angles, and **not** to the pos values

A way I found to circumvent this Z=-180cm "limit" issue, is to "cheat" a little bit with the Neuralnet FOV value
Increasing the FOV value will decrease the Z values, not hitting the 180cm barrier
Of course, this is just a turnround, not a perfect solution

sthalik, 2022-10-05:
See if this build help: \<https://www.dropbox.com/s/ertw1uqquysrqd7/opentrack-test-20221005_01.7z>

Pieloth, 2022-10-05:
Thanks @sthalik for this test build
As far as I have tested:
- The Z issue with Accela/Neuralnet is fixed! 
the Z values now varies correctly according to mapping, no more sudden changes at 180
-  But there's now a **new issue with Roll angle output**, issue which is not in 2022.3.0
   When my head is around 0 deg roll angle (standby position), Neuralnet outputs Roll angles around +180 or -180 deg
   * With 2022.3.0, the transition output between +180 to -180 and vice versa is immediate, so the output is continuous and Ok
   * But with this test build, the output roll angle smoothly varies from +180 to -180 and vice versa during about 2 seconds, so the Roll movement is not natural
   This can be seen easily with the Octopus

sthalik, 2022-10-06:
Alright, see if \<https://www.dropbox.com/s/lj404brrb4wbzzs/opentrack-test-20221006_01.7z> works.

Pieloth, 2022-10-06:
All 6 DOF now working fine with Neuralnet Tracker + Accela filter, with this last build "opentrack-test-20221006_01" , as far as I've tested on my side

I noticed however another issue with the test builds: 
In the Neuralnet Tracker settings window, the camera choice is not working anymore 
I have 2 webcams, but the camera name is not saved, if I choose the other camera available 
Confirmed in the .ini profile file, in the [neuralnet-tracker] section, the camera name field stays unchanged whatever camera is choosen
Note the camera choice works fine in 2022.3.0, which updates also correctly the camera name in the .ini profile

Thanks again @sthalik for your help!

Pieloth, 2022-10-06:
Thanks @sthalik for this next update, I will be able to test it on sunday
Will keep you posted...

Pieloth, 2022-10-09:
I can confirm Webcam selection in Neuralnet tracker settings window now works fine with the latest build: opentrack-test-20221006_02
AFAIK, all issues are now fixed on Neuralnet tracker + Accela filter 
Thanks @sthalik for the great job!

CappuccinoRU, 2023-01-19:
Unfortunately, the problem with the Z axis is still relevant (
There is no smooth change in the parameter along the Z axis, or the parameter is set to 75, or -75

opentrack 2022.3.2 Latest on Oct 19, 2022

Pieloth, 2023-01-19:
Strange it works perfectly on my side

Do you see this on the Raw tracker Data with Neuralnet, or on the Game data?
What are you Neuralnet tracker settings (FOV, resolution, Head center offset?
Did you try to activate "Show network input" to see if everything is smooth on the image feedback?

CappuccinoRU, 2023-01-19:
> Strange it works perfectly on my side
> 
> Do you see this on the Raw tracker Data with Neuralnet, or on the Game data? What are you Neuralnet tracker settings (FOV, resolution, Head center offset? Did you try to activate "Show network input" to see if everything is smooth on the image feedback?

I'm sorry, I played a little with the settings, everything seems to fall into place.
Before that, I just used AITrack,