opentrack/opentrack #1353: Hotkey Issue With An Xbox One Gamepad

MinamiHasaki, 2021-10-15:
When using buttons on an Xbox One gamepad as hotkeys in OpenTrack, they don't register if the OpenTrack window isn't focused. This is happening even though OpenTrack is set to "run this program as an administrator",

DrFeelgood66, 2021-11-04:
I'm having the same problem with my Logitech G910 Macro Keys

sthalik, 2021-11-04:
> Xbox One gamepad

opentrack doesn't support xinput hotkeys at this moment.

> Logitech G910 Macro Keys

That's not the same problem. Anyway, what are the `VK_*` constants?

DrFeelgood66, 2021-11-04:
> > Logitech G910 Macro Keys
> 
> That's not the same problem. Anyway, what are the `VK_*` constants?

What do you mean by VK_* constants?

645285985, 2022-05-26:
> > Xbox One gamepad
> 
> opentrack doesn't support xinput hotkeys at this moment.

any work around?

sthalik, 2024-03-03:
Would it be sufficient to implement it using QtGamepad?

645285985, 2024-03-03:
I have since began to use reWASD to bridge support. if QtG is similar, i would see no issue.

sthalik, 2024-06-04:
We could probably use this one https://github.com/pumphaus/qtgamepadlegacy after the migration to Qt 6 is done. Using it on Linux with the evdev backend isn't a full solution, as it doesn't support an arbitrary amount of keys and axes that HOTAS typically possess. But for Windows it'll do.

sthalik, 2025-07-31:
Implemented in 5029dce. You must install Microsoft's GameInput library though.

closed 2025-07-31 by sthalik