opentrack/opentrack #1366: Steam VR Yaw pitch interference problem

hanmike, 2021-11-04:
There is no problem even if you turn on each function separately or activate the roll.
but moment yaw and pitch are turned on together, there is a problem that the yaw value follows the roll value.
i'm using htc vive tracker 2.0

This is a video of testing the yaw and roll movements.
https://youtu.be/hI0x3yUKEVU
yaw move(1, 2)
roll move(3, 4)
turned on together(1, 4)

Is there any way to solve it?

donoli1st, 2021-12-27:
Almost all my problems went away after i choose "Point" as "Center Method"
Maybe it help's

tsvetoslav95, 2024-03-11:
I see this is years old but if someone comes here googling for help here is a hint what to google next:
extrinsic vs intrinsic yaw pitch roll