opentrack/opentrack #1367: Command line syntax for auto loading a profile - or a solution for MS Flight Sim 2020

JBlyth, 2021-11-06:
Back in 2015 a user asked for "Support command line options for autostart #159" to start OpenTracker with a given profile" - I presume from the shortcut icon. I think the suggested usage was --autostart <profile>

Did this ever get implemented as I can seem to get it to work? If it was implemented, what is the correct syntax?

I ask because I can't get Game Detector to work with Microsoft Flight Simulator 2020, as it doesn't seem to have an .exe that functions with Game Detector

sthalik, 2021-11-06:
It hasn't been implemented. Can you tell me more about the .exe thing? The detector lists all processes running on the system.

JBlyth, 2021-11-07:
Thanks Stanislaw. I have used the Game picker and chosen C:/Program Files/WindowsApps/Microsoft.FlightSimulator_1.20.6.0_x64__8wekyb3d8bbwe/FlightSimulator.exe, which is the correct exe, but it doesn't seem to trigger a change when starting Flight Sim or JoystickGremin.

It DOES work as expected with Condor 2, so I am not sure if it is because Flight Sim 2020 is a Microsoft Store download that is causing the problem.

I'm very happy to try any suggestions!

Cheers
John

sthalik, 2021-11-07:
Run Task Manager and get the correct process name that way. It's the one that uses all the CPU.

JBlyth, 2021-11-08:
I have done as requested ;-) My "usual" settings are:-
C:/Program Files/WindowsApps/Microsoft.FlightSimulator_1.20.6.0_x64__8wekyb3d8bbwe/FlightSimulator.exe
C:/Users/xxxx/OneDrive/Documents/Thrustmaster/joystick gremlin/FS2020 3.xml

I have also tried:- Microsoft Flight Simulator - 1.20.6.0, but that doesn't work either I'm afraid.

Here's a screen shot of Task Manager....
[image]

MeaningOfLights, 2022-02-20:
Hi Stanislaw, can you please add support for the AutoStart CLI argument? There's thousands of people who start up Flight Sim with half a dozen plugins. This is the only app that can't be started automatically and since SendKeys doesn't work on Win10 the options appear limited. Big fan of your work, thanks!

sthalik, 2022-02-20:
Alright, fine. This is going to be done through shared memory.