opentrack/opentrack #1371: Opentrack Microsoft Defender virus warning

0redge, 2021-11-08:
Hi!

Got a warning when I started opentrack today warning me about a trojan called "Trojan:Script/Phonzy.A!ml".
The affected file is C:\Program Files (x86)\opentrack\modules\opentrack-tracker-freepie-udp.dll

Just wondering if it's an actual virus or a false positive.

tepa74, 2021-11-08:
Got same warning.

sthalik, 2021-11-08:
False positive.

DrFeelgood66, 2021-11-10:
The same here, but it is a False positive warning.

Len62, 2021-11-10:
Seeing the same here, I kind of figured it was a false positive.

tenevas666, 2021-11-11:
Hello les mecs....idem, défender me signale sur W11 que Trojan dans le dernier open track, vous etes certain faux positif ?

T3NGHU, 2021-11-14:
Just installed the 2021.1.13 on windows 11 and windows defender shows a warning for the trojan called "Trojan:Script/Phonzy.A!ml". Most probably a false positive but very annoying as WDefender never stops warning for the virus.

tenevas666, 2021-11-15:
Bonjour.  C’est donc un faux positif ! ?

Len62, 2021-11-15:
I overwrote the offending file with the one from version 2.3.12. Stopped the warnings.

sthalik, 2021-11-19:
There's nothing I can really do about it. But replacing the file won't solve anything long-term. It's better to wait for Windows Defender to stop blacklisting the file. Just report is as a false positive and wait... wait a lot.

tenevas666, 2021-11-19:
Yes.    Ok.   Merci beaucoup

arcadien, 2021-11-19:
note: Avast also detect a virus in the same file

sthalik, 2021-11-20:
I'm going to get back to building opentrack with Visual Studio Preview 2022 as apparently building it with Visual Studio 2019 is the problem.

And it's really a false positive. The source file has nothing interesting to an antivirus in it, compared to any other opentrack module.

https://github.com/opentrack/opentrack/blob/master/tracker-freepie-udp/ftnoir_tracker_freepie-udp.cpp

sthalik, 2021-11-20:
I think it's done in the new release. Please confirm that nothing is seriously broken so that I may mark it as a stable release. There are barely any changes from the previous version.

GO63-samara, 2021-11-20:
I'm at a loss. opentrack 2021.2.0 is in the source, but not in the releases. Why?

sthalik, 2021-11-21:
@GO63-samara What do you mean?

GO63-samara, 2021-11-21:
OT2021.2.0 was not on the releases page yesterday. Everything is in order today. Thanks!

Len62, 2021-11-28:
@sthalik Started getting another positive hit for a trojan with opentrack-tracker-freepie-udp.dll. This is with the current release, 2021.2.0. It was fine until yesterday. Probably after a defender update and still false I assume.

CdtSylvestre, 2021-11-28:
Ditto here. Opentrack-tracker-freepie-udp.dll from version2021.2.0 downloaded from github is now flagged as a Trojan:Win32/Wacatac.B!ml (threat level: severe).

sthalik, 2021-11-29:
I'm not getting anything. See: <<https://www.virustotal.com/gui/file/b75dfa21e53d2c821693ea9750b92879286c4ac4ac5d358e1a2580dfac269415?nocache=1>>.

Len62, 2021-11-29:
Yeah, this was strange as initially it was OK. I'll delete and reinstall OT just to see what happens. This is on Win 11, not that it should matter.

CdtSylvestre, 2021-12-05:
For the record, re-downloaded 2021.2.0 today (the only one that was ever installed on that PC). Microsoft Defender does not flag Opentrack-tracker-freepie-udp.dll as a virus anymore.

All is well.

Slion, 2022-04-26:
Won't fix as we certainly won't bother doing a re-release of 2021.1.3 and other releases are not affected.