opentrack/opentrack #1376: Opentrack gives uncontrollable output

pizzapizze, 2021-11-11:
Suddenly after everything worked fine for a long time, opentrack outputs uncontrollable wobbly muddle., like in the video. 
I tried to change the Filter, the input, the output and tried to wait a few minutes for the filter to become effective. Nothing helped.
And I reinstalled Opentrack ofcourse.

https://user-images.githubusercontent.com/62241655/141353733-3b237706-b15a-4808-8572-b1a2987a9a97.mp4

SirBalth, 2021-12-18:
I have a very similar issue.

dyerelog, 2021-12-18:
Same problem, any solution ?

eggalatarian, 2021-12-19:
Quit open track, go to task manager (Win-R, taskmgr.exe) and look for "Dummy TrackIR Executable"  and/or "opentrack", then right click and "end task" them.

dyerelog, 2021-12-19:
Do not change anything