opentrack/opentrack #1378: PS3Eye Freeze: "ps3eye: transfer status 4"

dssrtfx, 2021-11-15:
Hello! I'm not sure what happened, maybe a Windows 10 update? But Opentrack Point Tracker with the PS3 Eye open driver (previously working for a month or so with the "libusb-win32", please see #1325 why I don't use libusbK) freezes a few seconds after starting tracking.

When I run opentrack from the command line, each freeze coincides with "ps3eye: transfer status 4" is written to the console.

I have tried using zadig to install libusbK, tried unplugging/replugging, checked that power management Windows sleep is turned off for all USB devices, and reinstalling libusb-win32 using zadig.

Looking through #1337, it may not be the same problem.

dssrtfx, 2021-11-21:
I'm completely open to it being my fault somehow, I just don't know where to decipher "transfer status 4," I wasn't able to find anything on Google.

sthalik, 2021-11-21:
Did you try replugging to a different USB slot?

dssrtfx, 2021-11-30:
I did not until now. Now it works. Thank you for pushing me to try it.
I didn't try another port until now because the camera worked before for, heck, years (even with the new open driver!), so why would it stop working now?

closed 2021-11-30 by dssrtfx