opentrack/opentrack #1380: PS3 Eye Red LED blink

deeph46, 2021-11-19:
Hi,

Have my opentrack worked great!
But I'm having issue with my PS3 Eye's red LED keep blink (not blinking, but blink 1 time) randomly. When it's blink, it also re-center the picture. 

I don't where this issue come from.

deeph

sthalik, 2021-11-19:
Are you using the open driver?

deeph46, 2021-11-19:
No, don't know exactly if I have installed open driver yet.

deeph46, 2021-11-19:
My bad, just didn't read the note in release page.
Have installed libusb-win32, but there is Interface 0 and Interface 1?. Noted that choose Interface 0. I have installed in Interface 0.

sthalik, 2021-11-19:
Run opentrack in a console window:

1. Press Win+R, type "cmd", press Enter
2. Drag `opentrack.exe` onto that window, then press Enter

And see if any text spews out when the LED blinks. If it does, make a screenshot (select console window, press alt+printscreen), then ctrl+v the screenshot into the text box on this issue.

deeph46, 2021-11-19:
OK, 1 hour driving sim has solid Red LED light on. I uninstalled CL Eye driver and using per your instruction in release page. It looks like it runs smoothly now. If it's run not smooth again, I will back to this thread.

Thanks for your app, I'm late to this tracking stuff, but your app doing great for simulator, thumbsup

deeph46, 2021-11-19:
Now, just have another problem. PS3Eye not detected. There was a sound of USB disconnecting and connected. In device manager the camera detected also in libusb-win32, but I can't choose PS3Eye anymore in device manager.

GO63-samara, 2021-11-19:
You need to select the `PS3 Eye open driver` not in the device manager, but in the `PointTracker Settings` (Device):

[image]

sthalik, 2021-11-20:
> In device manager the camera detected also in libusb-win32, but I can't choose PS3Eye anymore in device manager.

It's normal, you installed libusb onto the "interface 0", so there's no driver for that part of the composite USB device in "Imaging devices" anymore.

deeph46, 2021-11-20:
@GO63-samara, yes, I cross check if there is P3Eye open driver in there. Doesn't it automatically show PS3 Eye in the drop down?
@sthalik, I ended up updating interface 0 and interface 1 into libusb-win32. 

So, I was fiddling the device manager uninstalling driver, install driver, reboot PC, etc, about 3 hours. And in the end I have working PS3 Eye again more solid I think, no USB disconnected or connected sound again till now.

This is what I did :
  1. Uninstalled all the drivers from my PS3 Eye.
  2. Install libusb-win32.
  3. Right-click the Camera device.
  4. Click update.
  5. Browse my computer drivers.
  6. Let me pick from a list of available drivers on my computer.
  7. Last windows, I saw PS3 Eye driver, I clicked this.
  8. And PS3 Eye showed up in opentrack app.

Though, after I Reboot my PC. I didn't see PS3 Eye as Imaging Device (like after my steps above it pop up as Imaging Device) anymore in Device Manager but in opentrack I still can use it as a PS3 Eye open driver..

sthalik, 2021-11-20:
Interface 1 is the microphone. There's absolutely no need to use libusb for this. opentrack doesn't spy on you through the microphone or anything.

By the way, after rebooting I must always replug the ps3 eye to get the microphone to work. On Windows and Linux. It's always been like this.

@deeph46 are you experiencing anymore problems? Feel free to give it time, issues might not be immediately apparent.

GO63-samara, 2021-11-20:
> By the way, after rebooting I must always replug the ps3 eye to get the microphone to work. On Windows and Linux. It's always been like this.

This is very useful information, thanks! I did not know. I've always had microphone problems on PS3.

MathijsG, 2021-11-21:
> By the way, after rebooting I must always replug the ps3 eye to get the microphone to work. On Windows and Linux. It's always been like this.

You mean with the Open Driver? With the CL-Eye driver I never need to replug anything. And I use the microphone of the PS3 eye daily.

Len62, 2021-12-05:
@sthalik Regarding the LED blink, I am seeing it both with libusbK and libusb-win32 on Windows 11. Below is the output from the cmd window. Not a big deal, just something I've been meaning to pass on to you.

[image]

deeph46, 2021-12-06:
@sthalik, I have problem again with PS3 Eye not detecting in Opentrack, but I have fixed it by uninstalling the PS3 eye driver until I didn't see checkbox 'Delete the driver software this driver' option when uninstalling the driver. After that, I reinstalled it with Zadig. I don't know clear where/what makes the Opentracker don't see the PS3Eye Open driver.

sthalik, 2021-12-07:
@deeph46 when the driver is installed with zadig, you can see 'ps3eye open driver' on the list of available cameras with the PT. Confirm that it's installed to the right device through 'scan for hardware changes':

[image]

deeph46, 2021-12-08:
yes, that ID the same with mine. Still working fine today.