opentrack/opentrack #1382: Can't open Camera. PS3 eye driver error: no camera found

Iknowlongnamerightwhatwasithinking, 2021-11-22:
I have new windows7 install, no CL Eye driver installed.
 It tries to find the camera but comes back with errors. 

"Can't open Camera.
PS3 eye driver error: no camera found. "

so yeah, back to using the CL Eye Driver... which works fine. 
No problem finding the camera. 

Is there some FAQ on the open driver ?

sthalik, 2021-11-22:
It's in the release notes.

Iknowlongnamerightwhatwasithinking, 2021-11-23:
sorry for being a pest but where are the "release notes".

Iknowlongnamerightwhatwasithinking, 2021-11-23:
Oh yes I rememebr now the little dance I have to do. I think I did suggest you make a README on how to do this that is super obvious to find a while back. Like in the program directory. 

Something like this:

Download Zadig.exe  from  https://zadig.akeo.ie/
Run the program.
Click Options/List All Devices, check "list all devices". 
In the top window drop down menu find "USB Camera blah blah blah (Interface 0) etc etc "

In the right hand window, select the driver you want to install, from the up/down menu  find "LibUSBK". 
Click the big "install Driver" button. 

( If you have a driver already installed that button will read "replace driver"  and the driver  should be displayed in the left window. You will replace that one, ok. ) 

NOTE:
The left window called "Driver" is the currently installed driver !!!
If it is blank, there is no current driver installed.  Woo! 
After you install/replace the LibUSBK driver, the left window will update and show you which driver is currently installed.  Result! 

This confused the heck out of me the first time, oh many months ago. and then I forgot how to do this then I had to relearn with some help. It works now! Horray!

closed 2021-11-23 by sthalik

Iknowlongnamerightwhatwasithinking, 2021-11-23:
So where are the release notes ???

sthalik, 2021-11-23:
https://github.com/opentrack/opentrack/releases/tag/opentrack-2021.1.3

Iknowlongnamerightwhatwasithinking, 2021-11-23:
So the information a new user is going to need is buried in the old release pages? 
Please tell me you are going to add the "how to install the ps3 eye device driver.txt " in the installed docs folder, right? 
So when you install the program the information you may need is right there. Not right here, somewhere... 
A first time user of open track has NO IDEA where to find the information!  
Just saying.

sthalik, 2021-11-24:
You're right. For now I added the instructions to the project's wiki. The driver should be visible on the list of cameras only to people who already installed libusb anyway, so there's less confusion.

The install instructions are a mess, don't work for everyone but at least there are no crashes or blue screens.

TheDiviler, 2023-07-24:
> 

DUDE THANK YOUUUU. I was like on a couple hour search to make this camera work (I got the Ps3 Usb Camera-B4.04.27.1).
And you finally got me able to get the right driver lol. Thank you brotha