opentrack/opentrack #1383: PS3 Eye - access denied (insufficient permission)

emcek, 2021-11-22:
I update from 2.3.12 -> 2021.2.0 and when click start in opentrack I got:
**"PE3 Eye driver error: camera init failed: Access denied (insufficient permissions)"**
What I did:
* remove CL Eye driver (in Control Panel / Programs and Features)
* remove Opentrack 2.3.12 (in Control Panel / Programs and Features)
*  uninstall WinUSB driver for PS3 Eye (in Control Panel / Device Manager
* install the libusb-win32 driver onto interface 0 of the camera
* install OpenTrack 2021.2.0
* Reboot
* Re-plug camera to different port
* Start opentrack as administrator

Opentrack:
[image]

Device Manager:
[image]

Result:
[image]
[image]

emcek, 2021-11-25:
As mention in #1135 I try **libusbK** driver and it worked without re-plug camera to different port.
[image]

JoeWerts, 2021-12-03:
Can confirm, I had the same issue with libusb-win32 and installing libusbK fixed the problem.

dolmoso, 2021-12-08:
Same Issue! Same Fix it!
Thank now works fine and much more precision before with  older CL Eye driver.
Good work!

fbuchmeier, 2022-01-15:
I've had the same error "insufficient permissions" today with Opentrack 2022.1.0.

Environment:

Opentrack 2022.1.0
Windows 10 Pro 21H1
PS3 Eye Camera SLEH-00448 connected to a USB 3.0 port on my Laptop

What I did:

1. Install Opentrack 2022.1.0 (clean install)
2. Download Zadig 2.7 (2021.11.02)
3. Install libusb0 (v1.2.6.0)
4. Start Opentrack, select the Aruco Tracker with the PS3 Eye Open driver

The things I tried to fix this issue:

- start Opentrack as admin
- restart the OS

I then switched the driver from libusb0 (v1.2.6.0) to libusbK (v3.1.0.0). This worked in Opentrack, but was painfully slow (between 20 and 45fps and very erratic).

I then switched back from libusbK to libusb0 (without rebooting or even restarting Opentrack) and now Opentrack also works with libusb0 and a constant 75 frames per second.

Hope this helps,
Florian.

born2beflyin, 2022-03-08:
> I then switched the driver from libusb0 (v1.2.6.0) to libusbK (v3.1.0.0). This worked in Opentrack, but was painfully slow (between 20 and 45fps and very erratic).
>
> I then switched back from libusbK [...]

Same issue here. Confirm this worked for me.

emcek, 2022-03-08:
Maybe add all information to wiki page (and link it in README) and then close this issue?

sthalik, 2022-03-08:
How are these instructions? https://github.com/opentrack/opentrack/wiki/PS3-Eye-open-driver-instructions

lifeifbrian, 2022-03-21:
Non of the above worked but this driver from JoDaEmPa worked after 30 mins of playing with the settings in Opentrack.
https://github.com/opentrack/opentrack/issues/1443#issuecomment-1073392781

nitinvaid20, 2022-04-14:
Is there any proper steps by steps instructions as i also have lagging when using head tracking while playing ETS 2 then i came to know its the buggy ps3 camera software please let me know what needs to be done here
i did used Zadig and selected libusbK (v3.1.0.0) but it was more worse soo much lagg

LaurenceGough, 2022-05-17:
I have tried all of the above but nothing works. Multiple restarts, completely clearing the driver, different USB 2 ports directly on the motherboard. 
I have tried installing all driver types using Zadig, it installs fine but it gives the message Can't open Camera PS3 Eye driver error. Unknown error.

First I was getting the Access denied message (insufficient permission) after following the official instructions but I haven't seen that since trying to follow the above fixes.

emcek, 2022-05-17:
Looks like you tried everything... What can I suggest:
* try on different machine/computer - PS3 Eye open driver
* use older [opentrack 2.3.11](https://github.com/opentrack/opentrack/releases/tag/opentrack-2.3.11) lat release without PS3 Eye open driver
* try use CL Eye driver with current release of opentrack

LaurenceGough, 2022-05-17:
Thank you, I forgot to update on this issue. I tried an external USB 3 hub and it worked straight away. Beats me totally as I always thought using the built in motherboard USB ports were the best ones to use (USB 2 and USB3 I tried!).

Thank you anyway. Hopefully that will help others.

haze1986, 2022-05-30:
> 

Exactly as mentioned, thanks for sharing.

I realised I got it to work only when I kept opentrack open while changing from libusbK to libusb-win32 using zadig. I had opentrack closed thus it did not work initially.  As weird as it sounds.

Darkstar67, 2022-11-08:
With the prevalence of leisure flyers relying on this Opentrack stuff may I say I have had a total nightmare and none of the above legtimate but fruitless steps with any iteration of OT or Libsub versions worked to the point where after 3 hours I felt like chucking all the kit out the window...
then I found this-which is legacy I admit but it all fired up just fine after that, latest version of OT go into Camra Settings and select the old skool CL eye driver (after an easy test so you also know the cam isn't bust) and job done apart from curve bending.

https://delanengineering.com/manual/

And no I did not have to unplug and USBs or reboot either-just run OT as admin.

There may be newer drivers now...but dam if I can get them to work with a current tracker. so if you are having to start from scratch-start here. and Yes my Eye has an aftermarket IR filter for anyone wondering for the IR crystals on a head tracking clip.

sthalik, 2022-11-10:
I removed the old instructions that involved reinstalling the driver multiple times. The [new ones](https://github.com/opentrack/opentrack/wiki/PS3-Eye-open-driver-instructions) are more palatable.