opentrack/opentrack #1384: Opentrack octopus is moving but Star Citizen is not get any TrackIR output from it. Was once working

Zuffle, 2021-11-23:
Hello this is a great tool thanks for the awesome work. As stated in the subject, I previously had this working for maybe an hour in Star Citizen using my rift headset as a tracking source (Star Citizen does not have native VR support yet). But I don't really know why but it stopped while I was playing.

Ever since then I have not been able to get Star Citizen to recognize any motion data from opentrack. The octopus is moving and I have freetrack set to only output TrackIR as that is how I was able to get it working the first time. 

I downloaded the debug info archive but I have no idea how to use it. I think somehow there is something getting in the way of the data flowing from opentrack to the game and I am not sure which side it is on. I am inclined to believe it is on opentrack's side sicne it was working beautifully for a little while in Star Citizen.

Any help would be appreciated!

poboy975, 2021-11-27:
I'm interested in this as well. I've been able to get the "neuralnet tracker" to follow my face very well, but Star Citizen doesn't seem to be receiving the data

zackadiax24, 2024-02-01:
I am also having this issue and would really appreciate some help.

DarinDev1000, 2025-02-22:
Same issue, Star Citizen doesn't seam to be connected to opentrack.