opentrack/opentrack #1394: There is no tracker/neuralnet in the list.

derrikgort, 2021-12-19:
I don't see no tracker/neuralnet, though the opentrack was run under the admin root. This issue is on 2021.2.0, 2021.1.3.
[image]

sthalik, 2021-12-19:
Probably missing Visual Studio 2015 Redistributable.

derrikgort, 2021-12-19:
All visuals and frameworks are installed.

sthalik, 2021-12-19:
`opentrack-tracker-neuralnet.dll` exists in the modules directory, no?

derrikgort, 2021-12-19:
Of course it exists
[image]
.

sthalik, 2021-12-19:
Then drag opentrack.exe onto a Windows Console window, press Return and see why it fails to load neuralnet.

derrikgort, 2021-12-19:
[image]

sthalik, 2021-12-19:
Then run Dependency Walker. Just need to set up PATH first. Here's how.

[image]
[image]

On my end no libraries are missing, so the thing works:

[image]

But really I should be making `-MT` builds for the ONNX runtime.

derrikgort, 2021-12-19:
Sorry, I'm not a programmer, I can run the opentrack in cmd.exe but then I just don't get what I should set up a path for and what is dependency walker, and what I have to do with them. I just report an issue I got as a user.

GO63-samara, 2021-12-19:
I see that you don't have `neuralnet` and `kinectface` trackers on your list.
Do you have Windows 7? These trackers do not work in Windows 7. You must be using Windows 10.
When I launch Opentrack-21.2.0 from cmd in Windows 7, two libraries are not loaded:
```
D:\TOOLS_WIN\AVIA\OpenTrack\opentrack-2021.2.0>
DEBUG [F:\dev\opentrack\api/plugin-support.hpp:107]: library "kinect-face" faile
d: "Cannot load library D:\\TOOLS_WIN\\AVIA\\OpenTrack\\opentrack-2021.2.0\\modu
les\\\\opentrack-tracker-kinect-face.dll: DEBUG [F:\dev\opentrack\api/plugin-sup
port.hpp:107]: library "neuralnet" failed: "Cannot load library D:\\TOOLS_WIN\\A
VIA\\OpenTrack\\opentrack-2021.2.0\\modules\\\\opentrack-tracker-neuralnet.dll:
```
But in Windows 10 is all fine. The list contains all 17 trackers.

Try this:
Windows 7 runs `neuralnet-0.2` from here:
https://github.com/DaWelter/opentrack/releases
Direct link:
https://github.com/DaWelter/opentrack/releases/download/tracker-neuralnet-0.2/opentrack-portable-tracker-neuralnet-v0.2.7z

derrikgort, 2021-12-19:
> I see that you don't have `neuralnet` and `kinectface` trackers on your list. Do you have Windows 7? These trackers do not work in Windows 7. You must be using Windows 10. When I launch [...]

Yes, you are right, I'm on win7. I've downloaded the build you offered and it works fine...Now all I have to test it in work. I'm using aruco, it works ok, I wonder if the neuralnet will be better...

GO63-samara, 2021-12-19:
@sthalik
I see in your screenshot there is no `kinect-face tracker`, but there is an unknown `TrackHat Point Tracker`. This is fine?

derrikgort, 2021-12-19:
Can the neuralnet be working on the next versions of opentrack under win7? Thanks for advance.

sthalik, 2021-12-19:
@GO63-samara 
> I see in your screenshot there is no kinect-face tracker, but there is an unknown TrackHat Point Tracker. This is fine?

If your `neuralnet` build works, is this because you're building with `-MT`, or for some other reason?

The TrackHat sensor support hasn't been released yet. Kinect is missing because I only have parts of the SDK on the build machine.

sthalik, 2021-12-19:
@derrikgort please see if this build works. It was built with the official release version of onnxruntime. https://www.dropbox.com/s/hllwehp4j8q0cpi/opentrack-master-20211219_01.7z

derrikgort, 2021-12-19:
No, it is still absent... Track hat point tracker appeared ( It wasn't last time too), but there is no neuralnet again..
[image]
.

GO63-samara, 2021-12-19:
> If your neuralnet build works, is this because you're building ONNX with -MT, or for some other reason?

This is not my build, this is the build from @DaWelter.

I didn't manage to build `neuralnet` myself. I did not use the sources, but the `onnxruntime-1.8.1` release. I have configured three lines in `sdk-paths`:

```
setq(ONNXRuntime_INCLUDE_DIR "onnxruntime-1.8.1/include")
setq(ONNXRuntime_LIBRARY "onnxruntime-1.8.1/lib/onnxruntime.lib")
setq(ONNXRuntime_RUNTIME "onnxruntime-1.8.1/lib/onnxruntime.dll")
```
but apparently this is not enough. I got build errors.

GO63-samara, 2021-12-19:
When launched in Windows7 `OT-20211219_01` `neuralnet` does not appear:

```
D:\TOOLS_WIN\AVIA\OpenTrack\OT-20211219_01>
DEBUG [F:\dev\opentrack\api/plugin-support.hpp:106]: library "kinect-face" faile
d: "Cannot load library D:\\TOOLS_WIN\\AVIA\\OpenTrack\\OT-20211219_01\\modules\
\\\opentrack-tracker-kinect-face.dll: DEBUG [F:\dev\opentrack\api/plugin-support
.hpp:106]: library "neuralnet" failed: "Cannot load library D:\\TOOLS_WIN\\AVIA\
\OpenTrack\\OT-20211219_01\\modules\\\\opentrack-tracker-neuralnet.dll:
```

sthalik, 2021-12-20:
See if this build works: https://www.dropbox.com/s/7m27azwabkc9ik2/opentrack-master-20211220_01.7z

DaWelter, 2021-12-20:
Hi guys
thank you for the support.

> but apparently this is not enough. I got build errors.

When I started using the official dll's, setting the ONNX dir like so `set(ONNXRuntime_DIR "D:/dev/onnxruntime-official-1.9-x86-release/" CACHE PATH "" FORCE)` in my `sdk-paths-...@MSVC-windows.cmake` worked for me.

derrikgort, 2021-12-20:
> See if this build works: https://www.dropbox.com/s/7m27azwabkc9ik2/opentrack-master-20211220_01.7z

Finally it works! Thanks. It works a bit strange, something wrong with centering, but perhaps it needs adjusting, anyway I have to test it later.

GO63-samara, 2021-12-20:
Yes, in Windows 7, when launching OT-20191220_01, neuralnet appeared.
Maybe in a similar way, you can launch kinect-face in Windows 7?
```
D:\TOOLS_WIN\AVIA\OpenTrack\OT-20191220_01>
DEBUG [F:\dev\opentrack\api/plugin-support.hpp:106]: library "kinect-face" faile
d: "Cannot load library D:\\TOOLS_WIN\\AVIA\\OpenTrack\\OT-20191220_01\\modules\
\\\opentrack-tracker-kinect-face.dll:
```
Thanks!

sthalik, 2021-12-21:
@GO63-samara,
> Maybe in a similar way, you can launch kinect-face in Windows 7?

Kinect SDK is binary with no source available. There's nothing that could be linked statically.

sthalik, 2021-12-21:
It's no longer necessary to run `cmake .` after adding or removing a file, or switching branches.

```cmake
file(GLOB cxx CONFIGURE_DEPENDS "${dir}/*.cpp")
```

Now the build system does that by itself.

GO63-samara, 2021-12-21:
> When I started using the official dll's, setting the ONNX dir like so set(ONNXRuntime_DIR "D:/dev/onnxruntime-official-1.9-x86-release/" CACHE PATH "" FORCE) in my sdk-paths-...@MSVC-windows.cmake worked for me.

@DaWelter
Yes, this line replaces three lines at once and does the same:
```
#setq(ONNXRuntime_INCLUDE_DIR "onnxruntime-1.8.1/include")
#setq(ONNXRuntime_LIBRARY "onnxruntime-1.8.1/lib/onnxruntime.lib")
#setq(ONNXRuntime_RUNTIME "onnxruntime-1.8.1/lib/onnxruntime.dll")

setq(ONNXRuntime_DIR "onnxruntime-1.8.1" CACHE PATH "" FORCE)
```
But I still get the error on the `ninja install` command:
```
ftnoir_tracker_neuralnet.cpp.obj : error LNK2001: unresolved external symbol _OrtGetApiBase@0
tracker-neuralnet\opentrack-tracker-neuralnet.dll : fatal error LNK1120: 1 unresolved externals
```
Any ideas?

sthalik, 2022-01-10:
It works for me when using the toolchain file. Disabled all cmake options with `SHARED` in the name, except for `onnxruntime_BUILD_SHARED_LIB`.

Then,

```cmake
FLAGS_LD:STRING=-WX:NO dbghelp.lib # toolchain specific
ONNX_USE_MSVC_STATIC_RUNTIME:BOOL=TRUE
```

@GO63-samara, can you verify that there's `onnxruntime.lib` on the linker invocation command-line?

GO63-samara, 2022-01-10:
@sthalik 
I tried to rebuild the `opentrack` today
I re-cloned `opentrack` and `opentrack-depends` 
I was unable to build the release.

Problem in `ninja install` command after connecting in `sdk-paths` path to OpenCV:

A lot of warning C4275 appeared like this:
```
Building CXX object filter-kalman\CMakeFiles\opentrack-filter-kalman.dir\moc_kalman.cpp.obj
D:\Dev\master\opentrack\api/plugin-api.hpp(94): warning C4275: non dll-interface struct 'TR' used as base for dll-interf
ace class 'Metadata'
D:\Dev\master\opentrack\api\../compat/tr.hpp(8): note: see declaration of 'TR'
D:\Dev\master\opentrack\api/plugin-api.hpp(94): note: see declaration of 'Metadata'
```

And the error appeared:
```
fatal error C1047: The object or library file 'D:\Dev\master\opentrack\..\..\opentrack-depends\libusb-msvc-x86\libusb-1.
0.lib' was created by a different version of the compiler than other objects like 'video-ps3eye\ps3eye-driver\CMakeFiles
\ps3eye-mode-test.dir\mode-test.cxx.obj'; rebuild all objects and libraries with the same compiler
LINK : fatal error LNK1257: code generation failed
[106/619] Linking CXX shared library compat\opentrack-compat.dll
ninja: build stopped: subcommand failed.
```

Maybe it's incompatible with **MSVC2019**?

sthalik, 2022-01-11:
Rebuild libusb from version 1.0.23 as a static binary (after adjusting .vcxproj for -MT and -GL). Then replace `opentrack-depends/libusb-msvc-x86/libusb-1.0.lib`.

GO63-samara, 2022-01-11:
I have no idea how to build libusb.
Instead, I took the previous version of libusb, before the update https://github.com/opentrack/opentrack-depends/commit/4dc314e16298f7243988a4b0e68b4c797c04d7d6

My main mistake was that I didn't rebuild OpenCV after the upgrade. Now I have rebuilt openCV again, with an updated `msvc.msvc` toolchain.
I also included the `WITH_ONNX ON` flag.

The `ninja install` command still has a lot of warnings like this:
```
[264/619] Building CXX object tracker-udp\CMakeFiles\opentrack-tracker-udp.dir\ftnoir_tracker_udp_dialog.cpp.obj
D:\Dev\master\opentrack\api/plugin-api.hpp(94): warning C4275: non dll-interface struct 'TR' used as base for dll-interf
ace class 'Metadata'
D:\Dev\master\opentrack\api\../compat/tr.hpp(8): note: see declaration of 'TR'
D:\Dev\master\opentrack\api/plugin-api.hpp(94): note: see declaration of 'Metadata'
```

The release now includes video trackers and even a neuralnet tracker. And it even works!

[image]

When I close the built Opentrack, I get a question in cmd:
```
D:\Dev\opentrack-build\install>
[ INFO:0] global C:\OpenCV\opencv-4.5.2\modules\core\src\parallel\registry_parallel.impl.hpp (96) cv::parallel::Parallel
BackendRegistry::ParallelBackendRegistry core(parallel): Enabled backends(0, sorted by priority): N/A
WARN QWinEventNotifier: no event dispatcher, application shutting down? Cannot deliver event.

D:\Dev\opentrack-build\install>

```
Thanks for the support.

I think you are better able to update your `libusb` version in `opentrack-depends`, right?

sthalik, 2022-01-11:
> I have no idea how to build libusb.

```console
% git clone https://github.com/libusb/libusb.git
% cd libusb
% git checkout v1.0.23
Note: switching to 'v1.0.23'.
% start msvc/libusb_static_2017.vcxproj
```

Upgrade the SDK to latest. Switch to Release. Build.

Replace `libusb.h`, `libusb-1.0.lib` and `libusb-1.0.pdb` from the files that were created.

> I think you are better able to update your libusb version in opentrack-depends, right?

It's made for VS 2022.

GO63-samara, 2022-01-12:
Thanks, this is a good guide.

I built `libusb v1.0.23` in MSVC2019.  /MT and /GL have already been set in `libusb Debug Properties`.

The `ninja install` command still has a lot of warnings like:
```
[449/627] Building CXX object spline\CMakeFiles\opentrack-spline.dir\axis-opts.cpp.obj
D:\Dev\master\opentrack\api/plugin-api.hpp(94): warning C4275: non dll-interface struct 'TR' used as base for dll-interf
ace class 'Metadata'
D:\Dev\master\opentrack\api\../compat/tr.hpp(8): note: see declaration of 'TR'
D:\Dev\master\opentrack\api/plugin-api.hpp(94): note: see declaration of 'Metadata'
```
But the question disappeared when exiting opentrack.

I think it would be more correct for me to install MSVC2022 and build OT in it.

kanzaki-kunbr, 2022-04-21:
helo I'm having a problem in neuralnet it doesn't appear in opentrack
[image]

sthalik, 2022-04-21:
@kanzaki-kunbr Install the Visual Studio 2015 Redistributable. I'm going to fix this for future releases.

kanzaki-kunbr, 2022-04-21:
@sthalik 
[image]

sthalik, 2022-04-21:
@kanzaki-kunbr install the 32-bit version for VC++ 2015.

sthalik, 2022-04-21:
@kanzaki-kunbr the simplest solution is this: https://www.techpowerup.com/download/visual-c-redistributable-runtime-package-all-in-one/

kanzaki-kunbr, 2022-04-21:
[image]

it worked thanks for the help

jorgelfrias89, 2023-03-03:
Hola por alguna razón que desconozco cuando ejecuto el opentrack no me sale en la lista el neuralnet, estuve viendo arriba y tengo las librerías de c++ todas instaladas tanto en 32bit como en 64bit, uso SO Windows 10

Aniubik, 2024-02-12:
Also having same problem. Already tried everything and still not working. I have al Visuals installed and still nothing. This is showing up in cmd: DEBUG [F:\dev\opentrack\api/plugin-support.hpp:106]: library "neuralnet" failed: "Cannot load library C:\\Program Files (x86)\\opentrack\\modules\\\\opentrack-tracker-neuralnet.dll: A dynamic link library (DLL) initialization routine failed." I have no programing skills, can someone help me please?

sthalik, 2024-02-12:
It's a Windows 7 issue that now affects way more trackers and I'm looking into it right now.

sthalik, 2024-02-12:
I switched to `-MD` and guess what? it works

sthalik, 2024-02-12:
Now's the time you test the build: https://ananke.misaki.pl/opentrack/opentrack-test-20240212_01.7z

Tguy3100, 2024-02-12:
I seem to be experiencing the same issue neuralnet does not show up in the list however I do have all the redistributables installed Im running windows 10  

DEBUG [F:\dev\opentrack\api/plugin-support.hpp:106]: library "neuralnet" failed: "Cannot load library C:\\Users\\thatg\\OneDrive\\Desktop\\Open track\\install\\modules\\\\opentrack-tracker-neuralnet.dll: A dynamic link library (DLL) initialization routine failed."

not sure why onedrive is involved but I could sure use some help

stewi2, 2024-02-24:
I'm on Windows 11 and am having the same issue using the latest build. Perhaps this is because I'm running on ARM? Will try to build the DLL myself for lack of a better idea.

sthalik, 2024-02-29:
@DaWelter can you make a temporary build with tracker/nn that works? Thanks.

CaptainCorgi269, 2024-03-04:
also couldnt seem to find the neuralnet tracker in windows 11 arm 64 just to mention.  Am not a programmer so im not so sure, i will fiddle around to find a solution.

sthalik, 2024-03-19:
@Leko20001 https://github.com/opentrack/opentrack/issues/1394#issuecomment-1938864870

Leko20001, 2024-03-19:
It doesn't work on me

Leko20001, 2024-03-19:
Also mine is Windows 10

sthalik, 2024-03-19:
You can use 2023.1.0.

Leko20001, 2024-03-20:
I have tried to run on 2023.1.0 and I cant even run it