opentrack/opentrack #1395: Formatted my pc, using the same profiles but opentrack acts so weird that i cant even tune it out.

hatuk, 2021-12-20:
I've formatted my system last night. Before that, opentrack was running all fine in DCS. After the format, i installed the new version from here, but as i said even if i use my old profiles it doesnt acts the same and its really annoying, cant find a fix.

TrikePJ, 2021-12-27:
I formatted my PC too and now tracking is very Jittery! I reverted back to 2.3.13. I still have Issues but they are far less then the latest release

dglava, 2021-12-27:
Other people have experienced a change in tracking behaviour with the latest versions too. See the discussion here for possible hints and solutions: https://github.com/opentrack/opentrack/discussions/1381

sthalik, 2021-12-27:
@hatuk @TrikePJ is this the PT tracker? Then, are you using auto treshold and/or the open ps3eye driver? 

Working with @dglava and @Vortex76, I've been unable to pinpoint the exact change causing these issues. The results were contradictory (between the users and for each individual users as well).

TrikePJ, 2021-12-27:
[image]
[image]
This are my settings

sthalik, 2021-12-27:
@TrikePJ first see if enabling auto threshold helps at all. The outlines should be slightly larger than points as visible in the preview. One of the users I helped with this via IM said that it solved the 2.3.1.3+ issue.