opentrack/opentrack #1396: UDP over network flipping

Detr0y, 2021-12-25:
I play DCS on a service called Shadow PC which provides me with a better computer than my local hardware. To use head tracking, I have to use UDP over network to patch my head tracker data to the virtual machine. When I look up or down far enough, the tracker on the virtual machine flips, however the one on the local machine is fine. I've had this issue for as long as I've been using UDP over network. I've tried everything, but nothing fixes the issue.

I've attached a video that visualizes my issue:
https://user-images.githubusercontent.com/74055427/147387698-902a3f54-2c5c-4e21-a954-603a8d712126.mp4

My tracking hardware:
Modified PS3 eye camera
Trackhat clip (right side)

GO63-samara, 2021-12-25:
I would try the following steps for virtual machine:

1. Correct the mapping curves in the virtual machine so that they are 1 to 1 diagonal. `Max input` for Yaw and Roll should be 180. `Max input` and `Max output` should be 90 for Pitch.
2. The `dead zone` is not needed in the center of the mapping.
3. Disable `Center at startup` and set `Relative translation` to `Disabled`.
4. The `filter` must also be disabled.

These steps should provide a 1-in-1 overview on real and virtual machines.
But it needs to be checked.

Detr0y, 2021-12-25:
Did what you told me to, the issue persists. I've noticed that the flipping starts when the pitch goes above 90 on my local PC. Setting pitch input & output to 90 does not solve it; it looks like the tracker keeps tracking and relaying data to the virtual machine even with these constraints set.

Detr0y, 2021-12-25:
I have a soft fix for this issue: don't let the local tracker's pitch go over 90. I've set max pitch input & output to 90, left the curve as an increasing line, and set a pitch curve on my virtual machine. From the looks of it, this fixed my issue.

sthalik, 2021-12-26:
The best practice is to pass data unchanged on the sender as-is, i.e. disable centering, filtering and use 1:1 mappings (i.e. the default when you delete all control points). Now the bit your issue seems to stem from, set max pitch input:180, max pitch output:180. Same for other mappings on the sender, set max input:180 for yaw/roll and 75 for translation.

DanielKinsman, 2022-04-06:
I ran into the same issue. Regardless of what "best practice" is there is clearly a bug in the udp tracker somewhere. Here is another video for what it is worth.

https://user-images.githubusercontent.com/15639/161938892-8b54f213-223a-4aaa-85b8-aad055ce7e1e.mp4

GO63-samara, 2022-04-06:
This is not visible in the video, but you need to check:
On the remote machine, you need to disable the filter and set the maximum Pitch=180

MrFreaky430, 2022-04-25:
> I ran into the same issue. Regardless of what "best practice" is there is clearly a bug in the udp tracker somewhere. Here is another video for what it is worth.
> 
>  opentrackbug.mp4

how you set this via UDP ?