opentrack/opentrack #1397: Aruco on Linux

franciscot, 2021-12-27:
Hi,
Is it Aruco tracker compatible with Linux?
I can't build with aruco on manjaro or ubuntu,

Thanks,

sthalik, 2021-12-27:
Yes. Use the fork. https://github.com/opentrack/aruco

franciscot, 2021-12-28:
Is there any complete building guide? I'm not a Linux developer. I'm a normal user and It is very difficult to try to understand how to build everything. 
I'm fighting to build against WINE, x-plane plugin or aruco. For example.

franciscot, 2022-01-01:
I found this bash script and It' was very useful for arch/manjaro: https://github.com/JT8D-17/opentrack/blob/main/z_Build.sh
It was easy to adapt to debian/ubuntu.

I'm thinking to do a guide for normal Linux users. 

Sorry but I don't agree with the FAQ about "It's assumed Linux users are software developers", there are a lot not-dev users too. ;-)

closed 2022-01-01 by franciscot

sthalik, 2022-01-03:
> Sorry but I don't agree with the FAQ about "It's assumed Linux users are software developers", there are a lot not-dev users too. ;-)

This is why it's an assumption. It's not falsifiable.

franciscot, 2022-01-05:
Well, Linux users have a new script to install opentrack and plugins. It's not for beginners user yet.
Sorry, the instructions are in Spanish:
https://gitlab.com/jugandoenlinux/opentrack-script